esphome/issues #460: PCF8574 don't working if INPUT connect to ground on the start

IvanHorban, 2019-06-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):*ESP8266-07*

**Affected component:*PCF8574*

**Description of problem:*If the input is connected to the ground when ESP8266 loading , it does not react to the state change.*

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp8266_work_01
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: ""
  password: ""

# Enable logging
logger:
   level: VERY_VERBOSE

# Enable Home Assistant API
api:
  password: '1234'

ota:
  password: '1234'

status_led:
  pin:
    number: GPIO2
    inverted: True
i2c:
  sda: 4
  scl: 5
  scan: True
dallas:
  - pin: 13
    update_interval: 30000 ms
#sensor:
#  - platform: dallas
#    address: 0x610316643E3FFF28
#    resolution: 12
#    name: "Office Temperature"
#-------------------------------------
#  - platform: uptime
#    name: Uptime Sensor

pcf8574:
  - id: 'pcf8574_hub'
    address: 0x20
    pcf8575: False

# Individual outputs
binary_sensor:
  - platform: gpio
    name: "Input_01"
    device_class: window
    pin:
      pcf8574: pcf8574_hub
      number: 0
      mode: INPUT
      inverted: False
  - platform: status
    name: "Office Temperature"
```

**Traceback (if applicable):**
[... 350 lines cut ...]
[16:45:58][VV][i2c:113]:     Received 0b11111110 (0xFE)
[16:45:58][VV][i2c:083]: Requesting 1 bytes from 0x20:
[16:45:58][VV][i2c:113]:     Received 0b11111110 (0xFE)
[16:45:58][VV][i2c:083]: Requesting 1 bytes from 0x20:
[16:45:58][VV][i2c:113]:     Received 0b11111110 (0xFE)
[16:45:58][VV][i2c:083]: Requesting 1 bytes from 0x20:
[16:45:58][VV][i2c:113]:     Received 0b11111110 (0xFE)
[16:45:58][VV][i2c:083]: Requesting 1 bytes from 0x20:
[16:45:58][VV][i2c:113]:     Received 0b11111110 (0xFE)
[16:45:58][VV][i2c:083]: Requesting 1 bytes from 0x20:
[16:45:58][VV][i2c:113]:     Received 0b11111110 (0xFE)
[16:45:58][VV][i2c:083]: Requesting 1 bytes from 0x20:
[16:45:58][VV][i2c:113]:     Received 0b11111110 (0xFE)
[16:45:58][VV][i2c:083]: Requesting 1 bytes from 0x20:
[16:45:58][VV][i2c:113]:     Received 0b11111110 (0xFE)
[16:45:58][VV][i2c:083]: Requesting 1 bytes from 0x20:
[16:45:58][VV][i2c:113]:     Received 0b11111110 (0xFE)
[16:45:58][VV][i2c:083]: Requesting 1 bytes from 0x20:
[16:45:58][VV][i2c:113]:     Received 0b11111110 (0xFE)
[16:45:58][VV][i2c:083]: Requesting 1 bytes from 0x20:
[16:45:58][VV][i2c:113]:     Received 0b11111110 (0xFE)
[16:45:58][VV][i2c:083]: Requesting 1 bytes from 0x20:
[16:45:58][VV][i2c:113]:     Received 0b11111110 (0xFE)
```
**Additional information and things you've tried:**

OttoWinter, 2019-06-19:
> If the input is connected to the ground when ESP8266 loading , it does not react to the state change.

Could you be more specific? What input is connected to ground? What point in time do you mean by ESP8266 loading?

From the logs I only see that the PCF8574 is sending 0xFE and 0xFF data - for me it sounds like the issue is with the PCF8574 unit and not the sw.

IvanHorban, 2019-06-19:
The PCF8574 module has 8 pins. If at the moment of the start of the ESP8266 any pin is connected to the ground, it does not react to a change of state. To fix this you need to unplug it from the ground, restart the ESP8266. Then, when connected or disconnected this pin from the ground, this message will appear in the logs. ('Input_01': Sending state OFF or 'Input_01': Sending state ON)

OttoWinter, 2019-06-20:
Sounds like a problem with the chip again.

The PCF has a really simple protocol - to read the chip state ESPHome only reads 8bits from the chip over i2c which represents the binary state of each pin. There's notable no setup routine or so involved, so there can't be an issue with wrong sw setup.

onkelbeh, 2019-07-14:
Had a similar issue some days ago. I replaced the PCF8574 board and everything worked as expected. I believe there are some bad fake chips on the market. Here's a pic of the bad module:

[image]

OttoWinter, 2019-10-12:
I guess that concludes the issue, I'm pretty certain there's nothing wrong with the code (it's mainly only 1 line, and I checked that pretty thoroughly). The bad chip explanation would also explain why it only happens to some people.

closed 2019-10-12 by OttoWinter