esphome/issues #462: Reading scratchpad from Dallas Sensor failed

nicholh, 2019-06-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 - Sonoff Basic
**Affected component:**

https://esphome.io/components/sensor/dallas.html

**Description of problem:**
I am having an issue whereby I get the error Reading scratchpad from Dallas Sensor failed using a DS18B20 sensor

The interersting thing is that

it works after the first boot following a flash. After subsequent power cycles it gets this message.
I have an mqqt on_message section in the config - when i remove this it works for every boot. The on_message config works.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
mqtt:
  broker: hidden
  discovery: false
  topic_prefix: home/beer_chiller_thermostat
  on_message:
    - topic: home/beer_chiller_thermostat/climate/thermostat/target_temperature/command
      qos: 0
      then:
        - mqtt.publish:
            topic: some/topic
            payload: "Something happened!"

dallas:
  - pin:  GPIO14
    update_interval: 15s

sensor:
  - platform: dallas
    index: 0
    id: current_temperature
    name: current_temperature
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

I have treid removing the mqtt on_message section and it works fine.

OttoWinter, 2019-06-19:
Please post your setup here, how you wired the sensor and if you're using any resistors or parasitic power mode.

Sounds a lot like a HW issue to me.

nicholh, 2019-06-20:
Thanks for your responce OttoWInter :)

I have a single sensor (DS18B20) that is connected to the sonoff. It is connected using three wires Vcc, Gnd and Data with data on GPIO14, GPIO14, VCC and GND are all accessced via the five pin header on the basic.

I have a 10K resistor between data and vcc, however I have tried with no resistor and have the same experince. The lead to the sensor is around 1m long. 

I thought hardware at first also, however all I have to do is comment out the on_message section as below and it works reliably - I was not sure and do not have good enougth software skills to debug to this level.
'''
mqtt:
  broker: hidden
  discovery: false
  topic_prefix: home/beer_chiller_thermostat
#  on_message:
#    - topic: home/beer_chiller_thermostat/climate/thermostat/target_temperature/command
#      qos: 0
#      then:
#        - mqtt.publish:
#            topic: some/topic
#            payload: "Something happened!"
''''

OttoWinter, 2019-07-12:
I'm going to tag this as a probable instance of https://github.com/esphome/issues/issues/330#issuecomment-503752668

The fact that it gets "fixed" by commenting out code, means it's probably not directly related to software. These parts of code do not communicate with each other, hardware like insufficient power supply is more likely.

closed 2019-07-12 by OttoWinter

nicholh, 2019-07-12:
Thanks,

I have worked around the problem by using a mqtt_subscribe sensor rather than on_message. once i change to this the Dallas sensor started working fine.