esphome/issues #463: Binary sensor with delayed_on + delayed_off

Erelen-Laiquendi, 2019-06-19:
**Affected component:** Binary Sensor - https://esphome.io/components/binary_sensor/

**Description of problem:**
Wrong work with delayed_on + delayed_off
As I understand, in example below, binary sensor shoul not turn off at all, because there was only 2 seconds of long distance, but delayed_off is 5 seconds.

**Problem-relevant YAML-configuration entries:**
```yaml
binary_sensor:
  - platform: template
    name: "Test sensor"
    lambda: |-
      if (id(ultrasonic_sensor).state < 0.5) {
        return true;
      } else {
        return false;
      }
    filters:
      - delayed_on: 5s
      - delayed_off: 5s
```

**Traceback (if applicable):**
```
[00:34:49][D][ultrasonic.sensor:030]: 'ultrasonic_sensor' - Got distance: 0.16 m
[00:34:50][D][ultrasonic.sensor:030]: 'ultrasonic_sensor' - Got distance: 0.16 m
[00:34:51][D][ultrasonic.sensor:030]: 'ultrasonic_sensor' - Got distance: 0.16 m
[00:34:52][D][ultrasonic.sensor:030]: 'ultrasonic_sensor' - Got distance: 0.16 m
[00:34:53][D][ultrasonic.sensor:030]: 'ultrasonic_sensor' - Got distance: 0.16 m
[00:34:54][D][ultrasonic.sensor:030]: 'ultrasonic_sensor' - Got distance: 1.29 m <-- long distance
[00:34:55][D][ultrasonic.sensor:030]: 'ultrasonic_sensor' - Got distance: 1.33 m <-- long distance
[00:34:56][D][ultrasonic.sensor:030]: 'ultrasonic_sensor' - Got distance: 0.18 m
[00:34:57][D][ultrasonic.sensor:030]: 'ultrasonic_sensor' - Got distance: 0.16 m
[00:34:58][D][ultrasonic.sensor:030]: 'ultrasonic_sensor' - Got distance: 0.16 m
[00:34:59][D][ultrasonic.sensor:030]: 'ultrasonic_sensor' - Got distance: 0.16 m
[00:34:59][D][binary_sensor:033]: 'Test sensor': Sending state OFF
[00:35:00][D][ultrasonic.sensor:030]: 'ultrasonic_sensor' - Got distance: 0.16 m
[00:35:01][D][ultrasonic.sensor:030]: 'ultrasonic_sensor' - Got distance: 0.18 m
[00:35:01][D][binary_sensor:033]: 'Test sensor': Sending state ON
[00:35:02][D][ultrasonic.sensor:030]: 'ultrasonic_sensor' - Got distance: 0.16 m
[00:35:03][D][ultrasonic.sensor:030]: 'ultrasonic_sensor' - Got distance: 0.15 m
[00:35:04][D][ultrasonic.sensor:030]: 'ultrasonic_sensor' - Got distance: 0.16 m
[00:35:05][D][ultrasonic.sensor:030]: 'ultrasonic_sensor' - Got distance: 0.16 m
[00:35:06][D][ultrasonic.sensor:030]: 'ultrasonic_sensor' - Got distance: 0.17 m
[00:35:07][D][ultrasonic.sensor:030]: 'ultrasonic_sensor' - Got distance: 0.16 m
```

**Additional information and things you've tried:**

OttoWinter, 2019-06-20:
The behavior is correct. The main thing to note is that binary sensor filters are processed in a pipeline (that means order matters!).

Base sensor generates these events:
 - Start: True
 - 0:34:54: False
 - 0:34:56: True

After delayed_on filter:
 - Start: True
 - 0:34:54: False
 - 0:34:56+5=0:35:01: True

After delayed_on+delayed_off:
 - Start: True
 - 0:34:54+5=0:34:59: False
 - 0:35:01: True

Erelen-Laiquendi, 2019-06-20:
That's true algorithmically, but make totally unusable delayed_on + delayed_off practically.

delayed_on - for debouncing button on press.
delayed_off - for debouncing button on unpress.
delayed_on + delayed_off - as result, I haven't debouncing on unpress, but have "bouncing with delay".

From docs, that describes perfect, useful behaviour, that matches exactly with common sence:

> delayed_off: When a signal OFF is received, wait for the specified time period until publishing an OFF state. If an ON value is received while waiting, the OFF action is discarded.

In my example, there is ON while waiting, but OFF was not discarded.

> Or in other words: Only send an OFF value if the binary sensor has stayed OFF for at least the specified time period.

There is no OFF for specified period in my example.

OttoWinter, 2019-06-20:
I'm not saying a new filter can't be added for this use case - if you have a good description of how this new binary_sensor filter would work and what it should be named, you can open a feature request for it no problem.

Also note that since order matters, for your case switching delayed_on/off around will yield the correct result (as long as the binary sensor is "normally ON").

As far as I see, the code does work as intended - the binary sensor is correctly debounced with the described algorithm. Feature request for the use case is welcome.

closed 2019-06-20 by OttoWinter