esphome/issues #464: cant compile cant connect

g14nn86, 2019-06-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio
**ESP (ESP32/ESP8266, Board/Sonoff):**

D1 mini
**Affected component:**

**Description of problem:**
I cant compile 

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: d1_mini_esphome
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "SSID"
  password: "password"
  manual_ip:
    static_ip: 192.168.1.240
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.1
    dns2: 0.0.0.0

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'password'

ota:
  password: 'password'

switch:
  - platform: gpio
    pin: D5
    name: "Living Room Dehumidifier"
    id: dehumidifier1
  - platform: gpio
    pin: D7
    name: "Living Room Dehumidifier2"
    id: dehumidifier2
  - platform: gpio
    pin: D6
    name: "Living Room Dehumidifier3"
    id: dehumidifier3
  - platform: gpio
    pin: D0
    name: "Living Room Dehumidifier4"
    id: dehumidifier4  

binary_sensor:
  - platform: gpio
    pin: D3
    name: "Living Room Dehumidifier Toggle Button1"
    on_multi_click:
    - timing:
       - ON for 1s to 2s
       - OFF for at least 0.5s
      then:
        - switch.toggle: dehumidifier1
  - platform: gpio
    pin: D3
    name: "Living Room Dehumidifier Toggle Button2"
    on_multi_click:
    - timing:
        - ON for at most 1s
        - OFF for at least 0.5s
      then:
        - switch.toggle: dehumidifier2
  
  - platform: gpio
    pin: D4
    name: "Living Room Dehumidifier Toggle Button3"
    on_double_click:
      then:
        - switch.toggle: dehumidifier3
  - platform: gpio
    pin: D4
    name: "Living Room Dehumidifier Toggle Button4"
    on_press:
      then:
        - switch.toggle: dehumidifier4    

    
   
```

**Traceback (if applicable):**

```
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/d1_mini_esphome
Processing d1_mini_esphome (framework: arduino; platform: espressif8266@1.8.0; board: d1_mini)
--------------------------------------------------------------------------------
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ESP8266WiFi%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f909ba490>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)) (incremented retry: max=5, total=1)
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ESP8266WiFi%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f909ba690>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)) (incremented retry: max=5, total=2)
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ESP8266WiFi%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f909ba250>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)) (incremented retry: max=5, total=3)
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ESP8266WiFi%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f909babd0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)) (incremented retry: max=5, total=4)
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ESP8266WiFi%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f90a13890>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)) (incremented retry: max=5, total=5)
Error: [API] Could not connect to PlatformIO API Service. Please try later.
```

**Additional information and things you've tried:**

I can see the d1 mini i can control it and it is online but when i try to change the code and compile upload i get this.

i also using Nginx and i was using dnsmasq i disable it.

Any help?

g14nn86, 2019-06-20:
i also tried restart home assistant restart esp home and after all reinstall. I tried creating a new one device also the same thing,

OttoWinter, 2019-06-20:
Duplicate of https://github.com/esphome/issues/issues/302

closed 2019-06-20 by OttoWinter