esphome/issues #467: binary sensor with "delayed_on" for SW-420 never fires #466

jeffehobbs, 2019-06-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io/ESPHome add-on

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32-WROOM-32

**Affected component:**

binary_sensor

**Description of problem:**

If I add a "delayed_on" to the binary_sensor connected to the SW-420, the event never fires.

**Problem-relevant YAML-configuration entries:**

This code works fine:
```yaml

# Enables the SW-420 Vibration Sensor
binary_sensor:  
  - platform: gpio  
    name: Dryer  
    pin:
      number: GPIO13
      mode: INPUT
    device_class: vibration
    filters:
      - delayed_off: 5s

```

...but if I add a "delayed_on: 5s" to that YAML, the event will never fire.

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

Tried different measures of time (s, ms). Tried different order of statements. no luck.

OttoWinter, 2019-06-20:
Please attach debug logs to your post that show when the binary sensor turns off/on.

I think you might not understand what delayed_on does. If your vibration sensor is outputting a rapidly pulsing ON/OFF value and you wish to get an ON value during that time, you must use `delayed_off`. `delayed_on` - as the docs state - only forwards ON events if they have been longer than the configured timeout.

jeffehobbs, 2019-06-20:
I think you're right -- I think I'm misunderstanding the use case/utility of `delayed_on`. I'm going to close this up; thanks very much! Love the ESP Home project, it's so much fun.

closed 2019-06-20 by jeffehobbs