esphome/issues #471: Dual relay cover motor control

hugodelaroza, 2019-06-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio
**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff Dual
**Affected component:**

https://www.esphome.io/cookbook/dual-r2-cover.html

**Description of problem:**
I´ve implemented a Dual relay cover motor control exactly as the cookbook shows, and it work very fine. The problem is that when I have a power failure and the power is back again, the contol doesnt remember which was the cove position, so y have to close the cover (it closes to half the way), cut cover control power for 5 seconds, and power the cover control again, close it and take the real state again...

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: persiana01
  platform: ESP8266
  board: esp01_1m

wifi:
  manual_ip:
    static_ip: 192.168.0.140
    gateway: 192.168.0.1
    subnet: 255.255.255.0
  
  ssid: 'XXXXXX'
  password: 'XXXXXXXX'

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'XXXXXXXX'

ota:
  password: 'XXXXXXXXXX'

web_server:
  port: 80
  
binary_sensor:
- platform: gpio
  pin:
    number: GPIO10
    inverted: true
  id: button
  on_press:
    then:
      # logic for cycling through movements: open->stop->close->stop->...
      - lambda: |
          if (id(my_cover).current_operation == COVER_OPERATION_IDLE) {
            // Cover is idle, check current state and either open or close cover.
            if (id(my_cover).is_fully_closed()) {
              id(my_cover).open();
            } else {
              id(my_cover).close();
            }
          } else {
            // Cover is opening/closing. Stop it.
            id(my_cover).stop();
          }

switch:
- platform: gpio
  pin: GPIO12
  interlock: &interlock [open_cover, close_cover]
  id: open_cover
- platform: gpio
  pin: GPIO5
  interlock: *interlock
  id: close_cover

cover:
- platform: time_based
  name: "Cover"
  id: my_cover
  open_action:
    - switch.turn_on: open_cover
  open_duration: 20s
  close_action:
    - switch.turn_on: close_cover
  close_duration: 20s
  stop_action:
    - switch.turn_off: open_cover
    - switch.turn_off: close_cover
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-06-20:
See `esp8266_restore_from_flash` option.

The cookbook could be updated to include that option.

hugodelaroza, 2019-06-20:
Ok Otto, thank you very much! I really appreciate your work! I'll try this

Thanks!

closed 2019-06-23 by OttoWinter