esphome/issues #472: Slow performance with multiple remote receive binary sensors

PaulN33, 2019-06-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io running on ubuntu 18.04 lts
**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32

**Affected component:**
remote receive https://esphome.io/components/remote_receiver.html

**Description of problem:**
i am trying to create a 433mhz bridge to receive codes from pir sensors, defining anymore than 3 binary sensors results in vastly reduced response time, 3 or less picks up a pir trip in less than a second, 4+ takes between 5-8 seconds, is this to be expected? simply a limitation of the esp32?
is it fixable or is there a better way to do it?

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: ble_433mhz_receiver
  platform: ESP32
  board: esp32dev

wifi:
  ssid: "**********"
  password: "**********"

# Enable logging
logger:
# Enable Home Assistant API
api:
  password: "**********"

ota:
  password: "**********"
remote_receiver:
  pin:
    number: GPIO 15
    mode: INPUT_PULLUP
    inverted: False
  dump: rc_switch
  filter: 4us
  idle: 2ms
  tolerance: 60%
binary_sensor:
  - platform: remote_receiver
    name: "living room pir"
    device_class: motion
    rc_switch_raw:
      code: '110100100111011111101110'
      protocol: 1
    filters:
      - delayed_off: 1s
  - platform: remote_receiver
    name: "kitchen pir"
    device_class: motion
    rc_switch_raw:
      code: '110100110100100110001110'
      protocol: 1
    filters:
      - delayed_off: 1s
  - platform: remote_receiver
    name: "hallway pir"
    device_class: motion
    rc_switch_raw:
      code: '110100101100110001011110'
      protocol: 1
    filters:
      - delayed_off: 1s
#  - platform: remote_receiver
#    name: "bathroom"
#    device_class: motion
#    rc_switch_raw:
#      code: '110100101100110001111111'
#      protocol: 1
#    filters:
#      - delayed_off: 1s
#  - platform: remote_receiver
#    name: "test"
#    device_class: motion
#    rc_switch_raw:
#      code: '1101001111111110001111111'
#      protocol: 1
#    filters:
#      - delayed_off: 1s
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**
curiously turning logging off appears to cause a performance penalty of 1-2 seconds, also i think i have seen the performance with only 3 defined binary sensors degrade over time and improve upon a reboot - but i need more time to observe this thoroughly

OttoWinter, 2019-06-20:
No, it is not expected - and I don't _think_ the binary_sensors can cause something like that (they only decode the data and check if it matches; the computation for that is very simple).

Please attach debug logs while you see these slowdowns - without them it's pretty hard to understand what's going on.

PaulN33, 2019-06-20:
I've copied debug logs into my original post

OttoWinter, 2019-06-20:
@PaulN33 From the debug logs I'm not seeing any slowdown (and I only see 3 binary sensors). Could you show the debug logs with 4+ binary sensors and also annotate them with comments so that what's happening can be understood?

PaulN33, 2019-06-20:
fairly sure the code that was running when i took those debug logs had 5 binary sensors defined but i only tripped two, these are pir sensors, the delays i'm referring to are between me seeing the pir trigger and when i see the esp dump it to log, there is no obvious time difference between seeing the log and home assistants front end, anyway i will repeat the test to make sure 5 were defined and will trip all 5

PaulN33, 2019-06-20:
[19:55:18][D][debug:023]: ESPHome version 1.13.6
[19:55:18][D][debug:025]: Free Heap Size: 245348 bytes
[19:55:18][D][debug:053]: Flash Chip: Size=4096kB Speed=40MHz Mode=DIO
[19:55:18][D][debug:086]: Chip: Model=ESP32, Features=WIFI_BGN,BLE,BT, Cores=2, Revision=1
[19:55:18][D][debug:088]: ESP-IDF Version: v3.3-beta1-179-ge931fe9f5-dirty
[19:55:18][D][debug:091]: EFuse MAC: 000050C722A4AE30
[19:55:18][D][debug:143]: Reset Reason: Software Reset CPU
[19:55:18][D][debug:186]: Wakeup Reason: Unknown
[19:55:22][D][api:573]: Client 'Home Assistant 0.94.4 (192.168.0.162)' connected successfully!
[19:55:28][D][binary_sensor:033]: 'living room pir': Sending state ON
[19:55:28][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='1101001001110111111011'
[19:55:29][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='11010010011101111110'
[19:55:29][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='1101001001110111'
[19:55:29][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='110100100111011'
[19:55:30][D][binary_sensor:033]: 'living room pir': Sending state OFF
[19:55:34][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='11010010011101'
[19:55:34][D][binary_sensor:033]: 'living room pir': Sending state ON
[19:55:34][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='110100100111011'
[19:55:34][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='1101001001110111111'
[19:55:34][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='1101001001110111111'
[19:55:35][D][binary_sensor:033]: 'living room pir': Sending state OFF
[19:55:36][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='11010011010010011000'
[19:55:36][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='110100110100100110'
[19:55:36][D][binary_sensor:033]: 'kitchen pir': Sending state ON
[19:55:36][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='1101001101001001'
[19:55:36][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='1101001101000000110001110'
[19:55:36][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='110100110'
[19:55:36][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='1101001101001001'
[19:55:36][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='110100110100100'
[19:55:36][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=2 data='110100110100100110001110'
[19:55:37][D][binary_sensor:033]: 'kitchen pir': Sending state OFF
[19:55:44][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='11010011010010011001'
[19:55:44][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='11010011010010'
[19:55:44][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='11010011010010011000111'
[19:55:44][D][binary_sensor:033]: 'kitchen pir': Sending state ON
[19:55:44][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='110100110100'
[19:55:44][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='11010011010010011000111'
[19:55:44][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='1101001101001'
[19:55:44][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='1100000011'
[19:55:45][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='110000001111000100001'
[19:55:45][D][binary_sensor:033]: 'bathroom': Sending state ON
[19:55:45][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=4 data='110000011111000100001010'
[19:55:45][D][binary_sensor:033]: 'kitchen pir': Sending state OFF
[19:55:46][D][binary_sensor:033]: 'bathroom': Sending state OFF
[19:55:51][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='110000001111000100001110'
[19:55:51][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=2 data='110000001111000100001110'
[19:55:51][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='11000001'
[19:55:51][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=2 data='11000100'
[19:55:51][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='110000001111000100001110'
[19:55:51][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='110000001111000100001'
[19:55:51][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='110000001111000100001110'
[19:55:54][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='11010011010011'
[19:55:54][D][binary_sensor:033]: 'kitchen pir': Sending state ON
[19:55:54][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='1101001101001001'
[19:55:54][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='110100110'
[19:55:54][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='110100110'
[19:55:54][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='11010011'
[19:55:55][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='01101001101001001100011100010000'
[19:55:56][D][binary_sensor:033]: 'kitchen pir': Sending state OFF
[19:56:00][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=2 data='1101001101001'
[19:56:00][D][binary_sensor:033]: 'kitchen pir': Sending state ON
[19:56:00][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='110100110100100'
[19:56:00][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='11010011010010011000'
[... cut ...]
[19:56:44][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='01101001001110111111011101101001'
[19:56:44][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='01101001001110111111011101101001'
[19:56:45][D][binary_sensor:033]: 'living room pir': Sending state OFF
[19:56:50][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=2 data='110100100111011111101110'
[19:56:50][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='11010010011101'
[19:56:50][D][binary_sensor:033]: 'living room pir': Sending state ON

PaulN33, 2019-06-20:
that's the latest test,with all 5 sensors triggered, however i'm no longer sure the number of binary sensors is relevant, i have since witnessed quick performance with five sensors declared? i'm certainly seeing very variable performance though, which seems to stay consistent until i upload different code? for reference i have used the same hardware setup previously with the rc_switch demo to output sensor codes to node red and not seen this fluctuation in performance happen

OttoWinter, 2019-06-22:
Yes, that would be a different issue. I'm pretty sure there should not be any difference in speed due to the amount of binary sensors you have defined.

However, some delay can exist due to several reasons - either PIR remote is not sending stuff that fast or you need to decrease the idle timeout in your remote receiver config.

OttoWinter, 2019-07-12:
Closing because it's not the rcswitch as initially reported - probably the remote peripheral on the esp32 sometimes just takes some time (due to idle setting and spurious pulses) to push data.

closed 2019-07-12 by OttoWinter