esphome/issues #474: [BUG] I2C Stanza does not iterate over multiple pin values in constructor/ no support for multiple I2C buses

aenertia, 2019-06-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Fedora30 ; pip install 

**ESP (ESP32 Board/m5stick-c):**

Full pinout and arduino library for the platform available here: 

https://github.com/m5stack/M5StickC

**Affected component:**
i2c / playbook parser

**Description of problem:**
Currently the YAML playbook and parser does not support multiple I2C buses.

This is problematic for platforms that make extensive use of the multiple i2c buses in their design.

Currently am testing with the m5stickC https://github.com/m5stack/M5StickC

Which uses 3 different i2c buses one for grove, one for IMU and PMU and RTC and one for MEMs mic.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: m5stickcir
  platform: ESP32
  board: m5stack-core-esp32

wifi:
  ssid: "kainga-atawhai"
  password: ""

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

output:
  - platform: ledc
    pin: GPIO10
    id: redled
light:
  - platform: monochromatic
    output: redled
    name: "Red Inbuilt LED"

i2c:
  sda:  0
  scl:  34
  scan: True

time:
  - platform: sntp
    id: sntp_time
    timezone: Pacific/Auckland
    servers: 172.16.254.254

web_server:
  port: 80

sensor:
  - platform: bme280
    temperature:
      name: "BME280 Temperature"
      oversampling: 16x
    pressure:
      name: "BME280 Pressure"
    humidity:
      name: "BME280 Humidity"
    address: 0x76
    update_interval: 60s
~                                     

```

**Logs (if applicable):**

```

i2c bus scl/sda 21/22
[19:35:35][I][i2c:033]: Scanning i2c bus for active devices...
[19:35:35][I][i2c:040]: Found i2c device at address 0x34
[19:35:35][I][i2c:040]: Found i2c device at address 0x51
[19:35:36][I][i2c:040]: Found i2c device at address 0x6C

i2c bus scl/sda 32/33

[19:39:18][C][i2c:028]: I2C Bus:
[19:39:18][C][i2c:029]:   SDA Pin: GPIO33
[19:39:18][C][i2c:030]:   SCL Pin: GPIO32
[19:39:18][C][i2c:031]:   Frequency: 50000 Hz
[19:39:18][I][i2c:033]: Scanning i2c bus for active devices...
[19:39:18][I][i2c:040]: Found i2c device at address 0x76

---
note : delcaring multiple values in the playbook for the i2c constructor fails:

[aenertia@kiorewha esphome]$ esphome m5stickc.yaml run
INFO Reading configuration...
WARNING ESP32: Pin 10 (6-11) might already be used by the flash interface. Be warned.
WARNING ESP32: Pin 10 (6-11) might already be used by the flash interface. Be warned.
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/bin/esphome", line 10, in <module>
    sys.exit(main())
  File "/usr/lib/python2.7/site-packages/esphome/__main__.py", line 483, in main
    return run_esphome(sys.argv)
  File "/usr/lib/python2.7/site-packages/esphome/__main__.py", line 466, in run_esphome
    config = read_config(args.verbose)
  File "/usr/lib/python2.7/site-packages/esphome/config.py", line 803, in read_config
    res = load_config()
  File "/usr/lib/python2.7/site-packages/esphome/config.py", line 663, in load_config
    return _load_config()
  File "/usr/lib/python2.7/site-packages/esphome/config.py", line 651, in _load_config
    result = validate_config(config)
  File "/usr/lib/python2.7/site-packages/esphome/config.py", line 569, in validate_config
    validated = comp.config_schema(conf)
  File "/usr/lib/python2.7/site-packages/esphome/voluptuous_schema.py", line 32, in __call__
    res = super(_Schema, self).__call__(data)
  File "/usr/lib/python2.7/site-packages/voluptuous/schema_builder.py", line 267, in __call__
    return self._compiled([], data)
  File "/usr/lib/python2.7/site-packages/voluptuous/schema_builder.py", line 589, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/usr/lib/python2.7/site-packages/esphome/voluptuous_schema.py", line 131, in validate_mapping
    cval = cvalue(key_path, value)
  File "/usr/lib/python2.7/site-packages/voluptuous/schema_builder.py", line 811, in validate_callable
    return schema(data)
  File "/usr/lib/python2.7/site-packages/esphome/pins.py", line 289, in input_pin
    return validate_gpio_pin(value)
  File "/usr/lib/python2.7/site-packages/esphome/pins.py", line 267, in validate_gpio_pin
    value = _translate_pin(value)
  File "/usr/lib/python2.7/site-packages/esphome/pins.py", line 258, in _translate_pin
    return int(value)
TypeError: int() argument must be a string or a number, not 'list'

```

**Additional information and things you've tried:**

Note that the newer arduino core/platfromio espressif have support for the m5stick-c board definition ; using the older m5stack-core-esp32 one ... needs another bug.

OttoWinter, 2019-06-22:
Already, possible, you're just using the wrong syntax.

```yaml
i2c:
- scl: GPIO21
  sda: GPIO22
  id: bus0
- scl: GPIO32
  sda: GPIO33
  id: bus1

sensor:
  - platform: bme280
    # ...
    address: 0x76
    i2c_id: bus0  # or bus1
```

aenertia, 2019-06-22:
Thanks will give it a go.

Also support for sistronics displays (st7735) would be nice. 

The m5stickc has a weird gryo MMU sh200q and PMU AXP192 and a MEMS mic.

None of these should be hard to implement as components as example code is provided.

OttoWinter, 2019-06-23:
Those all go in the feature requests tracker, not here.

I will close the issue as multiple i2c buses do work.

closed 2019-06-23 by OttoWinter