esphome/issues #476: ESP32dev Board - Compile fails with Error in AsyncTCP / AsyncMqttClient

Markus-, 2019-06-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

I installed esphome fresh out of pip @ MacOS @ python 2.7

```
(esphome) mbair-m:esphome markus$ pip freeze
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
backports-abc==0.5
bottle==0.12.16
certifi==2019.3.9
chardet==3.0.4
click==5.1
colorama==0.4.1
colorlog==4.0.2
ecdsa==0.13
esphome==1.13.6
esphomeyaml==1.10.1
esptool==2.6
futures==3.2.0
idna==2.8
ifaddr==0.1.6
ipaddress==1.0.22
paho-mqtt==1.4.0
platformio==3.6.7
protobuf==3.7.1
pyaes==1.6.1
pyserial==3.4
pytz==2019.1
PyYAML==5.1
requests==2.21.0
semantic-version==2.6.0
singledispatch==3.4.0.3
six==1.12.0
tornado==5.1.1
typing==3.6.6
tzlocal==1.5.1
urllib3==1.24.1
voluptuous==0.11.5
```

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32dev - But compiling should be successful without any board attached...
**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:** MINIMAL-EXAMPLE
```yaml
esphome:
  name: esp32
  platform: ESP32
  board: esp32dev

wifi:
  ssid: "TEST"
  password: "TESTTEST"
  fast_connect: true

# Enable logging
logger:

mqtt:
  broker: 10.10.10.10

esp32_ble_tracker:
  scan_interval: 20s
```

**Logs (if applicable):**

```
esphome heizung_minimal.yaml compile
INFO Reading configuration...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
INFO Running:  platformio run -d esp32
Processing esp32 (platform: espressif32@1.6.0; framework: arduino; board: esp32dev)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
LibraryManager: Installing id=3837 @ 5.13.3
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
LibraryManager: Installing id=346 @ 0.8.2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
LibraryManager: Installing id=305 @ ^1.1.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
LibraryManager: Installing id=1826 @ ^1.0.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <AsyncMqttClient> 0.8.2
|   |-- <AsyncTCP> 1.1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Preferences> 1.0
|-- <WiFi> 1.0
Compiling .pioenvs/esp32/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp.o
Compiling .pioenvs/esp32/src/esphome/components/json/json_util.cpp.o
Compiling .pioenvs/esp32/src/esphome/components/logger/logger.cpp.o
Compiling .pioenvs/esp32/src/esphome/components/mqtt/custom_mqtt_device.cpp.o
Compiling .pioenvs/esp32/src/esphome/components/mqtt/mqtt_binary_sensor.cpp.o
Compiling .pioenvs/esp32/src/esphome/components/mqtt/mqtt_client.cpp.o
Compiling .pioenvs/esp32/src/esphome/components/mqtt/mqtt_climate.cpp.o
Compiling .pioenvs/esp32/src/esphome/components/mqtt/mqtt_component.cpp.o
Compiling .pioenvs/esp32/src/esphome/components/mqtt/mqtt_cover.cpp.o
Compiling .pioenvs/esp32/src/esphome/components/mqtt/mqtt_fan.cpp.o
Compiling .pioenvs/esp32/src/esphome/components/mqtt/mqtt_light.cpp.o
Compiling .pioenvs/esp32/src/esphome/components/mqtt/mqtt_sensor.cpp.o
Compiling .pioenvs/esp32/src/esphome/components/mqtt/mqtt_switch.cpp.o
Compiling .pioenvs/esp32/src/esphome/components/mqtt/mqtt_text_sensor.cpp.o
Compiling .pioenvs/esp32/src/esphome/components/wifi/wifi_component.cpp.o
Compiling .pioenvs/esp32/src/esphome/components/wifi/wifi_component_esp32.cpp.o
Compiling .pioenvs/esp32/src/esphome/components/wifi/wifi_component_esp8266.cpp.o
Compiling .pioenvs/esp32/src/esphome/components/xiaomi_ble/xiaomi_ble.cpp.o
Compiling .pioenvs/esp32/src/esphome/core/application.cpp.o
Compiling .pioenvs/esp32/src/esphome/core/component.cpp.o
Compiling .pioenvs/esp32/src/esphome/core/controller.cpp.o
Compiling .pioenvs/esp32/src/esphome/core/esphal.cpp.o
Compiling .pioenvs/esp32/src/esphome/core/helpers.cpp.o
Compiling .pioenvs/esp32/src/esphome/core/log.cpp.o
Compiling .pioenvs/esp32/src/esphome/core/preferences.cpp.o
Compiling .pioenvs/esp32/src/esphome/core/util.cpp.o
Compiling .pioenvs/esp32/src/main.cpp.o
Generating partitions .pioenvs/esp32/partitions.bin
Archiving .pioenvs/esp32/lib933/libArduinoJson-esphomelib_ID3837.a
Indexing .pioenvs/esp32/lib933/libArduinoJson-esphomelib_ID3837.a
Compiling .pioenvs/esp32/lib1b3/AsyncTCP_ID1826/AsyncTCP.cpp.o
Compiling .pioenvs/esp32/lib9b2/AsyncMqttClient_ID346/AsyncMqttClient.cpp.o
Compiling .pioenvs/esp32/lib9b2/AsyncMqttClient_ID346/AsyncMqttClient/Packets/ConnAckPacket.cpp.o
.piolibdeps/AsyncTCP_ID1826/src/AsyncTCP.cpp: In function 'bool _start_async_task()':
.piolibdeps/AsyncTCP_ID1826/src/AsyncTCP.cpp:205:141: error: 'xTaskCreateUniversal' was not declared in this scope
xTaskCreateUniversal(_async_service_task, "async_tcp", 8192 * 2, NULL, 3, &_async_service_task_handle, CONFIG_ASYNC_TCP_RUNNING_CORE);
^
*** [.pioenvs/esp32/lib1b3/AsyncTCP_ID1826/AsyncTCP.cpp.o] Error 1
.piolibdeps/AsyncMqttClient_ID346/src/AsyncMqttClient.cpp: In constructor 'AsyncMqttClient::AsyncMqttClient()':
.piolibdeps/AsyncMqttClient_ID346/src/AsyncMqttClient.cpp:40:61: warning: format '%x' expects argument of type 'unsigned int', but argument 3 has type 'uint64_t {aka long long unsigned int}' [-Wformat=]
sprintf(_generatedClientId, "esp32%06x", ESP.getEfuseMac());
^
.piolibdeps/AsyncMqttClient_ID346/src/AsyncMqttClient.cpp:40:61: warning: format '%x' expects argument of type 'unsigned int', but argument 3 has type 'uint64_t {aka long long unsigned int}' [-Wformat=]
============================================================================ [ERROR] Took 15.57 seconds ============================================================================

```

**Additional information and things you've tried:**

I just installed esphome with pip and put this example yaml into place. A simple compile doesn't work. Is there any issue?

OttoWinter, 2019-06-23:
Ugh, looks like AsyncTCP once again released an incompatible version without a proper major version bump.

Markus-, 2019-06-24:
So is there anything I can do until a fix?

Regards
Markus

sideeffectdk, 2019-06-24:
I've "fixed" it by specifying the previous version of AsyncTCP in platformio.ini in the folder for that project.
The version that works is: 1.0.3
The version that does not is: 1.1.0
The stuff after AsyncTCP specifies to use latest version before 1.1.0.

Like this:
```

; Auto generated code by esphome

[common]
lib_deps =
    AsyncTCP@<1.1.0
build_flags =
upload_flags =
```

> So is there anything I can do until a fix?
> 
> Regards
> Markus

Markus-, 2019-06-24:
Your entry is not recognized

`; Auto generated code by esphome

[common]
lib_deps =
    AsyncTCP@<1.1.0
build_flags =
upload_flags =

; ===== DO NOT EDIT ANYTHING BELOW THIS LINE =====
; ========== AUTO GENERATED CODE BEGIN ===========
[env:esp32]
board = esp32dev
board_build.partitions = partitions.csv
build_flags =
    -DESPHOME_LOG_LEVEL=ESPHOME_LOG_LEVEL_DEBUG
    -Wno-sign-compare
    -Wno-unused-but-set-variable
    -Wno-unused-variable
    -fno-exceptions
    ${common.build_flags}
framework = arduino
lib_deps =
    ArduinoJson-esphomelib@5.13.3                                                                                                                                                    
    AsyncMqttClient@0.8.2
    ESPmDNS
    Preferences
    ${common.lib_deps}
platform = espressif32@1.6.0
upload_speed = 115200
; =========== AUTO GENERATED CODE END ============
; ========= YOU CAN EDIT AFTER THIS LINE =========`

I think AsyncMqttClient forces the 1.1.0-version?
`Scanning dependencies...
Dependency Graph
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <AsyncMqttClient> 0.8.2
|   |-- <AsyncTCP> 1.1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Preferences> 1.0
|-- <AsyncTCP> 1.0.3
|-- <WiFi> 1.0
`

sideeffectdk, 2019-06-24:
Oh, also delete the new versions folder, so the only folder is the one with@1.0.3 at the end
Sorry I didn't say that in the first post

Markus-, 2019-06-24:
Strange. 

If I delete the whole .pio...-Folder it doesn't work (as above).
If I delete the subfolder (rm -rf esp32/.piolibdeps/AsyncTCP_ID1826) and just keep the 1.0.3 Version it compiles...

Ok. Thanks for the quick fix.

Regards
Markus

AlexDanault, 2019-06-27:
> Yeah, how do you make sure dependencies dependencies use the correct version 😄

Anybody has a working lib_deps config for those who need mqtt ? No combination of AsyncMqttClient worked for me so either I'm doing something wrong or I'm stuck waiting for a release with fixed dependencies.

Markus-, 2019-06-27:
See my "Strange"-Comment...

```
; Auto generated code by esphome

[common]
lib_deps =
    AsyncTCP@<1.1.0
build_flags =
upload_flags =

; ===== DO NOT EDIT ANYTHING BELOW THIS LINE =====
; ========== AUTO GENERATED CODE BEGIN ===========
[env:esp32og]
board = esp32dev
board_build.partitions = partitions.csv
build_flags =
    -DESPHOME_LOG_LEVEL=ESPHOME_LOG_LEVEL_DEBUG
    -Wno-sign-compare
    -Wno-unused-but-set-variable
    -Wno-unused-variable
    -fno-exceptions
    ${common.build_flags}
framework = arduino
lib_deps =
    ArduinoJson-esphomelib@5.13.3
    AsyncMqttClient@0.8.2
    ESPmDNS
    Hash
    Preferences
    ${common.lib_deps}
platform = espressif32@1.6.0
upload_speed = 115200
```

This is a working copy, BUT:

1.  - compile it --> it will fail
2. - make sure in the Version 1.0.3 is also downloaded
3. - go to .piolibdevs and delete AsyncTCP_ID1826 (without @1.0.3)
4. - compile again -> this should work

Regards
Markus

rdehuyss, 2019-06-27:
Also having trouble with this ...

AlexDanault, 2019-06-27:
Thanks @Markus- I can confirm your workaround works !

closed 2019-06-28 by OttoWinter

pfeerick, 2019-06-30:
I'm guessing upgrading to `platform = espressif32@1.8.0` (or 1.9.0) is not an alternative (yet?)... as  AsyncTCP 1.1.0 appears to be using features introduced in arduino-esp32-1.0.2 which was supported in platformio-espress32-1.8.0, and 1.9.0 improved espota support.

MajicCat, 2019-07-05:
Same issue.
Is there a working revision combination of espressif and Arduino that is valid (old or current)?

brandond, 2019-07-05:
@MajicCat this has been fixed in dev. You can use that branch, or wait for the next version to be released.

MajicCat, 2019-07-05:
"me-no-dev/ESPAsyncWebServer" does not have a 'dev' branch. Which are you referencing?

brandond, 2019-07-05:
The dev branch of this software, esphome. See up above where it says "OttoWinter closed this in esphome/esphome#653 7 days ago".

fah, 2019-07-17:
I use ESPhome as an addon in hassio. How can I delete the above mentioned file? I access my raspberry pi, where hassio lives on, with the ssh addon. 
But I can't find more then the project specific platformio.ini file inside "/config/esphome/...". 
For example "/data/projectname/.." does not exist.

fah, 2019-07-18:
Today I got an hassio update that fixed the AsyncTCP@1.0.3 issue.