esphome/issues #479: ESP32Cam Flash error "user code done" under EspHomeFlasher

Electronlibre2012, 2019-06-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Raspberry Pi 3B+ with hassio
File .img from HomeAssistant
EspHome 1.13.6
espHomeFlasher 1.0.1
ESP32CAM AI-thinker

**Affected component:**
https://esphome.io/

**Description of problem:**
After flashing in esphomeFlasher, stuck on "user code done" in log
So not connecting at all, OTA not possible

**Problem-relevant YAML-configuration entries:**
```yaml

esphome:
  name: terrasse
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "xxxxxxxxxxxxxxxxx"
  password: "xxxxxxxxxxxxxxxxxxxxxxxx"
  manual_ip:
    static_ip: 192.168.1.48
    gateway: 192.168.1.1
    subnet: 255.255.255.0

# Enable logging
logger: 
  level: VERBOSE

# Enable Home Assistant API
api:

ota:
  password: "xxxxxxxxxx"
  
# ESP32-CAM
esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  vertical_flip: false
  horizontal_mirror: false
  resolution: 640x480 

  name: terrasse
  # ... Frame Settings
  
sensor:
  - platform: wifi_signal
    name: "WiFi Signal Terrasse"
    update_interval: 60s
    id: signalWt

switch:
  - platform: restart
    name: "Restart Cam Terrasse"
  
binary_sensor:
  - platform: gpio
    pin: GPIO13
    name: "Terrasse PIR Sensor"
    device_class: motion

# Flashlight
output:
  - platform: gpio
    pin: GPIO4
    id: gpio_4
light:
  - platform: binary
    output: gpio_4
    name: $hostname lights
```

**Logs (if applicable):**

```
Connecting....
Detecting chip type... ESP32
Connecting...

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32D0WDQ5 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 24:6F:28:DB:76:2C
Uploading stub...
Running stub...
Stub running...
 - Flash Size: 4MB
 - Flash Mode: dio
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 10.0s
Compressed 15088 bytes to 9755...
Wrote 15088 bytes (9755 compressed) at 0x00001000 in 0.9 seconds (effective 141.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 143...
Wrote 3072 bytes (143 compressed) at 0x00008000 in 0.0 seconds (effective 786.5 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 4193.3 kbit/s)...
Hash of data verified.
Compressed 944448 bytes to 507559...
Wrote 944448 bytes (507559 compressed) at 0x00010000 in 45.0 seconds (effective 167.7 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Done! Flashing is complete!

Showing logs:
[18:22:28]ets Jun  8 2016 00:22:57
[18:22:28]
[18:22:28]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[18:22:28]configsip: 0, SPIWP:0xee
[18:22:28]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[18:22:28]mode:DIO, clock div:2
[18:22:28]load:0x3fff0018,len:4
[18:22:28]load:0x3fff001c,len:952
[18:22:28]load:0x40078000,len:6084
[18:22:28]load:0x40080000,len:7944
[18:22:28]entry 0x40080310
[18:22:28]user code done

```

**Additional information and things you've tried:**

Flash with flash_download_tools_v3.6.6_0, same problem, seem to be the binary file on EspHome that is not good

Electronlibre2012, 2019-06-24:
After 3 hours...reinstall Hassio with the original image on HomeAssistant web site, just install ESPHOME and ...same user code none after my first flash of a nex esp32-cam who works with a simple skecth arduino...

Is someone can help please? nobody have a solution?

 @OttoWinter please have you a solution, i m stuck...think a fresh install could help...

OttoWinter, 2019-06-24:
Please say specifically how you flashed it - did you have to press down any buttons during the flashing process? Add as much detail as possible. If the board has a reset button, press that after it has finished flashing.

Also try flashing an image without esp32_camera enabled.

Electronlibre2012, 2019-06-24:
thanks for your reponse :

1-open EspHomeflasher
2-connect the esp32-cam ai thinker : put the Gpio0 connect with gnd / put +5v and -gnd
3-flash. after it stay on "showing log"
4-i remove the +5v. remove the shunt between gpio0 and gnd
5-Put back the +5V
6-display :
```
[00:55:54]
[00:55:54]
[00:55:54]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[00:55:54]configsip: 0, SPIWP:0xee
[00:55:54]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[00:55:54]mode:DIO, clock div:2
[00:55:54]load:0x3fff0018,len:4
[00:55:54]load:0x3fff001c,len:952
[00:55:54]load:0x40078000,len:6084
[00:55:54]load:0x40080000,len:7944
[00:55:54]entry 0x40080310
[00:55:54]user code done

```

i try with a cp2102 now, same.
before i try with a FT232Bl, same

i do like that before. Or i press reset after the flash and have removed the shunt, but same result now.

Any idea?

Electronlibre2012, 2019-06-24:
i found the trick : it work with QIO on Arduino IDE. If i put DIO it reboot all the time in loop.

In esphomeflasher i see DIO mode when i flash it...can i change this? @OttoWinter please?

Electronlibre2012, 2019-06-24:
Is it possible for you to create a new ESP32 in EspHome like "ESP32-Cam AiThinker" named and have this config for flashing :
[image]

that the logs ok in IDE :

```
03:16:55.296 -> ets Jun  8 2016 00:22:57
03:16:55.296 -> 
03:16:55.296 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
03:16:55.296 -> configsip: 0, SPIWP:0xee
03:16:55.296 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
03:16:55.296 -> mode:DIO, clock div:2
03:16:55.296 -> load:0x3fff0018,len:4
03:16:55.296 -> load:0x3fff001c,len:1100
03:16:55.296 -> load:0x40078000,len:9232
03:16:55.296 -> load:0x40080400,len:6412
03:16:55.296 -> entry 0x400806a8
03:16:56.954 -> Booting
03:16:58.859 -> Ready
03:16:58.859 -> IP address: 192.168.1.48
```
I dont know why only QIO works but as you see, the chip choose DIO for boot...

I really understand nothing and nobody help...

Seem the there is a new version of esp32Cam ai Thinker and a lot of peaople will have this issue soon...

If you can make something for us please?

Electronlibre2012, 2019-06-25:
Hello @OttoWinter ,

i finally succeed : the only way is to put the USB-to-TTL adapter on the Raspberry USB /dev/ttyUSB0...

there is 2 files install before compile...dont know if the problem is that, here is the log :

```
========================= [SUCCESS] Took 85.43 seconds =========================
INFO Successfully compiled program.
INFO Running:  platformio run -d /config/esphome/terrasse -t upload --upload-port /dev/ttyUSB0
Processing terrasse (platform: espressif32@1.6.0; framework: arduino; board: esp-wrover-kit)
--------------------------------------------------------------------------------
PackageManager: Installing tool-mkspiffs @ ~2.230.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
PackageManager: Installing tool-espotapy @ ~1.1.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp-wrover-kit.html
PLATFORM: Espressif 32 > Espressif ESP-WROVER-KIT
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(ftdi) ON-BOARD(ftdi) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AsyncTCP> 1.0.3
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Preferences> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Retrieving maximum program size /data/terrasse/.pioenvs/terrasse/firmware.elf
Checking size /data/terrasse/.pioenvs/terrasse/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=         ]  12.2% (used 40004 bytes from 327680 bytes)
PROGRAM: [=====     ]  50.6% (used 829378 bytes from 1638400 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, esptool, ftdi, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: /dev/ttyUSB0
Uploading /data/terrasse/.pioenvs/terrasse/firmware.bin
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 24:0a:c4:bb:1e:10
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 16176 bytes to 10658...
Wrote 16176 bytes (10658 compressed) at 0x00001000 in 1.0 seconds (effective 132.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 143...
Wrote 3072 bytes (143 compressed) at 0x00008000 in 0.0 seconds (effective 785.5 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 3908.8 kbit/s)...
Hash of data verified.
Compressed 829488 bytes to 461562...
Wrote 829488 bytes (461562 compressed) at 0x00010000 in 41.9 seconds (effective 158.5 kbit/s)...
Hash of data verified.

Leaving...
[... 59 lines cut ...]
[10:02:45][C][wifi:255]:   IP Address: 192.168.1.48
[10:02:45][C][wifi:257]:   BSSID: 

[redacted]
[10:02:45][C][wifi:258]:   Hostname: 'terrasse'
[10:02:45][C][wifi:262]:   Signal strength: -18 dB ▂▄▆█
[10:02:45][C][wifi:263]:   Channel: 6
[10:02:45][C][wifi:264]:   Subnet: 255.255.255.0
[10:02:45][C][wifi:265]:   Gateway: 192.168.1.1
[10:02:45][C][wifi:266]:   DNS1: 192.168.1.1
[10:02:45][C][wifi:267]:   DNS2: 0.0.0.0
[10:02:45][C][logger:137]: Logger:
[10:02:45][C][logger:138]:   Level: DEBUG
[10:02:45][C][logger:139]:   Log Baud Rate: 115200
[10:02:45][C][logger:140]:   Hardware UART: UART0
[10:02:45][C][ota:029]: Over-The-Air Updates:
[10:02:45][C][ota:030]:   Address: terrasse.local:3232
[10:02:45][C][ota:032]:   Using Password.
[10:02:45][C][api:103]: API Server:
[10:02:45][C][api:104]:   Address: terrasse.local:6053
[10:02:52][D][api:075]: Disconnecting aioesphomeapi (192.168.1.96)

```

An idea why it doesnt work with EspHomeFlasher app on my PC, like before?

OttoWinter, 2019-06-25:
That is interesting - esphomeflasher and the esphome addon use the same code for flashing, so there should be no difference (unless something else changed in your setup in the meantime).

You can select a different flash mode with:
```yaml
esphome:
  # ...
  board_flash_mode: qio
```
but I doubt that's the issue.

Electronlibre2012, 2019-06-25:
I have already try it this morning 👍  : reboot in loop.

I ve try everything as possible since yesterday lol

I desinstalled/reinstall proprer python3.7 (uninstall with .msi, deleted folder)  of my Windows, same. I have installed the python gui esphomeflasher, same...

Dont understand. But it works with direct usb pi connexion...i think i have loose enought time...

Electronlibre2012, 2019-06-25:
it works with an esp8266 ...

Its direct connection, there is a micro USB (Wemos D1 Pro) so its not the same case, i thought there is a problem with the BL232RF i use to flash the esp on my window system because it work on direct USB Pi...

pcmester, 2019-06-30:
If you set the board flashing mode to dout, it will work. It worked for me.

Electronlibre2012, 2019-06-30:
hi @pcmester 
it doent work for me...

```
Showing logs:
[17:36:59]ets Jun  8 2016 00:22:57
[17:36:59]
[17:36:59]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[17:36:59]configsip: 0, SPIWP:0xee
[17:36:59]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[17:36:59]mode:DOUT, clock div:2
[17:36:59]load:0x3fff0018,len:4
[17:36:59]load:0x3fff001c,len:808
[17:36:59]load:0x40078000,len:6084
[17:36:59]load:0x40080000,len:6704
[17:36:59]entry 0x400802e4
```
there is no more "user code done" but it stuck after "entry 0x400802e4"...

MPJK77, 2019-06-30:
I had the same issue with two AI-Thinker esp32 cams. 
Flashing thru the USB port of the Pi directly solved my problems.

newbee112, 2019-07-25:
I have identical issue as described. Since i run on VM the workaround on PI install is not valid for me. It seems to be something with the new board type as stated above as well, all other ESP32's are working just fine.

andremiller, 2019-07-29:
Same issue here, flashing a sample program with Arduino IDE on windows works fine, but compiling a firmware withESPHome on RaspberryPi and then flashing it with esphome-flasher on same Windows machine as the Arduino IDE results in the following log output after a flash:

Leaving...
Hard Resetting...
Done! Flashing is complete!

Showing logs:
[17:41:09]ets Jun  8 2016 00:22:57
[17:41:09]
[17:41:09]rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
[17:41:09]waiting for download
[17:41:13]ets Jun  8 2016 00:22:57
[17:41:13]
[17:41:13]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[17:41:13]configsip: 0, SPIWP:0xee
[17:41:13]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[17:41:13]mode:DIO, clock div:2
[17:41:13]load:0x3fff0018,len:4
[17:41:13]load:0x3fff001c,len:952
[17:41:13]load:0x40078000,len:6084
[17:41:13]load:0x40080000,len:7944
[17:41:13]entry 0x40080310
[17:41:13]user code done

newbee112, 2019-08-03:
So what can be done about this? there are many running esphome on home assistant that is not running on Pi.

Electronlibre2012, 2019-08-11:
> 
> 
> So what can be done about this? there are many running esphome on home assistant that is not running on Pi.

Hello,

i have received 4 news esp32-cam from Bangood and now it works without problem with ESPFlasher...nothing to understand...

newbee112, 2019-08-12:
Still can not get it to work running on PC, i might just boot up a PI with ESPhome and flash from there. but i find it strange that for some reason we can not flash via windows (all other ESP's are just fine esp32, 8266, esp1) but these esp32 cam have the issue..

newbee112, 2019-08-12:
i can confirm that  when on fresh raspbian + esphome install on Pi3 it works fine to flash. but something is defiantly not right with this board under windows. as said other boards flash just fine.

oetelaar24, 2019-08-14:
Ok, I've also ran into this issue, but I have also finally found a feasible workaround for now. Since I also use a hassOs VM, the RPi-workaround isn't for me.

However, doing:
1. Follow step one of this tutorial, installing Arduino IDE and compiling/uploading the Wifiscan sketch: https://randomnerdtutorials.com/esp32-cam-video-streaming-web-server-camera-home-assistant/ (step 1 links to either Windows or MacOS tutorial of installing Arduino studio and testing it with a Wifiscan sketch in the first half, that is enough for us, just to get the right partitions and bootcode)
2. Compile and download the ESPHome binary from the interface.
3. Use esptool.py to flash the ESPHome binary on 0x10000: esptool.py --port /dev/tty.usbserial-A50285BI --baud 115200 --chip esp32 write_flash --flash_size 4MB -fm dio  0x10000 /path/to/esphomefirmware.bin
4. Disconnect the IO0/GND bridge and reboot.

I guess it has to do with the bootcode that the ESPHomeFlasher puts on the ESP32-CAM module, maybe this is an updated version, maybe the partitionlayout is a bit different.

Unfortunately, OTA doesn't work when you do it like this, but atleast the module/camera works :-D

Electronlibre2012, 2019-08-14:
Hello,

Thanks @oetelaar24 ,

another thing strange : i have the ```[Esp32 camera:091]: Setup Failed: ERROR``` with 4 new board received from bangood. It works to flash direct on PC with EspHomeFlasher (like the firsts boards i buy from Bangood month ago) but : no image is displayed in Hassio and got this error.

So i put the old camera lens instead of the one delivered with the "new" ESP32-cam board, and it works! no more Setup failed error, and camera stream is ok under Hassio...

[](https://github.com/esphome/issues/issues/435)

I understand nothing...

oetelaar24, 2019-08-14:
Sounds more like the new boards came with a new/different type of camera with either a different pinout or clock.
Fiddling with the setting:
external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32

could help... though, you'd need some documentation or otherwise it's gonna be hella difficult..

Electronlibre2012, 2019-08-14:
Yes same settings @oetelaar24 works with the old camera lens (previous ones from Bangood).

You are right! i just look and its not at all the same pinout and reference on the previous and the new model of camera lens...
Previous is AF2569/0927XA
New is XRZ001-V240 / V2.0 1903

i wil ask bangood for documentation...lol

Annaka007, 2019-08-22:
I had exactly the same problem when trying to upload code onto AI Thinker ESP32 Cam.

I logged into HASSOS, and then started ESPHome on my Intel NUC Win10, uploaded via USB, and the module just sits there and hangs at the ......user code done point. I tried uploading the Arduino web server program from the Arduino IDE and the camera worked fine but it just wont work via ESP Home on my NUC running Win10.

I then set up a Raspberry Pi with HASSOS and installed ESPHome, then uploaded the AIThinker code via the Raspberry Pi and everything works fine. Seems to be a problem with the NUC win10 combination.

myhrmans, 2019-09-06:
Anyone figured this out?

amordicus, 2019-10-25:
> Yes same settings @oetelaar24 works with the old camera lens (previous ones from Bangood).
> 
> You are right! i just look and its not at all the same pinout and reference on the previous and the new model of camera lens...
> Previous is AF256B/0927XA
> New is XRZ001-V240 / V2.0 1903
> 
> i wil ask bangood for documentation...lol

Have you find out the pinout I have almost the same OV2640, mine is XRZ00D1-V240 / V2.01903?  Or maybe the problem was the pinout of the ESP32S that is not the AI-thinker , or is it?

imShara, 2019-11-14:
Here is man who gets it to work, but only with 5v. https://github.com/Electronlibre2012/ESP-CAM-Ai-Thinker-Remote-Control-Tilt-and-Yaw/issues/2#issuecomment-526942984

But what this camera is? Does it built with real ov2640 sensor or XRZ001-V24 just chineese fake?

Electronlibre2012, 2019-11-14:
> Here is man who gets it to work, but only with 5v. Electronlibre2012/ESP-CAM-Ai-Thinker-Remote-Control-Tilt-and-Yaw#2
> (comment)
>
> But what this camera is? Does it built with real ov2640 sensor or
> XRZ001-V24 just chineese fake?

I dont know exactly...sorry

For me i just flash the esp32-cam on the raspberry directly with a USB/ttl
adapter, i dont find any other solution when it doesnt work with ESPHome
flasher tool on my PC...

Not a big deal, it works, that the most important, and just need 1 time,
after the OTA work great.

imShara, 2019-11-19:
> Sounds more like the new boards came with a new/different type of camera with either a different pinout or clock.
> Fiddling with the setting:
> external_clock:
> pin: GPIO0
> frequency: 20MHz
> i2c_pins: [...]

Just made working demo for ESP-CAM from Aliexpress with XRZ00D1-V240 camera module https://github.com/imShara/ESP32-CAM-XRZ00D1-V240-HTTP-AP

maccroyal, 2020-01-06:
Fix for me with new version of esphome-flasher.

closed 2020-08-07 by stale[bot]