esphome/issues #481: ESPhome zeroconf error

Wuppie007, 2019-06-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

arch | x86_64
-- | --
dev | false
docker | true
hassio | true
os_name | Linux
python_version | 3.7.3
timezone | Europe/Amsterdam
version | 0.94.4
virtualenv | false

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8966 and Sonoff

**Affected component:**

https://www.home-assistant.io/docs/mqtt/discovery/

**Description of problem:**
After fixing issues References: [#24187](https://github.com/home-assistant/home-assistant/issues/24187) and [#24578](https://github.com/home-assistant/home-assistant/issues/24578). The error changed, but still there is something wrong. Now a multiple of links are displayed on the integrations page ESPHome, but when it push "configure" a wait screen appear but nothing happend.

If added a traceback, what is available.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
Restart the esphome hassio addon

[image]

OttoWinter, 2019-06-24:
Yep, known. Try with the latest HA beta version - that has it fixed. But as the next HA release will come out very soon, there's not going to be a patch release for this small issue.

OttoWinter, 2019-07-12:
Fixed in recent HA versions

closed 2019-07-12 by OttoWinter