esphome/issues #483: WDT resets when using Waveshare e-paper

frog32, 2019-06-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio
**ESP (ESP32/ESP8266, Board/Sonoff):**

d1 mini and node MCU show the same behavior

**Affected component:**

waveshare e-paper

**Description of problem:**
I see WDT resets at the same interval that I use as update_interval which manifests in loosing connectivity to home assistant and the display cycling through inverted colors, black, white, black.

**Problem-relevant YAML-configuration entries:**
```yaml
font:
  - file: 'ComicNeue-Regular.ttf'
    id: comic
    size: 30

spi:
  clk_pin: D0
  mosi_pin: D1

display:
  - platform: waveshare_epaper
    cs_pin: D2
    dc_pin: D3
    busy_pin: D4
    reset_pin: D5
    model: 4.20in
    update_interval: 30s
    #full_update_every: 30
    lambda: |-
      it.printf(200, 0, id(comic), TextAlign::TOP_CENTER, "Test");
```

**Logs (if applicable):**

Very verbose and long https://pastebin.com/Lc2QJcVY

**Additional information and things you've tried:**

Changing the update interval changes the frequency at which the ESP restarts

OttoWinter, 2019-06-25:
If you're only seeing the WDT timeout with very verbose debugging enabled, that's expected.

The display just produces such amounts of data that will overflow logging - and very_verbose is expected to break things in some cases.

Please add if this also happens without very verbose debugging.

frog32, 2019-06-25:
Yes the same happens without the VERY_VERBOSE logging including falling off the network. If there is any other data you need or experiment I should run I would likely be able to work on that this evening EMEA hours.

frog32, 2019-06-25:
You are right. Once I set logging to DEBUG it started working again. Seems that I had some other issue in my config and while trying to debug I rabbitholed down the WDT reset issue.

closed 2019-06-25 by frog32