esphome/issues #484: UART Switch not sending CR / LF in string form

CountParadox, 2019-06-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io running on an Intel NUC using official image
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 - Wemos D1 Mini
**Affected component:**

https://esphome.io/components/switch/uart.html

**Description of problem:**
When sending a data string, including \r\n for carriage return and line feed, the characters "\r\n" are sent, rather than sending the actual CR and LF

For example, sending 

```
  - platform: uart
    name: "HDMI 1"
    data: 'INSP  10\r\n'
```

Results in the ESP sending
```
INSP  10\r\n
```
on a single line. 

As a result the UART command is not executed.

manually sending the hex as per the docs does seem to work:

```
  - platform: uart
    name: "HDMI 1"
    data: [0x49, 0x4E, 0x50, 0x53, 0x20, 0x20, 0x31, 0x30, 0x0D, 0x0A]
```

And correctly sends the CR and LF.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: sharptv
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: 'x'
  password: 'x'

api:

# Enable logging
logger:

ota:

mqtt:
  broker: 172.16.0.60
  username: rs232
  password: x

uart:
  baud_rate: 38400
  tx_pin: D4

switch:
  - platform: uart
    name: "Power On"
    data: [0x50, 0x4F, 0x57, 0x52, 0x20, 0x20, 0x20, 0x31, 0x0D, 0x0A]
    on_turn_on:
      then:
      - mqtt.publish:
          topic: esphome/rs232/sharp/state
          payload: "ON"

  - platform: uart
    name: "Power Off"
    data: [0x50, 0x4F, 0x57, 0x52, 0x20, 0x20, 0x20, 0x30, 0x0D, 0x0A]
    on_turn_on:
      then:
      - mqtt.publish:
          topic: esphome/rs232/sharp/state
          payload: "OFF"

  - platform: uart
    name: "HDMI 1"
    data: [0x49, 0x4E, 0x50, 0x53, 0x20, 0x20, 0x31, 0x30, 0x0D, 0x0A]
    on_turn_on:
      then:
      - mqtt.publish:
          topic: esphome/rs232/sharp/input
          payload: "HDMI1"

  - platform: uart
    name: "HDMI 2"
    data: [0x49, 0x4E, 0x50, 0x53, 0x20, 0x20, 0x31, 0x33, 0x0D, 0x0A]
    on_turn_on:
      then:
      - mqtt.publish:
          topic: esphome/rs232/sharp/input
          payload: "HDMI2"

  - platform: uart
    name: "HDMI 3"
    data: [0x49, 0x4E, 0x50, 0x53, 0x20, 0x20, 0x31, 0x38, 0x0D, 0x0A]
    on_turn_on:
      then:
      - mqtt.publish:
          topic: esphome/rs232/sharp/input
          payload: "HDMI3"
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

When using string, the output of the serial commands will be like so, assuming you turn on TV Power, and then HDMI2:

```
POWR   1\r\nINPS  13\r\n
```

where it SHOULD be like:

```
POWR   1
INPS  13
```

See community forums for some other tests done
https://community.home-assistant.io/t/uart-switch-help/123245/18

glmnet, 2019-06-25:
Interesting you’ve expected it to work. It doesn’t say on docs it should work. Or am I missing something?

CountParadox, 2019-06-25:
> Interesting you’ve expected it to work. It doesn’t say on docs it should work. Or am I missing something?

I've expected it as it seems logical to me, There is not many serial devices that will act on a command until receiving a CR or LF, so being able to send one of these, to me, would seem like a basic feature of a component designed to send serial commands...

brandond, 2019-06-26:
If you want to put escaped characters in your string you need to use double quotes, not single. This is per the YAML spec - it's not an issue with esphome:

https://yaml.org/spec/1.2/spec.html#id2776092
> Note that escape sequences are only interpreted in double-quoted scalars. In all other scalar styles, the “\” character has no special meaning and non-printable characters are not available.

OttoWinter, 2019-06-26:
Good catch @brandond ! I would not have thought of that, but it appears to be true.

An update to the docs would be welcome so that others do not encounter the same thing. I'll close the report as there's not really a bug.

closed 2019-06-26 by OttoWinter