esphome/issues #485: 'wait_time' configuration for remote transmitter component (timebase wrong)

fbelavenuto, 2019-06-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Raspbian - pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 generic

**Affected component:**

https://esphome.io/components/remote_transmitter.html

**Description of problem:**
The 'wait_time' configuration parameter is read as milliseconds in the code, but the wait platforms code uses the arduino's 'delay' function that expects to receive the time in microseconds.
For example, using a waiting time of 1 second causes a delay of only 1 millisecond.

**Problem-relevant YAML-configuration entries:**
```yaml
...
        - remote_transmitter.transmit_nec:
            address: 0x20DF
            command: 0xC03F
            repeat: 
              times: 10
              wait_time: 50ms
...

```

**Logs (if applicable):**

```

```

**Additional information and things you've tried:**
Analyzing the code I've corrected that part and attached a .diff file for the fix. Fix for ESP8266 and ESP32.

[remote_transmitter_send_wait.diff.txt](https://github.com/esphome/issues/files/3327682/remote_transmitter_send_wait.diff.txt)

glmnet, 2019-06-26:
Are you positive here?
Esphome supports ms and us as time units. That code uses delay to wait for whole milliseconds and then uses delayMicroseconds for the rest, makes sense. 
Please add a log statement there with the value of send_wait, it should be 50000 for your choice of 50ms, otherwise it is a bug but somewhere else.

fbelavenuto, 2019-06-26:
> Are you positive here?
No, I was developing my own solution and I discovered this great project only 2 weeks ago. I'm Brazilian, sorry for my bad english!
The code in python, including the dashboard, uses millisecond precision, does not accept microseconds, so the delay code is wrong, using 'delaymicroseconds()'. Another solution is to correct the precision of the fields to accept microsecond times and keep the delays as they are.

glmnet, 2019-06-26:
> Another solution is to correct the precision of the fields to accept microsecond times and keep the delays as they are.

Let's try to get that working
Could you please change [this line](https://github.com/esphome/esphome/blob/5cd7f23065d2925eeb507e5cbc2cdfc31e1078fd/esphome/components/remote_base/__init__.py#L91) to `positive_time_period_microseconds` and report back?

OttoWinter, 2019-06-26:
@glmnet Ah yes, true. the value is in microseconds not milliseconds, thanks!

glmnet, 2019-06-26:
I think you should look for RC Switch too

OttoWinter, 2019-06-26:
@glmnet Correct

closed 2019-06-26 by OttoWinter

fbelavenuto, 2019-06-26:
> Let's try to get that working
> Could you please change [this line](https://github.com/esphome/esphome/blob/5cd7f23065d2925eeb507e5cbc2cdfc31e1078fd/esphome/components/remote_base/__init__.py#L91) to [...]

Works perfectly!! Very thanks!