esphome/issues #491: PN532 NFC/RFID on Wemos D1 Mini, how to connect?

JBS5, 2019-06-28:
Just received a PN532 NFC/RFID (this one: https://nl.aliexpress.com/item/32899839956.html) and I am wondering how to connect this one to a Wemos D1 Mini. I know it is a different model then mentioned in the docs, but maybe this one could work too.

According to the docs (https://esphome.io/components/binary_sensor/pn532.html) I need clk, miso, mosi and cs pins. On the board there are these:

[image]

I suppose MOSI and MISO are clear, but not sure how to connect CLK and CS pins. Any experience with this kind of boards here?

OttoWinter, 2019-06-28:
CLK=SCK
CS=SS

OttoWinter, 2019-06-28:
It's even on the SPI page: https://esphome.io/components/spi.html

closed 2019-06-28 by OttoWinter

JBS5, 2019-06-28:
Connected it this way but the log shows:
[18:11:40][E][pn532:345]: Wake Up command failed!

Not sure if these mean this pn532 is not supported?