esphome/issues #493: Pulse counter debounce

blakkdukk, 2019-06-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Chromebook running pip 19.0.3 (python 2.7)
**ESP (ESP32/ESP8266, Board/Sonoff):**

DOIT ESP32 devkit v1
**Affected component:**

https://esphome.io/components/sensor/pulse_counter.html
https://esphome.io/components/sensor/index.html#throttle-heartbeat-debounce-delta

**Description of problem:**
The pulse counter component seems to ignore sensor filter 'debounce'. I've tried to set debounce up to 5 seconds, however even a few short "pulses" (or even a single pulse) far shorter than 5 seconds results in many pulses registred.

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: pulse_counter
    update_interval: 10s
    pin: 
      number: GPIO4
      mode: INPUT_PULLDOWN
    count_mode:
      rising_edge: DISABLE
      falling_edge: INCREMENT
    internal_filter: 13us
    unit_of_measurement: 'dl'
    name: 'Vattenmatare inkommande volym'
    filters:
      - lambda: return (x / 5.8);
      - debounce: 5s
```

**Logs (if applicable):**

```
[22:28:04][C][pulse_counter:147]: Pulse Counter 'Vattenmatare inkommande volym'
[22:28:04][C][pulse_counter:147]:   Unit of Measurement: 'dl'
[22:28:04][C][pulse_counter:147]:   Accuracy Decimals: 2
[22:28:04][C][pulse_counter:147]:   Icon: 'mdi:pulse'
[22:28:04][C][pulse_counter:148]:   Pin: GPIO4 (Mode: INPUT_PULLDOWN)
[22:28:04][C][pulse_counter:149]:   Rising Edge: DISABLE
[22:28:04][C][pulse_counter:150]:   Falling Edge: INCREMENT
[22:28:04][C][pulse_counter:151]:   Filtering pulses shorter than 13 µs
[22:28:04][C][ota:029]: Over-The-Air Updates:

<snip>

[22:28:54][D][pulse_counter:158]: 'Vattenmatare inkommande volym': Retrieved counter: 0.00 pulses/min
[22:28:59][D][sensor:092]: 'Vattenmatare inkommande volym': Sending state 0.00000 dl with 2 decimals of accuracy

< 3 rapid touches to the pin here>

[22:29:04][D][pulse_counter:158]: 'Vattenmatare inkommande volym': Retrieved counter: 90.00 pulses/min
```

**Additional information and things you've tried:**
I've tried internal pullup/pulldown and using a resistor to prevent floating errors.
I've tried the gpio sensors with delayed_on and delayed_off filter and that works as I expect 'debounce' to work.
I've tried two ESPs.

OttoWinter, 2019-06-29:
I think you don't understand what the debounce filter does. Citing from docs:

> Only send values if the last incoming value is at least specified time period old. For example if two values come in at almost the same time, this filter will only output the last value and only after the specified time period has passed without any new incoming values.

The debounce filter is a generic filter for sensors that prevents values being pushed out very quickly. It does **not** change the value pushed out.

That's what internal_filter is for: With that value you can filter out pulses that are shorter than a set time period.

I will close the issue as I think it's user error in misunderstanding the docs. If you think that's wrong, reopen the issue.

closed 2019-06-29 by OttoWinter

blakkdukk, 2019-06-30:
Thank you for the rapid response!

I might have misunderstood the filters as well as the component.

With an update interval of 60s, I expected touching the signal pin quickly 3 times would result in a pulse counter value of 3 pulses per minute. However, the pulse counter reported 90 pulses instead. I thought that the high value was caused by a 'switch bounce' and tried the 'debounce' filter to eliminate this.

As far as I can tell, neither internal_filter, debounce nor throttle solves my problem.

OttoWinter, 2019-07-01:
Yes, it's likely due to a fast-switching pin value - so debouncing with internal_filter will help. Higher values mean better/longer debouncing intervals.

Unfortunately, the internal filter on the ESP32 will only go to 12µs (limitation of hardware). You can work around it though with some external circuitry (see also switch debouncing in hardware).

blakkdukk, 2019-07-01:
I solved this by adding a binary sensor as debouncer that also controlled a gpio output. The pulse counter than counted debounced values on a pin connected to the gpio output (gpio 4 and 22 connected).

Thanks you for support and a very nice framework!

```
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO23
      mode: INPUT_PULLUP
    name: "pulse input"
    filters:
      - delayed_on: 10ms
      - delayed_off: 10ms
    on_press:
      then:
        output.turn_on: gpio_22
    on_release:
      then:
        output.turn_off: gpio_22
output:
  - platform: gpio
    pin:
      number: GPIO22
    id: gpio_22

sensor:
  - platform: pulse_counter
    update_interval: 60s
    internal_filter: 13us
    pin:
      number: GPIO4
      mode: INPUT_PULLUP
    unit_of_measurement: 'dl'
    name: 'Vattenmatare inkommande volym'
    filters:
      - lambda: return (x / 5.8);
```