esphome/issues #494: Coolix IR mqtt auto discovery causes errors in Home assistant

cvb941, 2019-06-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker Home Assistant 0.95.4
EspHome 1.13.0

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 Wemos D1 mini

**Affected component:**

Coolix IR Remote https://esphome.io/components/climate/coolix.html

**Description of problem:**
I'm using the Coolix climate component with MQTT and discovery enabled in home assistant and HA seems to be having issues with it. The homescreen is stuck on loading data, and the logs seem to print out the two errors bellow.

I also use recorder with mariadb if it is important..

Would this be a bug on the ESPHome side or home assistant?

**Problem-relevant YAML-configuration entries:**
HA:
```yaml
# Configure a default setup of Home Assistant (frontend, api, etc)
default_config:

# Uncomment this if you are using SSL/TLS, running in Docker container, etc.
# http:
#   base_url: example.duckdns.org:8123

# Text to speech
tts:
  - platform: google_translate

group: !include groups.yaml
automation: !include automations.yaml
script: !include scripts.yaml

recorder:
  db_url: mysql://xxx:xxx@tasks.mariadbnode/homeassistant?charset=utf8

mqtt:
  broker: tasks.mosquitto
  discovery: true
```
EspHome:
```yaml
esphome:
  name: byt_climate
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "xxx"
  password: "xxx"

# Enable logging
logger:

mqtt:
  broker: raspi1

ota:

# IR LED
remote_transmitter:
  pin: D5
  carrier_duty_percent: 50%

climate:
  - platform: coolix
    name: "Flat AC"
```

**Logs (if applicable):**

```
2019-06-29 20:24:44 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: AUTO
2019-06-29 20:25:26 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: AUTO
2019-06-29 20:25:28 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: OFF
2019-06-29 20:25:29 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: OFF
2019-06-29 20:25:31 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: OFF
2019-06-29 20:25:52 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: OFF
2019-06-29 20:25:53 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: OFF
2019-06-29 20:26:06 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: HEAT
```
```
2019-06-2920:20:32ERROR (MainThread)[  
   homeassistant.components.websocket_api.http.connection.1829211824
]Unable to serialize to JSON:Out of range float values are not JSON compliant{  
   'id':3,
   'type':'result',
   'success':True,
   'result':[  
      <state sun.sun=above_horizon; next_dawn=2019-06-30T02:11:31+00:00,
      next_dusk=2019-06-29T19:38:27+00:00,
      next_midnight=2019-06-29T22:54:59+00:00,
      next_noon=2019-06-30T10:54:53+00:00,
      next_rising=2019-06-30T02:52:58+00:00,
      next_setting=2019-06-29T18:56:56+00:00,
      elevation=4.34,
      azimuth=300.67,
      rising=False,
      friendly_name=Sun @ 2019-06-29T18:33:29.490524+02:00      >,
      <state person.**=unknown; editable=True,
      id=9f03d210def34b82908dba7549575520,
      user_id=3e917b0b9e8e429c9368f98c00df2a73,
      friendly_name=*** @ 2019-06-29T18:33:35.941958+02:00      >,
      <state climate.flat_ac=unknown; current_temperature=None,
      min_temp=17.0,
      max_temp=30.0,
      temperature=nan,
      target_temp_step=1.0,
      operation_mode=None,
      operation_list=      [  
         'auto',
         'off',
         'cool',
         'heat'
      ],
      friendly_name=Flat AC,
      supported_features=129 @ 2019-06-29T20:12:57.039633+02:00      >
   ]
}
```

**Additional information and things you've tried:**
I have managed to circumvent this bug by disabling the MQTT discovery in Home Assistant and defining the climate component there manually, like this:
```yaml
climate:
- platform: mqtt
  name: Flat AC
  modes:
      - "off"
      - "auto"
      - "cool"
      - "heat"
  mode_command_topic: "byt_climate/climate/flat_ac/mode/command"
  temperature_command_topic: "byt_climate/climate/flat_ac/target_temperature/command"
```
Still, the MQTT discovery would be great.

closed 2019-07-01 by OttoWinter

cvb941, 2019-07-02:
I loaded up the latest dev with the fix. Looks like the first of the two issues with the invalid mode is fixed now.

However, the second issue with JSON serializing still seems to persist, here's the log.

```
2019-07-02 12:15:01 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection.1913229072] Unable to serialize to JSON: Out of range float values are not JSON compliant

{
   'id':2,
   'type':'event',
   'event':<Event state_changed   [
      L
   ]:entity_id=climate.flat_ac,
   old_state=<state climate.flat_ac=auto; current_temperature=26.0,
   min_temp=17.0,
   max_temp=30.0,
   temperature=nan,
   target_temp_step=1.0,
   operation_mode=auto,
   operation_list=   [
      'auto',
      'off',
      'cool',
      'heat'
   ],
   friendly_name=Flat AC,
   supported_features=129 @ 2019-07-02T12:12:59.765986+02:00   >,
   new_state=<state climate.flat_ac=auto; current_temperature=26.0,
   min_temp=17.0,
   max_temp=30.0,
   temperature=nan,
   target_temp_step=1.0,
   operation_mode=auto,
   operation_list=   [
      'auto',
      'off',
      'cool',
      'heat'
   ],
   friendly_name=Flat AC,
   supported_features=129 @ 2019-07-02T12:12:59.765986+02:00   >>
}
```

OttoWinter, 2019-07-03:
Yep the second one is a bug in HA - it should not accept NAN values in the MQTT climate code (and convert those to None). Would be best to create a bug report about it in HA.