esphome/issues #495: Preferences save Guru Meditation Error

petr-simek, 2019-07-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

linux, pip, version 1.13.6
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32S

**Affected component:**

Custom

**Description of problem:**
I created custom component for Motor, component used encoder for position and break. It work ok but if I used `preferences` I got error. Preferences is used for store last encoder position. It is able to store encoder state 2-5 times than it break.
Can you help me what I do wrong?

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: rotary_encoder
    id: rotary_encoder_
    name: "Rotary Encoder"
    pin_a: 4
    pin_b: 16

```
no yaml
**Logs (if applicable):**

```
[07:15:33]Memory dump at 0x40162294: bad00bad bad00bad bad00bad
[07:15:33]Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
[07:15:33]Memory dump at 0x40162294: bad00bad bad00bad bad00bad
[07:15:33]Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
[07:15:34]Memory dump at 0x40162294: bad00bad bad00bad bad00bad
[07:15:34]Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
[07:15:34]Memory dump at 0x40162294: bad00bad bad00bad bad00bad
[07:15:34]Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
[07:15:34]Memory dump at 0x40162294: bad00bad bad00bad bad00bad
[07:15:34]Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
[07:15:34]Memory dump at 0x40162294: bad00bad bad00bad bad00bad
[07:15:34]Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
[07:15:34]Memory dump at 0x40162294: bad00bad bad00bad bad00bad
[07:15:34]Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
[07:15:34]Memory dump at 0x40162294: bad00bad bad00bad bad00bad
[07:15:34]Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
[07:15:34]MemoGuru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
[07:15:34]Debug exception reason: Stack canary watchpoint triggered (loop�      @4) 
```
``` 
#include <utility>

#include "motor.h"

static unsigned int position_abs(unsigned int a, unsigned int b) {
    return a >= b ? a - b : b - a;
}

static const char *motor_Tag = "motor";

void Motor::off() {
    esphome::ESP_LOGD(motor_Tag, "off 1");
    l_enable->digital_write(false);
    r_enable->digital_write(false);
    esphome::ESP_LOGD(motor_Tag, "off 2");
    l_pwm->set_level(0);
    r_pwm->set_level(0);
    state = OFF;
    speed = 0;
    esphome::ESP_LOGD(motor_Tag, "off 3");
    float state = encoder->state;
    esphome::ESP_LOGD(motor_Tag, "off 4 %d", rtc.is_initialized());
//FAILED HERE  this->rtc.save(&state);
    this->rtc.save(&state);
    esphome::ESP_LOGD(motor_Tag, "off 5");
}

void Motor::setup() {
    Component::setup();
    object_id_ = "motor_sensor";
    encoder->state = getInitialEncoder();
    off();
}

void Motor::update() {
    if (state == OFF || disabled) {
        return;
    }
    encoder_break();
}

float Motor::getInitialEncoder() {
    rtc = esphome::global_preferences.make_preference<float>(this->get_object_id_hash());
    float initial_state;
    if (!rtc.load(&initial_state))
        return 0;
    return initial_state;
}

void Motor::encoder_break() {
    unsigned long now = millis();
    unsigned long diff = get_period(last_encoder_move, now);
    unsigned long encoder_state = encoder->state;
    if (diff >= 300 && position_abs(last_position, encoder_state) <= 0) {
        esphome::ESP_LOGD(motor_Tag, "off encoder_break");
        off();
        return;
    } else if (diff >= 300) {
        last_encoder_move = now;
        last_position = encoder_state;
    }
}

uint32_t Motor::hash_base() { return 1985890752UL; }
```

```
class Motor : public esphome::PollingComponent, esphome::Nameable {
public:
    Motor(uint32_t updateInterval, esphome::GPIOPin *rEnable, esphome::GPIOPin *lEnable,
          esphome::ledc::LEDCOutput *rPwm, esphome::ledc::LEDCOutput *lPwm,
          esphome::rotary_encoder::RotaryEncoderSensor *encoder);

    void off();

    void update() override;

    void setup() override;

private:
    float getInitialEncoder();

    void encoder_break();

    uint32_t hash_base() override ;

     esphome::rotary_encoder::RotaryEncoderSensor *encoder;
    esphome::ESPPreferenceObject rtc;
};
```
**Additional information and things you've tried:**

OttoWinter, 2019-07-01:
Please post your entire code - including how you register and create the component. The error indicates that you're calling encoder_break from an interrupt, which will lead to an error.

petr-simek, 2019-07-01:
Component is create and register in main.cpp 
```
 motor = new Motor(10, new GPIOPin(22, OUTPUT), new GPIOPin(15, OUTPUT), output_r_pwm, output_l_pwm, rotary_encoder_);
   App.register_component(motor);
```
I used button for run/stop motor like this
```
  - platform: gpio
    pin:
      number: 33
      mode: INPUT_PULLUP
      inverted: True
    name: "up"
    on_click:
      then:
        - lambda: |-
            motor->dir_ccw();
```
I use lambda for `motor->dir_ccw();` which has 2 functions. First push enable motor and second push stop motor and when motor is stop it call `this->rtc.save(&state);` ` esphome::ESPPreferenceObject rtc;`. 
Is this function run from interrupt?

```yml
esphome:
  name: stul
  platform: ESP32
  board: esp32dev
  includes:
    - esphomelib-component/Button.h
    - esphomelib-component/Button.cpp
    - esphomelib-component/JC_Button.h
    - esphomelib-component/JC_Button.cpp

wifi:
  ssid: 'Malarodinka_iot'
  password: 'lalamrdi1X'

# Enable logging
logger:
  level: DEBUG
api:
ota:

esp32_touch:
  setup_mode: False

binary_sensor:
  - platform: gpio
    pin:
      number: 33
      mode: INPUT_PULLUP
      inverted: True
    name: "up"
    on_click:
      then:
        - lambda: |-
            motor->dir_ccw();
  - platform: gpio
    pin:
      number: 5
      mode: INPUT_PULLUP
      inverted: True
    name: "down"
    on_click:
      then:
        - lambda: |-
            motor->dir_cw();
#  - platform: gpio
#    pin:
#      number: 25
#      mode: INPUT_PULLUP
[... 789 lines cut ...]

    int L_SPEED_STEP_DURATION = 200;
    int position_first = 50;
    int position_second = 100;

    int position_third = 10;
    int speed = 0;

    esphome::rotary_encoder::RotaryEncoderSensor *encoder;
    MotorState state = OFF;
    bool disabled = false;
    bool disabled_first = true;
    long next_position = -1;
    long last_position = 0;
    void setSpeed(MotorState state, int speed);
    unsigned long last_tick = 0;

    unsigned long last_encoder_move = 0;

    esphome::ESPPreferenceObject rtc;
};

#endif //POKOJ_MOTOR_H

```

OttoWinter, 2019-07-01:
Well I don't immediately see what would cause the issue you see with the crash, but I see a couple of other bugs:

 - Motor::setup() calls setup functions of the ledc objects - but setup is already called from them from their own component lifecycle - remove those calls.
 - Motor::setup also sets encoder->state - but encoder->state is only an internal value. Changing it won't affect the states published in the future. Haven't looked at the code too closely, but this might make the component save the state many hundred times per second, in which case I think the esp-idf nvs SDK may also throw issues.

Please also try using the `globals` component for this saved data instead of using your own rtc variable.

petr-simek, 2019-07-01:
duplicit setups are removed. 
`encoder->state = custom_vule` is call only once in `Motor::setup()`. Is there any solution how to set init state?

How you mean this? `Please also try using the globals component for this saved data instead of using your own rtc variable.` 
I used same code as you used in `switch` and I don't know there is some other component which can be used for data store. Can you please navigate me where shoud I look?

glmnet, 2019-07-01:
Search globals in docs. Is under automation

petr-simek, 2019-07-12:
I rewrite preference(rtc) to globals and now it save value but then it throw same error as before.
```
[16:48:47]Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
[16:48:47]Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
[16:48:47]Memory dump at 0x40161040: bad00bad bad00bad bad00bad
```
Interaction with global is only in one method
```
void Motor::_off() {
     l_enable->digital_write(false);
    r_enable->digital_write(false);
    ESP_LOGD(motor_Tag, "off 2");
    l_pwm->set_level(0);
    r_pwm->set_level(0);
    state = OFF;
    speed = 0;
    globalEncoderState->value() = encoder->state;
    ESP_LOGD(motor_Tag, "off 3");
}
```
Method is call only from Component#update() which is not call from an interrupt and only once in while. 

In my case even if I stop motor, rotary encoder do other few steps. When I postponed encoder state save (no more steps) it start working properly. Is it by accident or rotary encoder has some bug or something else?

```
void Motor::update() {
    if (!encoder_store) {
        save_encoder();
    }
    if (state == SHUTDOWN) {
        _off();
        return;
    }
.......
}
void Motor::_off() {
    l_enable->digital_write(false);
    r_enable->digital_write(false);
    l_pwm->set_level(0);
    r_pwm->set_level(0);
    state = OFF;
    speed = 0;
    encoder_store = false;
}

void Motor::save_encoder() {
    unsigned long now = millis();
    unsigned long diff = get_period(last_encoder_move, now);
    unsigned long encoder_state = encoder->state;
    if (diff >= 1000 && position_abs(last_position, encoder_state) == 0) {
        globalEncoderState->value() = encoder->state;
        encoder_store = true;
    }
}
```

OttoWinter, 2019-07-12:
> I rewrite preference(rtc) to globals and now it save value but then it throw same error as before.

Can you try removing the globals saving and see if the error still occurs? I highly doubt it's the globals, I think I would have heard bug reports about it (including from myself) if there were a bug there.

> Method is call only from Component#update() which is not call from an interrupt and only once in while.

No, `update` is only called from non-ISR context.

> In my case even if I stop motor, rotary encoder do other few steps. When I postponed encoder state save (no more steps) it start working properly. Is it by accident or rotary encoder has some bug or something else?

I don't understand what you mean there. Could you try rephrasing and explaining with code?

petr-simek, 2019-07-13:
> Can you try removing the globals saving and see if the error still occurs? I highly doubt it's the globals, I think I would have heard bug reports about it (including from myself) if there were a [...]
If I remove `globalEncoderState->value() = encoder->state;` it is work without error.

> I don't understand what you mean there. Could you try rephrasing and explaining with code?

1) press button
2) call _off()
3) Disconnect electricity from motor (It take few ms before motor actually stop. I don't know how to explain it, If you look at https://youtu.be/NFvMC3l3fGY?t=79 1:20-1:30)
4) Rotary encoder count other 8 steps
5) If encoder has same count for 1s call  `globalEncoderState->value() = encoder->state;`
6) ok

but if I changed it
1) press button
2) call _off()
3) Disconnect electricity from motor and call  `globalEncoderState->value() = encoder->state;`
4a) 1 of 3 attempts it count
4b) 2 of 3 attempts it failed with error

If you don't understand what I mean, I'm not surprised, my English is not very good. You helped me with my problem and we can close the issue.

closed 2020-08-07 by stale[bot]