esphome/issues #497: on_shutdown hook not triggered before deep sleep

fondberg, 2019-07-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome 1.13.6

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266
**Affected component:**

https://esphome.io/components/esphome.html#on-shutdown

**Description of problem:**
`on_shutdown` hook not run before entering deep sleep.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: vaextsensor
  platform: ESP8266
  board: d1_mini
  on_boot:
    then:
      - logger.log: "on boot triggered"
      # - delay: 2s
  on_shutdown:
    then:
      - logger.log: "on shutdown triggered"
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

A fix is commited by @OttoWinter here: https://github.com/esphome/esphome/commit/61215b41c0be87106f31b6291446b335cf6eed

closed 2019-07-03 by OttoWinter

fondberg, 2019-07-08:
Unfortunatly it doesn't seem that the issue is solved. The hook is never triggered

brandond, 2019-07-08:
Are you sure it's not triggered? In my experience the shutdown hooks can be a bit finicky since any remote actions (such as logging) aren't waited for, so if you're just testing with sending a log message it won't necessarily wait for it to actually make it out across the network before shutting down. You could try putting a wait in there after the log so that it has time to make it out before sleeping.

fondberg, 2019-07-09:
Hi. I tested with switching on and off a led a few times as well as a delay after the log msg but it doesn't seem to trigger.

OttoWinter, 2019-07-10:
Huh, ok - i didn't check if the patch actually fixed the issue. Just saw that it was *one* issue.

Is the config you posted in the original description what you're using now? What exactly do you see when entering deep sleep (log output).

fondberg, 2019-07-10:
Here is the code I used to test it, I added a led blink to see what was going on because I realized that logging might not happen in time. Ultimate success would be if D3 pin is low in deep sleep.

```
esphome:
  name: vaextsensor
  platform: ESP8266
  board: d1_mini
  arduino_version: latest
  on_boot:
    priority: -100.0
    then:
      - logger.log: "on boot triggered"
      - switch.turn_on: red_led_switch
      - switch.turn_on: soil_sensor_vcc
  on_shutdown:
    then:
      - logger.log: "on shutdown triggered"
	  - switch.turn_off: soil_sensor_vcc
      - switch.turn_on: red_led_switch
      - delay: 1s
      - switch.turn_off: red_led_switch
      - delay: 1s
      - switch.turn_on: red_led_switch
      - delay: 1s
      - switch.turn_off: red_led_switch
      - delay: 1s
      - switch.turn_on: red_led_switch
      - delay: 1s
      - switch.turn_off: red_led_switch

wifi:
  ssid: XXX
  password: XXX

logger:

ota:

deep_sleep:
  id: deep_sleep_1
  run_duration: 1min
  sleep_duration: 1min

switch:
  - platform: gpio
    pin: D1
    id: red_led_switch
    internal: true
  - platform: gpio
    pin: D3
    id: soil_sensor_vcc
    internal: true
```

OttoWinter, 2019-07-10:
Ok so `delay` will not work in on_shutdown. `delay` is asynchronous, so the shutdown code will not wait for the shutdown hooks to be completed. All other stuff (before delay) should work though (but logs will probably only work over USB, because network is probably shut down before the logger can send the message).

I will try to test the on_shutdown hook later and see if I can replicate it.

fondberg, 2019-07-10:
I see. Well, the most important is to set the pin top low so maybe you just measure the pin to see that this is done?

OttoWinter, 2019-07-10:
Ah yes and another thing I should mention: I'm not sure pin states will persist during deep sleep - the GPIO mux is handled by the part of the system that's shut down during deep sleep. So it might well be that the pins go back to their default state (see ESP8266 page) during deep sleep.

glmnet, 2019-07-10:
I can't help making the obvious question here: What do you want to happen `on_shutdown`?

fondberg, 2019-07-10:
I wonder is this is something that is needed...

https://github.com/esp8266/esp8266-wiki/wiki/Memory-Map
Look for 
"Pull-down during sleep"

Is this something that I can add in a lambda in the shutdown hook?

fondberg, 2019-07-11:
@glmnet as stated a couple of times the intent is to set pins to low to preserve battery in deep sleep

fondberg, 2019-07-12:
I believe I have found a way around this. The pins behave differently and using different pins (GPIO13 and 15) they stay low during deep sleep.