esphome/issues #500: ESP32-D Dual core support

blademckain, 2019-07-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32-D

**Affected component:**
platform: ESP32

**Description of problem:**

hi,
esphome support esp32-d?
if yes, can esphome use both cores?

glmnet, 2019-07-03:
Can you expand the question? How should esphome use both cores? How it matters for the end user?

OttoWinter, 2019-07-03:
As @glmnet said, please elaborate. Some points from me:

 - Yes, in dual core setups the second core is used too for some tasks (I believe wifi and so). But ESPHome's main loop runs in a single thread.
 - Processor speed has not been a problem with ESPHome, in fact the ESPs are way overspecced for ESPHome's uses.
 - If the user needs to, they can absolutely create separate threads on other cores with the freeRTOS functions, but then you also need to use mutexes etc properly.

I will close the issue as the feature request is very sparsely filled out and no reason has been given (please add those for reopening it).

closed 2019-07-03 by OttoWinter