esphome/issues #502: 433MHz RC Transmitter not working

AnthonyKNorman, 2019-07-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
HASSIO

ESP8266 D1 Mini
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**Affected component:**

https://esphome.io/components/remote_transmitter.html

**Description of problem:**
I have two WEMOS D1 Mini. One has a 433MHz receiver, the other has the transmitter.
When I use the Arduino IDE I can use the rc-switch library to pass remote codes between the pair. This confirms the hardware config. Using the YAML file below, Home Assistant finds the switch, the logs say it's sending, but nothing is seen on the receiver.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: esp_rf_remote_1
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "TP-LINK_FBA9"
  password: "58523346"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "s2sfilwY"

ota:
  password: "s2sfilwY"

remote_transmitter:
  pin: D2
  # RF uses a 100% carrier signal
  carrier_duty_percent: 100%
  
switch:
  - platform: template
    name: RF Power Button
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '000000010001010100110011'
          protocol: 1

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
INFO Reading configuration...
INFO Starting log output from esp_rf_remote_1.local using esphome API
INFO Connecting to esp_rf_remote_1.local:6053 (192.168.1.127)
INFO Successfully connected to esp_rf_remote_1.local
[16:06:41][I][app:096]: esphome version 1.13.6 compiled on Jul  3 2019, 15:46:48
[16:06:41][C][wifi:372]: WiFi:
[16:06:41][C][wifi:254]:   SSID: [redacted]
[16:06:41][C][wifi:255]:   IP Address: 192.168.1.127
[16:06:41][C][wifi:257]:   BSSID: [redacted]
[16:06:41][C][wifi:258]:   Hostname: 'esp_rf_remote_1'
[16:06:41][C][wifi:262]:   Signal strength: -81 dB ▂▄▆█
[16:06:41][C][wifi:263]:   Channel: 1
[16:06:41][C][wifi:264]:   Subnet: 255.255.255.0
[16:06:41][C][wifi:265]:   Gateway: 192.168.1.1
[16:06:41][C][wifi:266]:   DNS1: 8.8.8.8
[16:06:41][C][wifi:267]:   DNS2: 8.8.4.4
[16:06:41][C][template.switch:058]: Template Switch 'RF Power Button'
[16:06:41][C][template.switch:059]:   Restore State: NO
[16:06:41][C][template.switch:060]:   Optimistic: NO
[16:06:41][C][logger:137]: Logger:
[16:06:41][C][logger:138]:   Level: DEBUG
[16:06:41][C][logger:139]:   Log Baud Rate: 115200
[16:06:41][C][logger:140]:   Hardware UART: UART0
[16:06:41][C][remote_transmitter:018]: Remote Transmitter...
[16:06:41][C][remote_transmitter:019]:   Carrier Duty: 100%
[16:06:41][C][remote_transmitter:020]:   Pin: GPIO4 (Mode: OUTPUT)
[16:06:41][C][ota:029]: Over-The-Air Updates:
[16:06:41][C][ota:030]:   Address: esp_rf_remote_1.local:8266
[16:06:41][C][ota:032]:   Using Password.
[16:06:41][C][api:103]: API Server:
[16:06:41][C][api:104]:   Address: esp_rf_remote_1.local:6053
[16:06:58][D][switch:021]: 'RF Power Button' Turning ON.
[16:06:58][D][remote_transmitter:066]: Sending remote code...

This is from the USB connection
[D][api:573]: Client 'ESPHome v1.13.6 (192.168.1.97)' connected successfully!
[I][app:096]: esphome version 1.13.6 compiled on Jul  3 2019, 15:46:48
[C][wifi:372]: WiFi:
[C][wifi:254]:   SSID: 'XXXXXXXXX''
[C][wifi:255]:   IP Address: 192.168.1.127
[C][wifi:257]:   BSSID: XXXXXXXXXXXX
[C][wifi:258]:   Hostname: 'esp_rf_remote_1'
[C][wifi:262]:   Signal strength: -81 dB ▂▄▆█
[C][wifi:263]:   Channel: 1
[C][wifi:264]:   Subnet: 255.255.255.0
[C][wifi:265]:   Gateway: 192.168.1.1
[C][wifi:266]:   DNS1: 8.8.8.8
[C][wifi:267]:   DNS2: 8.8.4.4
[C][template.switch:058]: Template Switch 'RF Power Button'
[C][template.switch:059]:   Restore State: NO
[C][template.switch:060]:   Optimistic: NO
[C][logger:137]: Logger:
[C][logger:138]:   Level: DEBUG
[C][logger:139]:   Log Baud Rate: 115200
[C][logger:140]:   Hardware UART: UART0
[C][remote_transmitter:018]: Remote Transmitter...
[C][remote_transmitter:019]:   Carrier Duty: 100%
[C][remote_transmitter:020]:   Pin: GPIO4 (Mode: OUTPUT)
[C][ota:029]: Over-The-Air Updates:
[C][ota:030]:   Address: esp_rf_remote_1.local:8266
[C][ota:032]:   Using Password.
[C][api:103]: API Server:
[C][api:104]:   Address: esp_rf_remote_1.local:6053
[D][switch:021]: 'RF Power Button' Turning ON.
[D][remote_transmitter:066]: Sending remote code...

```

**Additional information and things you've tried:**
This is the working Ardunio sketch
```
/*
  Example for different sending methods
  
  https://github.com/sui77/rc-switch/
  
*/

#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();

void setup() {

  Serial.begin(9600);
  
  // Transmitter is connected to Arduino Pin #10  
  mySwitch.enableTransmit(4);
  
  // Optional set protocol (default is 1, will work for most outlets)
  // mySwitch.setProtocol(2);

  // Optional set pulse length.
  // mySwitch.setPulseLength(320);
  
  // Optional set number of transmission repetitions.
  // mySwitch.setRepeatTransmit(15);
  
}

void loop() {

  /* See Example: TypeA_WithDIPSwitches */
  mySwitch.switchOn("11111", "00010");
  delay(1000);
  mySwitch.switchOff("11111", "00010");
  delay(1000);

  /* Same switch as above, but using decimal code */
  mySwitch.send(5393, 24);
  delay(1000);  
  mySwitch.send(5396, 24);
  delay(1000);  

  /* Same switch as above, but using binary code */
  mySwitch.send("000000000001010100010001");
  delay(1000);  
  mySwitch.send("000000000001010100010100");
  delay(1000);

  /* Same switch as above, but tri-state code */ 
  mySwitch.sendTriState("00000FFF0F0F");
  delay(1000);  
  mySwitch.sendTriState("00000FFF0FF0");
  delay(1000);

  delay(20000);
}
```

OttoWinter, 2019-07-05:
Please include what you mean by "receiver". The code posted above only includes transmitter code. If the receiver is an esphome device, please also include the RAW code captured.

AnthonyKNorman, 2019-07-05:
The receiver is an esp8266 running the arduino rc-switch library. I have also since found that adding a repeat of 10 this occasionally works. Please let me know if there is anything else I can do to help.

mikekay1, 2019-07-08:
Since I cannot message you directly, I would suggest you remove your password information from your ESPHome config log in the first post.

AnthonyKNorman, 2019-07-09:
I have now tested this against an ESPHome receiver

YAML
```
remote_receiver:
  pin: D1
  dump:
    - rc_switch
  tolerance: 50%
  filter: 250us
  idle: 4ms
  buffer_size: 2kb
  ```

LOGS
```
[11:40:37][C][remote_receiver.esp8266:060]:   Pin: GPIO5 (Mode: INPUT)
[11:40:37][C][remote_receiver.esp8266:065]:   Buffer Size: 2000
[11:40:37][C][remote_receiver.esp8266:066]:   Tolerance: 50%
[11:40:37][C][remote_receiver.esp8266:067]:   Filter out pulses shorter than: 250 us
[11:40:37][C][remote_receiver.esp8266:068]:   Signal is done after 4000 us of no changes
[11:40:37][C][ota:029]: Over-The-Air Updates:
[11:40:37][C][ota:030]:   Address: esp8266_test.local:8266
[11:40:37][C][ota:032]:   Using Password.
[11:40:37][C][api:103]: API Server:
[11:40:37][C][api:104]:   Address: esp8266_test.local:6053
[11:40:41][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='111111111'
[11:40:45][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=6 data='00000000'
[11:40:55][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='000000010001010100110011'
[11:40:55][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=2 data='000000000'
[11:40:55][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='000000010001010100110011'
[11:40:55][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='000000010001010100110011'
[11:40:55][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='000000010001010100110011'
[11:40:55][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='0000000100010'
[11:40:55][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='000000010001010100110011'
[11:40:55][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='000000010001010100110011'
[11:40:55][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=1 data='000000010001010100110011'
[11:41:01][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=6 data='000000001'
[11:41:02][D][remote.rc_switch:240]: Received RCSwitch Raw: protocol=6 data='000000000000000000'
```
The 11:40:55 lines are when I activated the switch. The others are spurious.

Current transmitter YAML
```
remote_transmitter:
  pin: D2
  # RF uses a 100% carrier signal
  carrier_duty_percent: 100%
   
  
switch:
  - platform: template
    name: RF Power Button
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '000000010001010100110011'
          protocol: 1
          repeat:
            times: 10
            wait_time: 50ms
         
```

OttoWinter, 2019-07-10:
@AnthonyKNorman So from what I can see it *is* working? 

Also this looks sufficiently similar to https://github.com/esphome/issues/issues/508 that I will flag it as a duplicate of said issue.

closed 2019-07-10 by OttoWinter