esphome/issues #503: Reading ADS1115 timed out esphome

Luc3as, 2019-07-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio Stable version

**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 mini
**Affected component:**

https://esphome.io/components/sensor/ads1115.html

**Description of problem:**
After some time, component ADS1115 loops itself with message Reading ADS1115 timed out esphome, flooding the logs, and making the device unresponsive because it is trying to connect very quickly I think. After restarting the device it is working okay for some time again. Maybe some i2c timing of bus could be stretched but I did not find any parameter to change it. 

**Problem-relevant YAML-configuration entries:**
```yaml
i2c:
  sda: D2
  scl: D4
  scan: True

ads1115:
  - address: 0x48

sensor:
  - platform: ads1115
    multiplexer: 'A3_GND'
    gain: 4.096
    name: "Hladina vody v hydropónií"  

  - platform: ads1115
    multiplexer: 'A0_GND'
    gain: 4.096
    name: "Vlhkosť rajčiny vľavo" 
    filters:
      - lambda: return ( 100 - x / 4.096 * 100 - 30) / ( 66 - 30 ) * 100 ;
    unit_of_measurement: "%"
    icon: "mdi:water-percent"
    accuracy_decimals: 0

```

**Logs (if applicable):**

```
Reading ADS1115 timed out esphome

I restarted the device withtou copyying log , I will do when it occurs next time.

```

**Additional information and things you've tried:**

glmnet, 2019-07-04:
I am using i2c with arduino custom firmware (arduino is slave) and usually have no issues. Works very reliable. However if I reset arduino comm is lost and I need to reset esp8266. 
I guess there is no real issue with ads here, however it will be interesting to discuss an auto healing routine for i2c bus. 
Btw i2c speed is configurable. Try other speed and see if problem persist or happens more often. 
https://esphome.io/components/i2c.html

Luc3as, 2019-07-04:
I already tried to change frequency of i2c bus, with higher frequency error
occured more quickly. I think there would be "solution" to wait some time
like few second between trying to communicate with ads again, so it would
not spam and load whole esp, reading would not work but all other things
hanged to esp should be responsive.

glmnet, 2019-07-04:
Ahhh I missed the part of the esp being unresponsive. That shouldn’t be the case at all. Which ability does the esp loses specifically? OTA upgrade? Device goes “unavailable” in ha?
How is your circuit? Are you using DuPont cables are they good? Otherwise try another wiring or another set of esp / ads. (If you already did please add the info) it will be nice to isolate the issue between hardware/software side if possible 
Do you have stuff connected to the ads? Something else that might be causing interference?

Luc3as, 2019-07-04:
It is partially unresponsive as it goes to unavailable and API itself is
disconnecting. After few presses of restart button ( the software one) it
resets and is good. I have only this wired to i2c, but there is also one
wire sensor, few npn transistors acting as pwm output, binary input
sensors. everything is soldered and its not moving so I think connections
should be good to go. ADS is reading 2 capacitive moisture sensors only. I
could try to disable ADS for some time and test how stable it is.

OttoWinter, 2019-07-05:
This warning is shown if the sensor does not set the new data flag in the config register:

https://github.com/esphome/esphome/blob/v1.13.6/esphome/components/ads1115/ads1115.cpp#L106

To me this sounds like a problem with the ADS1115 hw - i2c clock stretching or anything of those sorts should not play a role here (the register is polled).

But the logs should not be flooded unless you have chosen a very high update interval. What update interval do you have configured?

Also try esphome dev with ads1115 `continuous_mode: true`

Luc3as, 2019-07-05:
I have 60 seconds polling interval in config, but when timeout is hitted,
it does not take care about it.

phdelodder, 2019-09-09:
I'm also having issues of 'Reading ADS1115 timed out' the only thing that solves it, is by restarting the ESP32. The update interval is set to 30 seconds and also using exponential_moving_average. 

```
    filters:
      - exponential_moving_average:
          alpha: 0.1
          send_every: 12
```

OttoWinter, 2019-10-24:
ESPHome v1.14 has a new option `continuous_mode` for the ADS1115. There the won't be any timeouts anymore because the sensor is actively reading data at all times.

closed 2019-10-24 by OttoWinter