esphome/issues #504: HDC1080 issue with CCS811 HDC1080 combo module.

regnerus, 2019-07-04:
_This issue might be related to #294 however I could not make up if that issue also was related to the CCS811 HDC1080 combo module therefore opening this new issue._

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io ESPHome Beta 1.13.6

**ESP (ESP32/ESP8266, Board/Sonoff):**

DOIT ESP32 DevKit V1

**Affected component:**

HDC1080 Sensor

**Description of problem:**
I recently bought a CCS811 HDC1080 combo module from Aliexpress (https://nl.aliexpress.com/item/32866568687.html). The I2C scan function is able to find both the CCS811 address (0x5A) and the HDC1080 address (0x40) however it results in the following error `Communication with HDC1080 failed!`.  The HDC1080 sensor is however working as expected.

**Problem-relevant YAML-configuration entries:**
```yaml
  - platform: hdc1080
    temperature:
      name: "Master Bedroom Temperature"
      id: master_bedroom_temperature
    humidity:
      name: "Master Bedroom Humidity"
      id: master_bedroom_humidity
    address: 0x40
    update_interval: 60s
  - platform: ccs811
    eco2:
      name: "Master Bedroom eCO2"
    tvoc:
      name: "Master Bedroom TVOC"
    temperature: master_bedroom_temperature
    humidity: master_bedroom_humidity
    baseline: 0x8FBC
    address: 0x5A
    update_interval: 60s
```

**Logs (if applicable):**

```
[21:00:28][C][i2c:028]: I2C Bus:
[21:00:28][C][i2c:029]:   SDA Pin: GPIO21
[21:00:28][C][i2c:030]:   SCL Pin: GPIO22
[21:00:28][C][i2c:031]:   Frequency: 50000 Hz
[21:00:28][I][i2c:033]: Scanning i2c bus for active devices...
[21:00:28][I][i2c:040]: Found i2c device at address 0x40
[21:00:28][I][i2c:040]: Found i2c device at address 0x5A
[21:00:28][C][logger:137]: Logger:
[21:00:28][C][logger:138]:   Level: VERY_VERBOSE
[21:00:28][C][logger:139]:   Log Baud Rate: 115200
[21:00:28][C][logger:140]:   Hardware UART: UART0
[21:00:28][C][hdc1080:028]: HDC1080:
[21:00:28][C][hdc1080:029]:   Address: 0x40
[21:00:28][E][hdc1080:031]: Communication with HDC1080 failed!
[21:00:28][C][hdc1080:033]:   Update Interval: 60.0s
[21:00:28][C][hdc1080:034]:   Temperature 'Master Bedroom Temperature'
[21:00:28][C][hdc1080:034]:     Unit of Measurement: '°C'
[21:00:28][C][hdc1080:034]:     Accuracy Decimals: 1
[21:00:28][C][hdc1080:034]:     Icon: 'mdi:thermometer'
[21:00:28][C][hdc1080:035]:   Humidity 'Master Bedroom Humidity'
[21:00:28][C][hdc1080:035]:     Unit of Measurement: '%'
[21:00:28][C][hdc1080:035]:     Accuracy Decimals: 0
[21:00:28][C][hdc1080:035]:     Icon: 'mdi:water-percent'
[21:00:28][C][ccs811:111]: CCS811
[21:00:28][C][ccs811:112]:   Address: 0x5A
[21:00:28][C][ccs811:113]:   Update Interval: 60.0s
[21:00:28][C][ccs811:114]:   CO2 Sensor 'Master Bedroom eCO2'
[21:00:28][C][ccs811:114]:     Unit of Measurement: 'ppm'
[21:00:28][C][ccs811:114]:     Accuracy Decimals: 0
[21:00:28][C][ccs811:114]:     Icon: 'mdi:periodic-table-co2'
[21:00:28][C][ccs811:115]:   TVOC Sensor 'Master Bedroom TVOC'
[21:00:28][C][ccs811:115]:     Unit of Measurement: 'ppb'
[21:00:28][C][ccs811:115]:     Accuracy Decimals: 0
[21:00:28][C][ccs811:115]:     Icon: 'mdi:radiator'
[21:00:28][C][ccs811:117]:   Baseline: 8FBC

```

**Additional information and things you've tried:**

Tried both the normal and beta versions of ESPHome.

regnerus, 2019-07-08:
Closing this issue since it is probably due to a bad HDC1080 chip, received a second module which does not have this issue!

closed 2019-07-08 by regnerus

Skelly82, 2019-07-09:
Hello I have same issue with HDC1080 in ESPHome, but when I upload simple Arduino sketch and connect to sensor with ClosedCube_HDC1080.h library the measurements work. If I could somehow help to debug this issue, please let me know.

digiblur, 2019-07-14:
Looks like I'm running into the same issue with the CCS811 HDC1080 combo board.  It shows the address of both devices in the i2c scan on Esphome but of course it fails to communicate.  I was able to test ESPEasy on it and the HDC1080 reports back temperature and humidity just fine.

Skelly82, 2019-07-15:
I experienced same behavior on this board, the other two sensors work fine. https://www.aliexpress.com/item/32867694176.html