esphome/issues #506: 'Device Type' dropdown list is sticking out of the frame of the wizard.

W00D00, 2019-07-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
HASS.IO 167
Raspberry Pi3B + SSD boot + Docker
HA 0.95.4
Host system Raspbian GNU/Linux 9 (stretch)

**ESP (ESP32/ESP8266, Board/Sonoff):**
NodeMCU v2 ESP8266

**Affected component:**
-

**Description of problem:**
When I try to create a new configuration with the wizard, I want to set the 'Device Type' from the dropdown list, the list appairs but the list is sticking out of the frame of the wizard. I can see the top items of the list and I can select item very tricky way.

**Problem-relevant YAML-configuration entries:**
-

**Logs (if applicable):**
-

**Additional information and things you've tried:**
A screenshot is attached about the problem.
[image]

OttoWinter, 2019-07-05:
It's a problem with the CSS framework used (materializecss) and not easy to fix.

Please just use the scroll wheel (works great for me).

W00D00, 2019-07-05:
Thanks for your reply. However, I am not able to scroll up with the scroll wheel. I want to select the Nodemcu list item, which is not shown. The tricky way is that I select a shown list item and use the up arrow key to move the selection up but, the list is not moving and the selection is not shown and I do it at random.

closed 2020-04-29 by glmnet