esphome/issues #508: 433mhz not transmitting correct signal

mikekay1, 2019-07-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266 NodeMCU
**Affected component:**

**Description of problem:**
Working in PlatformIO perfectly using RC-Switch code below fires with no repeats and works Infinitely.

**PlatformIO Code-Working**
```
#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();

void setup() {
 mySwitch.enableTransmit(0);
 mySwitch.setProtocol(1, 396); 
 Serial.begin(9600);
}

void loop() {
 mySwitch.send("010000000100010100000001");
 Serial.print("Sent Code [On]\n");
 delay(5000);
 mySwitch.send("010000000100010100000011");
 Serial.print("Sent Code [Off]\n");
 delay(5000);
}
```

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: nodemcu_patiolights
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: --
  password: --
    
  manual_ip:
    static_ip: 192.168.1.198
    gateway: 192.168.1.1
    subnet: 255.255.255.0
# Enable logging
logger:

# Enable Home Assistant API
api:
  password: --

ota:
  password: --
  
remote_transmitter:
  pin: D1
  # RF uses a 100% carrier signal
  carrier_duty_percent: 100%
  
switch:
  - platform: template
    name: Patio_Lights_Power_Button
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '010000000100010100000001'
          protocol: 
            pulse_length: 396
            sync: [1,31]
            zero: [1,3]
            one: [3,1]
            inverted: false
          repeat: 10
```
_Using known RCSwitch protocol-1 values from https://github.com/sui77/rc-switch/blob/a9da9c36820b02fc5613dfe2437e1187bcf5b402/RCSwitch.cpp#L76_
**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
I have tried almost every combination of things possible to get this to work -- in HA the switch doesn't do anything. 

I have confirmed that in the Logs on the ESPhome it is saying it's sending the code. I just don't have the hardware to actually see what was sent. Tried different PINs to no avail.

The outputted values from RC-Switch dumped are in this picture and are all confirmed working using platformIO.

https://community-home-assistant-assets.s3.dualstack.us-west-2.amazonaws.com/original/3X/9/1/911884470427008b9d26f40ea0742a382cce09f1.png

mikekay1, 2019-07-09:
Just keeping this thread updated with my own answers as a log for someone else...

**Resolution**
```
esphome:
  name: nodemcu_patiolights
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid:--
  password: --
    
  manual_ip:
    static_ip: 192.168.1.198
    gateway: 192.168.1.1
    subnet: 255.255.255.0
# Enable logging
logger:

# Enable Home Assistant API
api:
  password: --

ota:
  password: --
  
remote_transmitter:
  pin: D1
  # RF uses a 100% carrier signal
  carrier_duty_percent: 100%
  
switch:

  - platform: template
    name: LED_Controller_Power_ON
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '010000000100010100000001'
          protocol: 
            pulse_length: 396
            sync: [1,31]
            zero: [1,3]
            one: [3,1]
            inverted: false
          repeat: 3
  - platform: template
    name: LED_Controller_Power_OFF
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '010000000100010100000011'
          protocol: 
            pulse_length: 396
            sync: [1,31]
            zero: [1,3]
            one: [3,1]
            inverted: false
          repeat: 3
          
  - platform: template
    name: LED_Controller_Power_LIGHT
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '010000000100010100000100'
          protocol: 
            pulse_length: 396
            sync: [1,31]
            zero: [1,3]
            one: [3,1]
            inverted: false
          repeat: 3

  - platform: template
    name: LED_Controller_Power_100
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '010000000100010100000111'
          protocol: 
            pulse_length: 396
            sync: [1,31]
            zero: [1,3]
            one: [3,1]
            inverted: false
          repeat: 3
          
  - platform: template
    name: LED_Controller_Power_50
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '010000000100010100001000'
          protocol: 
            pulse_length: 396
            sync: [1,31]
            zero: [1,3]
            one: [3,1]
            inverted: false
          repeat: 3
          
  - platform: template
    name: LED_Controller_Power_25
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '010000000100010100001001'
          protocol: 
            pulse_length: 396
            sync: [1,31]
            zero: [1,3]
            one: [3,1]
            inverted: false
          repeat: 3
```

**Description of Resolution:**
Not sure if specifying the values for sync,zero,one, inverted made it work because the defaults weren't 'kicking in'. Or whether the repeat 3 is what did it. Not sure but if I had to make an educated guess my bug is somehow related to [this](https://github.com/esphome/issues/issues/120)

OttoWinter, 2019-07-10:
@mikekay1 Please try with only the repeat set to three - and none of the other changes.

As described in the issue you linked to, ESPHome only sends the code once by default, whereas RCSwitch repeats it multiple times. If that is the issue the docs could be updated to add that repeat should be set to >1. Then it would also be a docs issue, not per se a bug.

mikekay1, 2019-07-10:
That makes perfect sense -- I have set the repeat to 3, and will remove the other defaults. I didn't know that RC-Switch in PlatformIO automatically repeats the signal. 

On that note if the above is correct and it is the repeat >1 do you think it would benefit to make that the default 3 [unless otherwise specified] maybe alot of people dont know that why they never end up sending the signal. I fail to see a scenario where one would want it to only send once.
**OR**
Like you say update the docs with the RF example to include the repeat option to let people know.

Finally: In either scenario I think it would be nice if esphome could do what RC switch does specify a pulse_length but default to the other protocol settings. As the settings are somewhat burried in the source of rcswitch.cpp.
```
mySwitch.setProtocol(1, 396); 
```

Agreed I dont think this is a bug, I will post back my findings and comment back.

glmnet, 2019-07-10:
Curiously I just [sent a pr](https://github.com/esphome/esphome-docs/pull/290/files) for adding some info on docs based on my first experiences with 433mhz rc switch

Maybe a good addition would be adding in the dump logs the delay time between codes decoded. I changed the delay between repetitions from the default to a much shorter, after seeing that messages where coming quickly when using the remote control instead of the esphome transmitter with a repeat of e.g. 10 times.  However I did not measure in any manner the actual delay of the remote transmitter, esphome log timestamps are in second resolution which does not allow for an easy estimation.

BertrumUK, 2019-11-22:
I have been trying to get this to work for a few days now and I cannot seem to get anything transmitted via 433Mhz. 

I have tried using an ESP01 and a NodeMCU but both exhibit the same issue. I have swapped GPIO's over but still the same. The Rx sniffer on ESPHome works fine just not the Tx. With both connected at the same time the ESPhome Rxer also picks nothing up.

I currently use OMG for 433Mhz Transmit and Receive so I know that my transmitter is working. I can monitor the 433Mhz rx using MQTT as I use if for a motion sensor and my doorbell so it will show if there is any 433Mhz transmission taking place but I get nothing.

I have tried different configs including the one marked "Resolution" from MikeKay1 above but still nothing. The logs show the command being sent but nothing is Rxed 

```
[09:55:03][D][switch:021]: 'esp on' Turning ON.
[09:55:03][D][remote_transmitter:066]: Sending remote code...
[09:55:03][D][switch:021]: 'esp off' Turning ON.
[09:55:03][D][remote_transmitter:066]: Sending remote code...
```

Here is my config 

```
esphome:
  name: esp_433mhz
  platform: ESP8266
  board: nodemcuv2 # for NodeMCU
  #board: esp01_1m for ESP01

wifi:
  ssid: 'SSID'
  password: !secret wifipw
  manual_ip:
    static_ip: 192.168.0.130
    gateway: 192.168.0.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp 433Mhz Fallback Hotspot"
    password: !secret wifipw

captive_portal:

# Enable logging
logger:
  level: debug

# Enable Home Assistant API
api:

ota:

debug:

remote_receiver:
  pin: D2
  dump:
    - rc_switch
  # Settings to optimize recognition of RF devices
  tolerance: 50%
  filter: 250us
  idle: 4ms
  buffer_size: 2kb

remote_transmitter:
  pin: D1
  # RF uses a 100% carrier signal
  carrier_duty_percent: 100%
  
switch:

  - platform: template
    name: esp on
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '10000000100010100000001'
          protocol: 
            pulse_length: 396
            sync: [1,31]
            zero: [1,3]
            one: [3,1]
            inverted: false
          repeat: 3
  - platform: template
    name: esp off
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '010101010001111100110110'
          protocol: 
            pulse_length: 306
            inverted: false
          repeat: 3
```

closed 2020-12-05 by stale[bot]