esphome/issues #510: Fastled not clearing previous color

ministryofsillywalks, 2019-07-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**Affected component:**

FastLed

**Description of problem:**
When switching to an effect the previous settings don't get cleared.
For example I have an effect which turns the first 20 Leds one color and doesn't do anything to the others. These then retain their color from the previous settings.
This isn't too bad as I can just turn them off in the effect.
I feel like fastled is missing the option
FastLED.clear();

However when I turn off the light coming from an effect it reverts to the previous static color for a second and then turns off.
For example if I come from a static white light and switch to rainbow then that works.
Turning off the light from rainbow has the leds flash white for one second and only then turn off.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```
esphome:
  name: badezimmer
  platform: ESP8266
  board: nodemcuv2

wifi:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

switch:
  - platform: gpio
    name: "Badezimmer Licht"
    pin: D0
    
light:
  - platform: fastled_spi
    chipset: WS2801
    data_pin: D5
    clock_pin: D6
    num_leds: 60
    rgb_order: RGB
    name: "FastLED WS2801 Light"
    effects:
      - addressable_rainbow:
          name: Rainbow Effect With Custom Values
          speed: 10
          width: 50
      - addressable_lambda:
          name: "NightLight"
          update_interval: 16ms
          lambda: |-
            it.range(0, 20) = light::ESPColor(255, 0, 18);
            it.range(21, 60) = light::ESPColor(0, 0, 0);

      - addressable_lambda:
         
          name: "Christmas RedGreen"
          lambda: |-
       
            for (int i = 1; i <  it.size(); i+=2) {
            it[i] = light::ESPColor(255, 0, 18);
            }
            
            for (int i = 0; i <  it.size(); i+=2) {
            it[i] = light::ESPColor(0, 0, 0);
            }
**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

ministryofsillywalks, 2019-07-10:
I have the same thing happen on ws2811.
Made a short video to showcase the issue.
https://gfycat.com/sandymintyhumpbackwhale
I start off with red as static color.
Then change the effect to random_twinkle.
Then turn the light off.
As you can see after the random_twinkle effect the full red flashes up again before turning off.
Also here is the logs:
`[18:32:52][D][light:261]: 'FastLED WS2811 Light' Setting:
[18:32:52][D][light:270]:   State: ON
[18:32:52][D][light:300]:   Transition Length: 1.0s
[18:32:55][D][light:261]: 'FastLED WS2811 Light' Setting:
[18:32:55][D][light:323]:   Effect: 'Random Twinkle'
[18:33:00][D][light:261]: 'FastLED WS2811 Light' Setting:
[18:33:00][D][light:270]:   State: OFF
[18:33:00][D][light:300]:   Transition Length: 1.0s
[18:33:00][D][light:306]:   Effect: 'None'`

Hope someone has an idea whats going on, as this is sadly quite a dealbreaker for the twinkling nightlight im building for my little one ;)

brandond, 2019-07-10:
@ministryofsillywalks this does seem broken, but have you tried setting the transition length to 0 on the OFF state? I'm guessing something's going on with the twinkle ending immediately, so it goes back to ON (solid red) for the duration of the transition to OFF.

OttoWinter, 2019-07-10:
Ah yes that's due to how the light base works - and I don't see a nice solution other than adding a hack for "if light is turning off, don't reset color" - but that's probably what will end up having to be done.

Workaround is using 0 transition length as @brandond said.

ministryofsillywalks, 2019-07-11:
Thanks for the input!
Looking at the documentation I can set transition to 0 as a global or for an effect only. Right? 
Is there a way to just set it to 0 for the "off" transition from effects? 
Having the light gradually dimming on and dimming off from a static color is quite nice and it would be a shame to ruin that with 0s transitions on everything...

ministryofsillywalks, 2019-07-11:
Setting global transitions to 0s did indeed solve the issue of the light flashing when turning off.
would of course be great to be able to continue using a gentle fade in and out but this will work for the time being.

closed 2020-12-05 by stale[bot]