esphome/issues #511: BME280 module does not collect dewpoint

BrettDikeman, 2019-07-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Not applicable

**ESP (ESP32/ESP8266, Board/Sonoff):**

Not applicable

**Affected component:**

BME280 (and BME680)

**Description of problem:**

The BME sensors with temperature and humidity provide dew point values, but these are not available in the corresponding component.

**Problem-relevant YAML-configuration entries:**

N/A

**Logs (if applicable):**

**Additional information and things you've tried:**

N/A

brandond, 2019-07-09:
I don't believe either device actually provides this as a raw measurement. Other firmwares will derive it for you based on temperature and humidity but I imagine @ottowinter will tell you to do this for yourself with a template sensor lambda.

Example: https://github.com/nodemcu/nodemcu-firmware/blob/master/app/modules/bme280.c#L457

BrettDikeman, 2019-07-09:
Yep, not a single mention of the word "dew" in the datasheet. Good job, me.

\*_facepalm_\*

I guess if anything this would end up being a feature request for the module to provide. Closing...

closed 2019-07-09 by BrettDikeman