esphome/issues #512: esp32 lolin freeze and hang without hass, no automation is working

afefelov, 2019-07-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

LOLIN32

Core

I'm using esphome with esp32 lolin for greenhouse automation, I've updated to latest version 2 week ago and today faced with strange situation: in case if hass is not available for several hours, esp32 completely lost even wifi connection. From documentation it seems kind of reason for intermal wdt for reset, but still strange for me, cause generally all the internal automation should perform even without wifi connection and it is at least at start when wifi is starting to connect.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: greenhouse
  platform: ESP32
  board: lolin32

wifi:
  ssid: 'f1'
  password: '*'

logger:

api:

ota:

esp32_ble_tracker:
  scan_interval: 300s
  
switch:
  - platform: gpio
    pin: GPIO25
    name: "Cucumber door (+)"
    id: sw_cucumber_pwr
    restore_mode: ALWAYS_OFF
    inverted: yes
  - platform: gpio
    pin: GPIO26
    name: "Cucumber door (-)"
    id: sw_cucumber_gnd
    restore_mode: ALWAYS_OFF
    inverted: yes
  - platform: gpio
    pin: GPIO27
    name: "Tomato door (+)"
    id: sw_tomato_pwr
    restore_mode: ALWAYS_OFF
    inverted: yes
  - platform: gpio
    pin: GPIO14
    name: "Tomato door (-)"
    id: sw_tomato_gnd
    restore_mode: ALWAYS_OFF
    inverted: yes
  - platform: gpio
    pin: GPIO16
    name: "Water (+)"
    id: sw_water_tomato_pwr
    restore_mode: ALWAYS_OFF
    inverted: yes
  - platform: gpio
    pin: GPIO4
    name: "Water (-)"
    id: sw_water_tomato_gnd
    restore_mode: ALWAYS_OFF
    inverted: yes

dallas:
  - pin: GPIO13
    update_interval: 10s

sensor:
  - platform: xiaomi_miflora
    mac_address: 'C4:7C:8D:60:E4:7E'
    temperature:
      name: "Xiaomi MiFlora Temperature"
    moisture:
      name: "Xiaomi MiFlora Moisture"
    illuminance:
      name: "Xiaomi MiFlora Illuminance"
    conductivity:
      name: "Xiaomi MiFlora Soil Conductivity"
    battery_level:
      name: "Xiaomi MiFlora Battery Level"

  - platform: dallas
    address: 0xF50417C1DEDDFF28
    name: "t. tomato"
    id: temp_tomato
    filters:
      - sliding_window_moving_average:
          window_size: 30
          send_every: 30
    on_value:
      then:
        - if:
            condition:
              lambda: 'return id(temp_tomato).state > 20;'
            then:
              - switch.turn_off: sw_tomato_pwr
              - switch.turn_on: sw_tomato_gnd
              - delay: 2s
              - switch.turn_off: sw_tomato_pwr
              - switch.turn_off: sw_tomato_gnd
            else:
              - switch.turn_on: sw_tomato_pwr
              - switch.turn_off: sw_tomato_gnd
              - delay: 25s
              - switch.turn_off: sw_tomato_pwr
              - switch.turn_off: sw_tomato_gnd
  - platform: dallas
    address: 0xB30417C1B5C5FF28
    name: "t. cucumber"
    id: temp_cucumber
    filters:
      - sliding_window_moving_average:
          window_size: 30
          send_every: 30
    on_value:
      then:
        - if:
            condition:
              lambda: 'return id(temp_cucumber).state > 25;'
            then:
              - switch.turn_off: sw_cucumber_pwr
              - switch.turn_on: sw_cucumber_gnd
              - delay: 2s
              - switch.turn_off: sw_cucumber_pwr
              - switch.turn_off: sw_cucumber_gnd
            else:
              - switch.turn_on: sw_cucumber_pwr
              - switch.turn_off: sw_cucumber_gnd
              - delay: 25s
              - switch.turn_off: sw_cucumber_pwr
              - switch.turn_off: sw_cucumber_gnd
  
binary_sensor:
  - platform: gpio
    name: "Tomato door sw"
    pin:
      number: GPIO19
      inverted: True
      mode: INPUT_PULLUP
    on_state:
      then:
        - switch.turn_off: sw_tomato_pwr
        - switch.turn_on: sw_tomato_gnd
        - delay: 25s
        - switch.turn_off: sw_tomato_pwr
        - switch.turn_off: sw_tomato_gnd
  - platform: gpio
    name: "Cucumber door sw"
    pin:
      number: GPIO23
      inverted: True
      mode: INPUT_PULLUP
    on_state:
      then:
        - switch.turn_off: sw_cucumber_pwr
        - switch.turn_on: sw_cucumber_gnd
        - delay: 25s
        - switch.turn_off: sw_cucumber_pwr
        - switch.turn_off: sw_cucumber_gnd
i2c:

display:
  - platform: lcd_pcf8574
    dimensions: 16x2
    address: 0x27
    lambda: |-
      it.printf("Temp: %2.0f", id(temp_tomato).state);
      it.strftime("It is %H:%M on %d.%m.%Y", id(gps_time).now());

uart:
  rx_pin: GPIO17
  baud_rate: 9600

# gps:

time:
  - platform: homeassistant
    id: gps_time
    on_time:
      - hours: 17
        seconds: 0
        minutes: /5
        then:
          - switch.turn_on: sw_water_tomato_pwr
          - switch.turn_off: sw_water_tomato_gnd
      - hours: 18
        seconds: 0
        minutes: /5      
        then:
          - switch.turn_off: sw_water_tomato_pwr
          - switch.turn_on: sw_water_tomato_gnd
      - hours: 19
        seconds: 0
        minutes: /5
        then:
          - switch.turn_off: sw_water_tomato_pwr
          - switch.turn_off: sw_water_tomato_gnd

web_server:
  port: 80

```

**Logs (if applicable):**

```
No logs collected before freeze

```

OttoWinter, 2019-07-11:
> From documentation it seems kind of reason for intermal wdt for reset, but still strange for me

Where do you get this info from? The logs?

> generally all the internal automation should perform even without wifi connection and it is at least at start when wifi is starting to connect.

They will, unless the device crashes of course. If your device crashes the automations will stop.

Please attach logs (taken by connecting the device via USB and viewing logs) to the post.

afefelov, 2019-07-11:
Should I connect device via usb and wait till crush?

closed 2019-11-01 by no-response[bot]