esphome/issues #515: `reboot_timeout: 0s` does not disable it

chriscrowe, 2019-07-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

- ESPHome v 1.13.6 running on HASS v 0.95.4
- HASS running in Ubuntu VM on Synology DS918+

**ESP (ESP32/ESP8266, Board/Sonoff):**

- Teckin SP20 smart plug running ESPHome v 1.13.6

**Affected component:**

https://esphome.io/components/wifi.html

**Description of problem:**

The docs cite the following about `reboot_timeout`:

- **reboot_timeout** (*Optional*, :ref:`time <config-time>`): The amount of time to wait before rebooting when no
  WiFi connection exists. Can be disabled by setting this to ``0s``, but note that the low level IP stack currently
  seems to have issues with WiFi where a full reboot is required to get the interface back working. Defaults to ``5min``.

I have my configs set to `reboot_timeout: 0s` but my switches still reset themselves when wifi is lost for some relatively short period of time (seems like 5 minutes or less). This is particularly problematic for the Teckin SP20 switches because a power cycle actually toggles the relay very briefly (I think it's an NO relay).

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
wifi: !include common/wifi.yaml
```

common/wifi.yaml

```yaml
  ssid: !secret wifi_ssid
  password: !secret wifi_pass
  reboot_timeout: 0s
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

brandond, 2019-07-12:
You didn't include your full config so I can't tell what you're actually using, but you should be aware that both the native api and mqtt components have reboot_timeout settings also. Have you disabled those as well?

chriscrowe, 2019-07-12:
Oof, I'm definitely not using the API's `reboot_timeout` directive, didn't know there was a separate one. I'm sure that's my issue here, will try it.

anusu90, 2019-09-18:
i have put following code:
api:
  reboot_timeout: 3600s 

This should make the reboot time gap to 1 hour right? It still doesn't work. 
1. I still don't know why my esp8266 keeps on disconnecting even when the router is in the next room
2. when it disconnects it reboots
3. it doesn't restore the states

I have esphome 1.13.6. Running HA on raspberry pi 3B+

glmnet, 2019-09-19:
Post full yaml file.

brandond, 2019-09-19:
Did you read the bit above where we discussed the fact that there are multiple reboot timeouts?

anusu90, 2019-09-19:
`esphome:
  name: ranchi3
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "######"
  password: "########"
  manual_ip:
    static_ip: 192.168.0.55 
    gateway: 192.168.0.1
    subnet: 255.255.255.0
    dns1: 192.168.0.1
    dns2: 192.168.0.1
  reboot_timeout: 3600s
    
switch:
  - platform: gpio
    pin: GPIO04
    name: "srrr_1"
    inverted: true
    restore_mode: RESTORE_DEFAULT_ON
    
  - platform: gpio
    pin: GPIO05
    name: "srrr_2"
    inverted: true
    restore_mode: RESTORE_DEFAULT_ON
    
  - platform: gpio
    pin: GPIO12
    name: "srrr_3"
    inverted: true
    restore_mode: RESTORE_DEFAULT_ON

  - platform: gpio
    pin: GPIO13
    name: "srrr_4"
    inverted: true
    restore_mode: RESTORE_DEFAULT_ON

# Enable logging
logger:

# Enable Home Assistant API
api:
  reboot_timeout: 3600s

ota:`

I did not. I am sorry. I have updated it. This should do it?

anusu90, 2019-09-19:
Update: it did not fix it:
[image]
The above pic above shows it disconnects. I am pinging it through my Mac. The following pic shows it becomes unavailable in Hassio and reboots, even after making relevant changes in my configuration file.

[image]

anusu90, 2019-09-19:
The frequency is also very high: as evident below.

[image]

please help

anusu90, 2019-09-19:
CONTD.

i ssh (ed?) into my raspberry pi using my Mac. And from there on i am pinging my node. I don't know what this actually will do. Maybe some insight

[image]

brandond, 2019-09-20:
How can you tell from the dropped pings that it is rebooting? Have you tried collecting logs via the UART? I can't tell from what you've shared if it is crashing, soft restarting, or just dropping off the wifi.

anusu90, 2019-09-20:
I am sorry but I dont know how to do that. From what I understand i should plug my node to my mac start Arduino IDE and open the serial monitor, see the logs and post it here. Before that I should put the following in my yaml file.

`logger:
  level: VERBOSE`

Am I on the right track?

glmnet, 2019-09-20:
Ok your symptoms points to a wifi issue, but we need to be sure

`restore_mode: RESTORE_DEFAULT_ON` should turn them on only if it does not know how they were previously, now I'm not 100% sure but a soft crash reboot, should keep the RTC memory and restore with previous value, so if this is the case this might be caused by an ugly brown out.

Can you tell us more differences? do you think there are more walls / furniture / [large water surfaces](https://news.mit.edu/2018/wireless-communication-through-water-air-0822) between the router and the node than in your old house? Can you test the node in your old house? Well may be that won't be handy :)

You're right about setting up the logger, proper syntax is:

```yaml
logger:
  level: VERBOSE
```

Now I have never tested this but in your case it will be worth trying the [debug component](https://esphome.io/components/debug.html) which should tell you the latest reboot reason (It will be great if it can tell without need to USB to it but I don't know)

anusu90, 2019-09-21:
The old house is more than 1000 KMs away so I can't test it there :)

I am also thinking that it could be a WIFI issue. My old house was a 2 bedroom apartment so I was using the router provided by the ISP with my raspberry pi connected to the router via lan cable. This house is two storied and following is my network setup. Since broadband connectivity in this area was a little unreliable I am using a load balancer.

Apart from that there not much difference in set up. Node is in next room from where router is (that is how it was in old house). While coming my luggage did under go x-ray scan at the airport. But I don't think that could be the reason.

[image]

Anyways I am going to check the logs via my MAC in couple of days.

anusu90, 2019-09-21:
I just enabled the debug component and put the log to verbose mode and flashed the esp over WIFI.
Following the the log out put just after flashing

[image]

glmnet, 2019-09-21:
Good, next step is waiting for it to restart? and connect to the logger again to see what it says.
I forgot to mention another useful thing is the [uptime sensor](https://esphome.io/components/sensor/uptime.html) so you know it restarted.

anusu90, 2019-09-24:
It's been more than 48 hours. The node is connected to my MAC via USB and it has not restarted once. Though this has happened before also when it did not restart for couple days, I am hoping that the problem was with my power supply. I was using a SAMSUNG USB adapter (5V / 2A). 
I will update accordingly.

glmnet, 2019-09-24:
Ok computer usb ports usually have good voltage / current supply. However 2A should be more than enough for most stuff, however e.g. a high torque servo will easily draw all that power.
By moving the node to your computer did you unplug anything from the ESP?
What devices are usually connected to the ESP? any chance there is a short circuit? a solder that looks ok but it is not? a bad male to male dupont cable? (those are very unreliable)

anusu90, 2019-09-24:
No nothing has changed. There are 4 relays that I have connected to node, same circuit that I was using in the old house. 

I have already checked for short circuit as well. It seems fine.

anusu90, 2019-09-29:
UPDATE

I got a 5V 2A usb charger from Amazon and replaced the Samsung one. It's been 3 days. Everything seems fine. No crashing/ rebooting/ dropping WIFI yet.

OttoWinter, 2019-10-20:
Closing as it works now

closed 2019-10-20 by OttoWinter