esphome/issues #516: Custom components doesn’t create mqtt elements

ezar, 2019-07-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP 01S
**Affected component:**

- custom_component
- mqtt

**Description of problem:**
I define several sensors, switches and binary sensors using custom components, but It doesn’t create mqtt elements using discovery at hassio. 

Other sensors defined inside `sensor:` group works. 

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: roomba
  platform: ESP8266
  board: esp01_1m
  includes:
    - includes/ESPHomeRoombaComponent.h
  libraries:
    - file:///config/esphome/includes/Roomba-1.4.zip

wifi:
  ssid: 'xxxxxxx’
  password: 'secret’

mqtt:
  broker: '192.168.1.1’
  username: 'secret’
  password: 'secret'
  discovery: true

logger:
  level: debug

ota:

custom_component:
 - lambda: |-
    auto custom_roomba = new RoombaComponent(
        "Roomba Distance",
        "Roomba Voltage",
        "Roomba Current",
        "Roomba Charge",
        "Roomba Capacity",
        "Roomba Charging",
        "Roomba Docked",
        "Roomba Cleaning",
        "Roomba On",
        "Roomba Off",
        "Roomba Dock",
        "Roomba Locate",
        "Roomba Spot Clean",
        "roomba/command", /* command_topic */
        2,                /* BRC Pin */
        5000              /* Update Interval */
      );
    App.register_component(custom_roomba);
    App.register_sensor(custom_roomba->distance_sensor);
    App.register_sensor(custom_roomba->voltage_sensor);
    App.register_sensor(custom_roomba->current_sensor);
    App.register_sensor(custom_roomba->charge_sensor);
    App.register_sensor(custom_roomba->capacity_sensor);
    App.register_binary_sensor(custom_roomba->chargingState_sensor);
    App.register_binary_sensor(custom_roomba->dockedState_sensor);
    App.register_binary_sensor(custom_roomba->cleaningState_sensor);
    App.register_switch(custom_roomba->on_switch);
    App.register_switch(custom_roomba->off_switch);
    App.register_switch(custom_roomba->dock_switch);
    App.register_switch(custom_roomba->locate_switch);
    App.register_switch(custom_roomba->spot_switch);
    return {custom_roomba};
    
switch:

sensor:
  - platform: uptime
    name: "Roomba Uptime Sensor"
  - platform: wifi_signal
    name: "Roomba WiFi Signal Sensor"
    update_interval: 30s

binary_sensor:
  - platform: status
    name: "Roomba Status"
```

[ESPHomeRoombaComponent.h Library](https://raw.githubusercontent.com/mannkind/ESPHomeRoombaComponent/master/ESPHomeRoombaComponent.h)

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

ezar, 2019-07-19:
Nobody have same problem? :(

OttoWinter, 2019-07-21:
That's correct - `register_sensor` etc won't create the associated MQTT etc elements starting some versions back. That's why ESPHome has the `sensor.custom` platforms: Those *will* do so. Please use the `sensor.custom` platform in yaml to register them.

closed 2019-07-21 by OttoWinter

ezar, 2019-07-22:
Using sensor.custom, Can I create switch and binary_sensor?
Regards,

ezar, 2019-07-22:
OK. I see a `switch.custom` and `binary_sensor.custom`
Regards,

ezar, 2019-07-22:
@OttoWinter It's possible to create a custom switch with serveral switches?
https://esphome.io/components/switch/custom.html

We make a similar aproximation: https://esphome.io/components/sensor/custom.html#bonus-sensors-with-multiple-output-values

Regards,

OttoWinter, 2019-10-13:
Yes.