esphome/issues #518: Compilation failed on fresh hassbian install

radunCZ, 2019-07-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** pip

fresh install of hassbian, fully updated OS (apt-get update / apt-get upgrade....)

```
root@hassbian:/home/homeassistant/esphome# pio --version
PlatformIO, version 4.0.0
root@hassbian:/home/homeassistant/esphome# python --version
Python 3.7.3
root@hassbian:/home/homeassistant/esphome# esphome config/sonoff_basic1_test version
Version: 1.13.6
```

EPSHome installed via _pip install esphome_ 

**ESP (ESP32/ESP8266, Board/Sonoff):** sonoff basic

**Affected component:** N/A

**Description of problem:**
I'm not able to finish the compile process on this environment 

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: sonoff_basic1_test
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "projectwifi"
  password: "wifipassword"

api:

logger:

ota:

web_server:
  port: 80

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Sonoff Basic Test01 - Button"
    on_press:
      - switch.toggle: relay

switch:
  - platform: gpio
    name: "Sonoff Basic Test01 - Sonoff Basic Relay"
    pin: GPIO12
    id: relay

status_led:
  pin:
    number: GPIO13
    inverted: yes
```

**Logs (if applicable):**

```
root@hassbian:/home/homeassistant/esphome# esphome config/sonoff_basic1_test.yaml compile
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d config/sonoff_basic1_test
Processing sonoff_basic1_test (board: esp01_1m; framework: arduino; platform: espressif8266@1.8.0)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 1.8.0 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES: toolchain-xtensa 1.40802.2 (4.8.2), framework-arduinoespressif8266 2.20402.4 (2.4.2), tool-esptool 1.413.0 (4.13)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Looking for Update library in registry
Warning! Library `{'name': 'Update', 'requirements': None}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'name': 'Update', 'requirements': None}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP Async WebServer> 1.1.1
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/sonoff_basic1_test/src/esphome/components/api/api_message.cpp.o
Compiling .pioenvs/sonoff_basic1_test/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/sonoff_basic1_test/src/esphome/components/api/basic_messages.cpp.o
Compiling .pioenvs/sonoff_basic1_test/src/esphome/components/api/command_messages.cpp.o
as: loadlocale.c:130: _nl_intern_locale_data: Assertion `cnt < (sizeof (_nl_value_type_LC_TIME) / sizeof (_nl_value_type_LC_TIME[0]))' failed.
xtensa-lx106-elf-g++: internal compiler error: Aborted (program as)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [.pioenvs/sonoff_basic1_test/src/esphome/components/api/api_message.cpp.o] Error 4
as: loadlocale.c:130: _nl_intern_locale_data: Assertion `cnt < (sizeof (_nl_value_type_LC_TIME) / sizeof (_nl_value_type_LC_TIME[0]))' failed.
xtensa-lx106-elf-g++: internal compiler error: Aborted (program as)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [.pioenvs/sonoff_basic1_test/src/esphome/components/api/command_messages.cpp.o] Error 4
as: loadlocale.c:130: _nl_intern_locale_data: Assertion `cnt < (sizeof (_nl_value_type_LC_TIME) / sizeof (_nl_value_type_LC_TIME[0]))' failed.
xtensa-lx106-elf-g++: internal compiler error: Aborted (program as)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [.pioenvs/sonoff_basic1_test/src/esphome/components/api/basic_messages.cpp.o] Error 4
as: loadlocale.c:130: _nl_intern_locale_data: Assertion `cnt < (sizeof (_nl_value_type_LC_TIME) / sizeof (_nl_value_type_LC_TIME[0]))' failed.
xtensa-lx106-elf-g++: internal compiler error: Aborted (program as)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [.pioenvs/sonoff_basic1_test/src/esphome/components/api/api_server.cpp.o] Error 4
================== [ERROR] Took 8.92 seconds ==================
```

**Additional information and things you've tried:**

I've tried to install the ESPHome within Python virtualenv with Python 2.7 and latest Platformio 3.x, but the result was same.

Same config wortks fine on hass.io with esphome addon v1.12.2 (and it's more or less the sample config for Sonoff Basic anyways)

glmnet, 2019-07-15:
PlatformIO 4 / Python 3 is not supported. 
Please upload the logs you get when building in PlatformIO 3 / Python 2

radunCZ, 2019-07-15:
unsupported, but installed as default...

```
root@hassbian:/home/homeassistant# mkdir epshome_python2
root@hassbian:/home/homeassistant# virtualenv -p /usr/bin/python2.7  /home/homeassistant/epshome_python2/
root@hassbian:/home/homeassistant# source /home/homeassistant/epshome_python2/bin/activate
(epshome_python2) root@hassbian:/home/homeassistant# python --version
Python 2.7.16

(epshome_python2) root@hassbian:/home/homeassistant# pip install 'platformio<4.0.0'
....
(epshome_python2) root@hassbian:/home/homeassistant# pio --version
PlatformIO, version 3.6.7
(epshome_python2) root@hassbian:/home/homeassistant# pip install esphome
(epshome_python2) root@hassbian:/home/homeassistant# esphome /home/homeassistant/esphome/config/sonoff_basic1_test version
Version: 1.13.6
```

versions are fine, I hope

```
(epshome_python2) root@hassbian:/home/homeassistant# esphome /home/homeassistant/esphome/config/sonoff_basic1_test.yaml compile
INFO Reading configuration...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Deleting /home/homeassistant/esphome/config/sonoff_basic1_test/.pioenvs
INFO Deleting /home/homeassistant/esphome/config/sonoff_basic1_test/.piolibdeps
INFO Compiling app...
INFO Running:  platformio run -d /home/homeassistant/esphome/config/sonoff_basic1_test
***************************************************************************************************************************
Obsolete PIO Core v3.6.7 is used (previous was 4.0.0)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
***************************************************************************************************************************
Processing sonoff_basic1_test (framework: arduino; platform: espressif8266@1.8.0; board: esp01_1m)
---------------------------------------------------------------------------------------------------------------------------
LibraryManager: Installing id=3837 @ 5.13.3
Unpacking  [####################################]  100%
LibraryManager: Installing id=306 @ 1.1.1
Unpacking  [####################################]  100%
LibraryManager: Installing id=305
Unpacking  [####################################]  100%
LibraryManager: Installing id=1826
Unpacking  [####################################]  100%
Warning! Library `{'requirements': None, 'name': 'Update'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'Update'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
CorePackageManager: Installing tool-scons @ ~2.20501.7
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
tool-scons @ 2.20501.7 has been successfully installed!
Uninstalling tool-scons @ 3.30005.0:    [OK]
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz 80KB RAM (1MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP Async WebServer> 1.1.1
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/sonoff_basic1_test/src/esphome/components/api/api_message.cpp.o
Compiling .pioenvs/sonoff_basic1_test/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/sonoff_basic1_test/src/esphome/components/api/basic_messages.cpp.o
Compiling .pioenvs/sonoff_basic1_test/src/esphome/components/api/command_messages.cpp.o
as: loadlocale.c:130: _nl_intern_locale_data: Assertion `cnt < (sizeof (_nl_value_type_LC_TIME) / sizeof (_nl_value_type_LC_TIME[0]))' failed.
xtensa-lx106-elf-g++: internal compiler error: Aborted (program as)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [.pioenvs/sonoff_basic1_test/src/esphome/components/api/api_message.cpp.o] Error 4
as: loadlocale.c:130: _nl_intern_locale_data: Assertion `cnt < (sizeof (_nl_value_type_LC_TIME) / sizeof (_nl_value_type_LC_TIME[0]))' failed.
xtensa-lx106-elf-g++: internal compiler error: Aborted (program as)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [.pioenvs/sonoff_basic1_test/src/esphome/components/api/command_messages.cpp.o] Error 4
as: loadlocale.c:130: _nl_intern_locale_data: Assertion `cnt < (sizeof (_nl_value_type_LC_TIME) / sizeof (_nl_value_type_LC_TIME[0]))' failed.
xtensa-lx106-elf-g++: internal compiler error: Aborted (program as)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [.pioenvs/sonoff_basic1_test/src/esphome/components/api/basic_messages.cpp.o] Error 4
as: loadlocale.c:130: _nl_intern_locale_data: Assertion `cnt < (sizeof (_nl_value_type_LC_TIME) / sizeof (_nl_value_type_LC_TIME[0]))' failed.
xtensa-lx106-elf-g++: internal compiler error: Aborted (program as)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [.pioenvs/sonoff_basic1_test/src/esphome/components/api/api_server.cpp.o] Error 4
============================ [ERROR] Took 20.21 seconds ============================

```

radunCZ, 2019-07-15:
I have uninstalled also the platformio outside the virtual environment, but the result is the same

glmnet, 2019-07-15:
Great, your issue is not very specific to ESPHome.
Searching for `cnt < (sizeof (_nl_value_type_LC_TIME) / sizeof (_nl_value_type_LC_TIME[0]))` gives some interesting results

Eg. try some of these dpkg-recongure ... https://unix.stackexchange.com/questions/269159/problem-of-cant-set-locale-make-sure-lc-and-lang-are-correct

radunCZ, 2019-07-15:
kind of black magic :)

this worked:
[https://unix.stackexchange.com/questions/444102/loadlocale-c-nl-intern-locale-data-assertion-error](https://unix.stackexchange.com/questions/444102/loadlocale-c-nl-intern-locale-data-assertion-error)

```
(epshome_python2) root@hassbian:/home/homeassistant# LANG=/usr/lib/locale/en_US
(epshome_python2) root@hassbian:/home/homeassistant# locale-gen
Generating locales (this might take a while)...
  cs_CZ.ISO-8859-2... done
  cs_CZ.UTF-8... done
  en_GB.ISO-8859-1... done
  en_GB.ISO-8859-15... done
  en_GB.UTF-8... done
Generation complete.
```

```
(epshome_python2) root@hassbian:/home/homeassistant# esphome /home/homeassistant/esphome/config/sonoff_basic1_test.yaml compile
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /home/homeassistant/esphome/config/sonoff_basic1_test
...
...
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [====      ]  43.1% (used 35304 bytes from 81920 bytes)
PROGRAM: [====      ]  35.5% (used 363740 bytes from 1023984 bytes)
=============== [SUCCESS] Took 87.91 seconds ===============

***************************************************
There is a new version 4.0.0 of PlatformIO available.
Please upgrade it via `platformio upgrade` or `pip install -U platformio` command.
Changes: https://docs.platformio.org/en/latest/history.html
***************************************************

***************************************************
There are the new updates for platforms (espressif8266)
Please update them via `platformio platform update` command.

If you want to manually check for the new versions without updating, please use `platformio platform update --only-check` command.
***************************************************

INFO Successfully compiled program.

```

closed 2019-07-15 by radunCZ