esphome/issues #519: Library `{'requirements': None, 'name': 'Hash'}` has not been found in PlatformIO Registry.

richard-scott, 2019-07-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

DOIT ESP32 DEVKIT V1
esphome 1.14.0-dev

**Affected component:**

Compiling binary prior to flashing.

**Description of problem:**

Compile routine is looking for library it can't find.

**Problem-relevant YAML-configuration entries:**
```yaml
Not a YAML issue.

```

**Logs (if applicable):**

```
Processing bluetooth_mqtt (platform: espressif32@1.6.0; framework: arduino; board: esp32doit-devkit-v1)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 1.6.0 > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10001.190111 (1.0.1), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Looking for Hash library in registry
Warning! Library `{'requirements': None, 'name': 'Hash'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'Hash'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AsyncTCP> 1.0.3
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <AsyncMqttClient> 0.8.2
|   |-- <AsyncTCP> 1.0.3
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <Update> 1.0
|-- <WiFi> 1.0

```

**Additional information and things you've tried:**

brandond, 2019-07-15:
> You can ignore this message, if `{'requirements': None, 'name': 'Hash'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.

This is normal. Is your compile failing, or you just don't like seeing the warning?

richard-scott, 2019-07-16:
It is a new warning since the add-on updated PlatformIO to v4. I figured it may have been a typo or some issue with the upgrade configuration... Other than that things work compile OK.

closed 2019-08-29 by richard-scott