esphome/issues #520: AsyncMqttClient.cpp compile warning

richard-scott, 2019-07-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io 

**ESP (ESP32/ESP8266, Board/Sonoff):**

DOIT ESP32 DEVKIT V1
esphome 1.14.0-dev

**Affected component:**

Compiling binary prior to flashing.

**Description of problem:**

Warning during compile.

**Problem-relevant YAML-configuration entries:**
```yaml
Not a YAML issue.

```

**Logs (if applicable):**

```
Compiling /data/bluetooth_mqtt/.pioenvs/bluetooth_mqtt/lib6e6/AsyncMqttClient_ID346/AsyncMqttClient/Packets/ConnAckPacket.cpp.o
/data/bluetooth_mqtt/.piolibdeps/bluetooth_mqtt/AsyncMqttClient_ID346/src/AsyncMqttClient.cpp: In constructor 'AsyncMqttClient::AsyncMqttClient()':
/data/bluetooth_mqtt/.piolibdeps/bluetooth_mqtt/AsyncMqttClient_ID346/src/AsyncMqttClient.cpp:40:61: warning: format '%x' expects argument of type 'unsigned int', but argument 3 has type 'uint64_t {aka long long unsigned int}' [-Wformat=]
   sprintf(_generatedClientId, "esp32%06x", ESP.getEfuseMac());
                                                             ^
/data/bluetooth_mqtt/.piolibdeps/bluetooth_mqtt/AsyncMqttClient_ID346/src/AsyncMqttClient.cpp:40:61: warning: format '%x' expects argument of type 'unsigned int', but argument 3 has type 'uint64_t {aka long long unsigned int}' [-Wformat=]
Compiling /data/bluetooth_mqtt/.pioenvs/bluetooth_mqtt/lib6e6/AsyncMqttClient_ID346/AsyncMqttClient/Packets/PingRespPacket.cpp.o

```

**Additional information and things you've tried:**

wolfg1969, 2019-08-18:
I got it fixed by using
```
esphome:
  arduino_version: espressif32@>1.8.0
```

OttoWinter, 2019-08-27:
That's an issue in the AsyncMQTTClient library esphome uses. But fortunately the warning is just the compiler being overly verbose, the code will work fine. There's not really a solution to this other than forking the library (the library author hasn't released new versions in quite a while)

closed 2020-12-06 by stale[bot]