esphome/issues #521: How can I declare two UARTs in yaml?

anisart, 2019-07-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 Heltec WiFi kit 8

**Affected component:**

uart

**Description of problem:**
I need to connect two serial devices (mh-z19 and pms7003) to Heltec Wi-Fi kit 8. I tried to connect it to GPIO13 and GPIO15 separately, it works fine. But how to declare second software serial to get working two devices at the same time?

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: air_sensor
  platform: ESP8266
  board: heltec_wifi_kit_8

wifi:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

i2c:
  sda: GPIO4
  scl: GPIO5

font:
  - file: "Ubuntu-R.ttf"
    id: ubuntu_20
    size: 20

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x32"
    reset_pin: GPIO16
    address: 0x3C
    lambda: |-
      if (id(vcc_voltage).has_state()) {
        it.printf(64, 0, id(ubuntu_20), TextAlign::TOP_CENTER , "%.2fV", id(vcc_voltage).state);
      }

uart:
  tx_pin: GPIO15
  rx_pin: GPIO13
  baud_rate: 9600
  id: uart_1

sensor:
  - platform: pmsx003
    type: PMSX003
    pm_1_0:
      name: "PM <1.0µm Concentration"
    pm_2_5:
      name: "PM <2.5µm Concentration"
    pm_10_0:
      name: "PM <10.0µm Concentration"
    uart_id: uart_1
  # - platform: mhz19
  #   co2:
  #     name: "MH-Z19 CO2 Value"
  #   temperature:
  #     name: "MH-Z19 Temperature"
  #   update_interval: 60s
  #   uart_id: uart_1
  - platform: adc
    pin: VCC
    name: "VCC Voltage"
    id: vcc_voltage

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

brandond, 2019-07-15:
Just make it a list. Keep in mind that the ESP8266 has two hardware UARTs with some odd restrictions. See this for more info: https://esphome.io/components/logger.html#hardware-uarts

```yaml
uart:
  - tx_pin: 15
    rx_pin: 13
    baud_rate: 9600
    id: uart0_swap
  - tx_pin: 2
    rx_pin: None
    baud_rate: 9600
    id: uart1_txonly
```

anisart, 2019-07-15:
@brandond thanks. Seems like it works
```yaml
uart:
  - id: uart_1
    tx_pin: GPIO15
    rx_pin: GPIO13
    baud_rate: 9600
  - id: uart_2
    rx_pin: GPIO12
    baud_rate: 9600
```
but now I have a problem like in #488 :(

closed 2019-07-21 by OttoWinter