esphome/issues #522: [ESP32] Please enable CONFIG_FREERTOS_UNICORE option in menuconfig

pcmester, 2019-07-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Windows 10, esphomeflasher

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP-WROOM-32D
**Affected component:**

**Description of problem:**
After flashing, the ESP cannot boot

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
E (93) cpu_start: Running on single core chip, but application is built with dual core support. 
E (93) cpu_start: Please enable CONFIG_FREERTOS_UNICORE option in menuconfig.

```

**Additional information and things you've tried:**
I've tried to change boards, but the issue is still the same

brandond, 2019-07-15:
I think this is going to be a platformio issue. Not sure they have support for single-core boards at the moment.

OttoWinter, 2019-07-21:
Correct, this is a limitation of the current version of the arduino-espressif32 package we're using (some stuff like BLE is much less stable in some newer versions)

Also, as soon as we update, the issue will be fixed automatically, the unicore/etc stuff is handled by the platformio `board` option. Selecting the correct `board` setting will auto-set this.

closed 2019-07-21 by OttoWinter