esphome/issues #524: ZeroDivisionError: float division by zero  Error when Compiling

cassio299, 2019-07-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io 95.4
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESPHome 1.13.6
**Affected component:**

https://esphome.io/cookbook/brilliant-mirabella-genio-smart-plugs.html?highlight=smart%20plug#use-tuya-convert-to-install-esphome-firmware

**Description of problem:**

I am trying to update my smart plug (Gosund SP!-4 from Februar 2019)ESPhome version 1.2.2 to version 1.3.6 by recompile and the upload. I have deleted the old plug_03.yaml and tried to copy the code to the new created plug_03.yamel via ESPHome dashboard / edit. I am ale to create a new file with wifi via the dashboard and complete and up load it but when I edit the file with the code (that worked in 1.12.2) I get:

INFO Reading configuration...
INFO Detected timezone 'CET' with UTC offset 1 and daylight savings time from 03/29/20 02:00:00 to 10/25/20 03:00:00
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 11, in <module>
    load_entry_point('esphome', 'console_scripts', 'esphome')()
  File "/opt/esphome/esphome/__main__.py", line 483, in main
    return run_esphome(sys.argv)
  File "/opt/esphome/esphome/__main__.py", line 473, in run_esphome
    return POST_CONFIG_ACTIONS[args.command](args, config)
  File "/opt/esphome/esphome/__main__.py", line 272, in command_compile
    exit_code = write_cpp(config)
  File "/opt/esphome/esphome/__main__.py", line 147, in write_cpp
    CORE.flush_tasks()
  File "/opt/esphome/esphome/core.py", line 616, in flush_tasks
    next(task)
  File "/opt/esphome/esphome/core.py", line 421, in _wrapper_generator
    for x in var:
  File "/opt/esphome/esphome/core.py", line 421, in _wrapper_generator
    for x in var:
  File "/opt/esphome/esphome/core.py", line 421, in _wrapper_generator
    for x in var:
  File "/opt/esphome/esphome/core.py", line 421, in _wrapper_generator
    for x in var:
  File "/opt/esphome/esphome/core.py", line 421, in _wrapper_generator
    for x in var:
  File "/opt/esphome/esphome/core.py", line 421, in _wrapper_generator
    for x in var:
  File "/opt/esphome/esphome/core.py", line 421, in _wrapper_generator
    for x in var:
  File "/opt/esphome/esphome/core.py", line 421, in _wrapper_generator
    for x in var:
  File "/opt/esphome/esphome/core.py", line 417, in _wrapper_generator
    var = gen.send(var)
  File "/opt/esphome/esphome/components/sensor/__init__.py", line 193, in calibrate_linear_filter_to_code
    k, b = fit_linear(x, y)
  File "/opt/esphome/esphome/components/sensor/__init__.py", line 300, in fit_linear
    r = _correlation_coeff(x, y)
  File "/opt/esphome/esphome/components/sensor/__init__.py", line 294, in _correlation_coeff
    return s_xy / math.sqrt(s_sq_x * s_sq_y)
ZeroDivisionError: float division by zero

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

substitutions:
  plug_name: plug_03

  
  # Model Gosund SP1-4

esphome:
  name: ${plug_name}
  platform: ESP8266
  board: esp8285
  
wifi:
  ssid: "Rungstedlund"
  password: "Knygberg299"
  power_save_mode: light

logger:

api:
  password: 'Tyskland299'
ota:
  password: 'Tyskland299'

web_server:
  port: 80

time:
  - platform: homeassistant
    id: homeassistant_time
    
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO3
      inverted: True
    name: "${plug_name}_button"
    on_press:
      - switch.toggle: "${plug_name}_Relay"
    
switch:
  - platform: gpio
    name: "${plug_name}_Relay"
    id: "${plug_name}_Relay"
    pin: GPIO14
    restore_mode: ALWAYS_ON
    on_turn_on:
      - switch.turn_on: "${plug_name}_LED_Blue"
    on_turn_off:
      - switch.turn_off: "${plug_name}_LED_Blue"
  - platform: gpio
    name: "${plug_name}_LED_Blue"
    id: "${plug_name}_LED_Blue"
    pin: GPIO1
    inverted: True
    restore_mode: ALWAYS_OFF

status_led:
  pin: 
    number: GPIO13
    inverted: True

sensor:
  - platform: hlw8012
    sel_pin:
      number: GPIO12
      inverted: True
    cf_pin: GPIO04
    cf1_pin: GPIO05
    # Higher value gives lower watt readout
    current_resistor: 0.00599
    current:
      name: "${plug_name}_Current"
      unit_of_measurement: A
      filters:
        - calibrate_linear:
            # Map 0.0 (from sensor) to 0.0 (true value)
          - 0.0 -> 0.203
          - 0.0 -> 0.203
          - 0.0 -> 0.201
    voltage:
      name: "${plug_name}_Voltage"
      unit_of_measurement: V
      filters:
        - calibrate_linear:
            # Map 0.0 (from sensor) to 0.0 (true value)
          - 612.4 -> 230.8
          - 609.1 -> 230.7
          - 611.6 -> 231.1
    power:
      name: "${plug_name}_Wattage"
      unit_of_measurement: W
      id: "${plug_name}_Wattage"
    change_mode_every: 8
    update_interval: 10s
    
  - platform: wifi_signal
    name: "${plug_name}_wifi"
    update_interval: 60s
    filters:
    - sliding_window_moving_average:
        window_size: 15
        send_every: 15
  
  - platform: uptime
    name: ${plug_name}_Uptime Sensor 
  
  - platform: total_daily_energy
    name: "${plug_name}_Total Daily Energy"
    power_id: "${plug_name}_Wattage"
    filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001
    unit_of_measurement: kWh
      
text_sensor:
  - platform: version
    name: "${plug_name}_version"
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```INFO Reading configuration...
INFO Detected timezone 'CET' with UTC offset 1 and daylight savings time from 03/29/20 02:00:00 to 10/25/20 03:00:00
INFO Starting log output from plug_03.local using esphome API
INFO Connecting to plug_03.local:6053 (192.168.1.105)
INFO Successfully connected to plug_03.local
[20:29:47][I][app:096]: esphome version 1.13.6 compiled on Jul 15 2019, 18:05:21
[20:29:47][C][wifi:372]: WiFi:
[20:29:47][C][wifi:254]:   SSID: 'RXXXXXXXXX'[redacted]
[20:29:47][C][wifi:255]:   IP Address: XXXXXXX
[20:29:47][C][wifi:257]:   BSSID: F8:32:XXXXXXXX[redacted]
[20:29:47][C][wifi:258]:   Hostname: 'plug_03'
[20:29:47][C][wifi:262]:   Signal strength: -54 dB ▂▄▆█
[20:29:47][C][wifi:263]:   Channel: 2
[20:29:47][C][wifi:264]:   Subnet: 255.255.XXXX
[20:29:47][C][wifi:265]:   Gateway: XXXXXX
[20:29:47][C][wifi:266]:   DNS1: XXXXX
[20:29:47][C][wifi:267]:   DNS2: 0.0.0.0
[20:29:47][C][logger:137]: Logger:
[20:29:47][C][logger:138]:   Level: DEBUG
[20:29:47][C][logger:139]:   Log Baud Rate: 115200
[20:29:47][C][logger:140]:   Hardware UART: UART0
[20:29:47][C][web_server:125]: Web Server:
[20:29:47][C][web_server:126]:   Address: plug_03.local:XXX
[20:29:47][C][ota:029]: Over-The-Air Updates:
[20:29:47][C][ota:030]:   Address: plug_03.local:XXX
[20:29:47][C][ota:032]:   Using Password.
[20:29:47][C][api:103]: API Server:
[20:29:47][C][api:104]:   Address: plug_03.local:XXX

**Additional information and things you've tried:**

Delete the old plug.yamel file, reinstalled ESPHome reboot home assistant.

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

brandond, 2019-07-15:
I commented on your ticket that got auto-closed. Why are you mapping 0 to multiple values? I'm pretty sure this is what's breaking it, since your stack is in `calibrate_linear_filter_to_code`:

```yaml
      filters:
        - calibrate_linear:
            # Map 0.0 (from sensor) to 0.0 (true value)
          - 0.0 -> 0.203
          - 0.0 -> 0.203
          - 0.0 -> 0.201
```

cassio299, 2019-07-15:
> I commented on your ticket that got auto-closed. Why are you mapping 0 to multiple values? I'm not sure how you would expect this to work, but I'm pretty sure this is what's breaking it since your stack is in `calibrate_linear_filter_to_code`:
> 
> ```yaml
>       filters:
>         - calibrate_linear:
>             # Map 0.0 (from sensor) to 0.0 (true value)
>           - 0.0 -> 0.203
>           - 0.0 -> 0.203
>           - 0.0 -> 0.201
> ```

Hi Brandond
Thank you for you reply.
I am following the instructions on filter sensors: https://esphome.io/components/sensor/index.html
The 0.00 are previous internal measurements for the current and the 0.2... are measurements by my volt meter.
You are right if I change the 0 measurements in version 13.6 I can compile, however the calibration are now wrong.
Doo you have a solution with a 0 measurement that works??

brandond, 2019-07-15:
You should be able to map 0 to things, you just can't map multiple readings to the same thing. From the example:

```yaml
filters:
  - calibrate_linear:
      - 0.0 -> 0.0
      - 40.0 -> 45.0
      - 100.0 -> 102.5
```

brandond, 2019-07-15:
@OttoWinter maybe some sort of sanity check here to ensure that the readings aren't all at the same measured value would help avoid the divide by zero?

cassio299, 2019-07-16:
I have deleted all my measurements 

And started all over the new debug readings for the currents came as a surprise:

[D][hlw8012:063]: Got power=42.0W, current=0.0A
[D][sensor:092]: 'plug_03_Current': Sending state 0.04183 A with 2 decimals of accuracy

I might have taken the wrong readout, I just wonder why it worked back in version 12.2.

Anyway I also discovered that I had a old left over from the previous calibration method:
current_resistor: 0.00599

I then took some new measurements:

 filters:
        - calibrate_linear:
          # Model Gosund SP1-4 (0.013A)
          # Map 0.0 (from sensor) to 0.0 (measured value)
          - 0.00 -> 0.013
          - 0.24624 -> 0.202
          - 0.25106 -> 0.200
          - 0.25106 -> 0.198
          - 0.24624 -> 0.198
          - 0.24524 -> 0.197
          - 0.24141 -> 0.197
        - lambda: if (x < (0.01 - 0.013)) return 0; else return (x - 0.013);

Debug:
Got power=252.4W, current=0.2A
[15:58:46][D][sensor:092]: 'plug_03_Current': Sending state 0.18734 A with 3 decimals of accuracy

Thank you so much for your help, Brandond. I was totally lost.

closed 2019-07-16 by cassio299

OttoWinter, 2019-07-21:
This is because the provided values can not be used to form a linear equation. You cannot construct a linear equation from just values that have the same x value.