esphome/issues #526: Update ssid and password through WebServer hosted on AP

manju-rn, 2019-07-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io on RPI3 Model B+

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**Affected component:**

WIFI and WebServer

**Description of problem:**
The WIFI setup in my YAML config has both Stand and AP mode. I have also included the WebServer  component.  And it works great - meaning when WIFI ssid or password changes, I am able to connect to webpage hosted through AP of ESP8266.  However, is there a way that a new WIFI ssid and password can be included in the Webserver's webpage so that I do not have to go through YAML hosted in hass.io.  

The reason is that once the WIFI connectitivity is lost, the ESP8266 should be manually brought over to the RPI and flashed through USB port of RPI and I would like to avoid that.

I understand the WebServer component of the ESPHome may not allow addition of extra fields, so kindly suggest way to override using custom webserver for this purpose.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
I have tried to override the WIFI components in the main.cpp by psuedocode below

setup()
{
// generated code
...
//after generated code
esphome::wifi::WiFiAP.set_ssid("new ssid");
esphome::wifi::WiFiAP.set_password("new password");
esphome::wifi::WiFiComponent::WiFiComponent->retry_connect ();
APP.setup();
}

But this code never connect, I believe the ESP reboots and always tries to connect to the original ssid and password from YAML

manju-rn, 2019-07-19:
This is resolved.  Made following changes in main.cpp
1. Declared the AysyncWebServer and WIFIAP as global variables
// ========== AUTO GENERATED INCLUDE BLOCK END ==========="
AsyncWebServer mywebserver(9090);
wifi::WiFiAP wifi_wifiap_global;

2. Within the Setup() - Assigned the object of WIFIAP which was instantiated within the generated code block to the global variable so that it could be used inside the Lambda call.  Used the .on call to trigger the Lambda call. The lambda call sets the new Password (and SSID call, if required) and used the wifi_component pointer to start reconnecting
  // =========== AUTO GENERATED CODE END ============
  // ========= YOU CAN EDIT AFTER THIS LINE =========
  wifi_wifiap_global = wifi_wifiap;
  mywebserver.on("/reconnect", HTTP_GET, [wifi_wificomponent, &wifi_wifiap_global](AsyncWebServerRequest *request)
  {
          // get the password from the request 
          std::string newPassword = <code>
	  request->send(200, "text/plain", "Passwod changed...reconnecting");
	  wifi_wifiap_global.set_password(newPassword);
	  wifi_wificomponent->start_connecting(wifi_wifiap_global, false);
  }); 

3. Start the AsyncWebserver that was declared globally - just before the App.setup() which was generated by ESPHome
  mywebserver.begin(); 	
  App.setup();
}

Summary:
1. The above starts a new Webserver at port 9090 apart from the usual existing Webserver defined via a YAML.
2. Change the password of the SSID currently ESP device connected to.  The ESPHome should immediately complain that the ESP Device is disconnected.
3. The YAML definition already had both STA and AP mode.  Hence, the ESP Device switches to AP mode 
2. Use the URL (of the AP mode) /reconnect?password=newpassword to pass the new password - I know it is a insecure way to pass the password, but the objective here was just to see if SSID and Password could be changed directly incase the original ssid/password has changed without the need to change the YAML and regenerate and flash the firmware from ESPHome back to ESP device
3. The ESP Device should now be connected to same SSID whose password was changed (or any other as desired)

Known Issues:
1.  If the original SSID & password has no changes, then change of the password through /reconnect does not work. It does disconnect momentarily (as the ESPHome reports that ESP device is disconnected) but immediately reconnects after say 2 secs to the same SSID (with original password)- looks like it does not recognize changed password
2. When in AP mode, after the change of the password through /reconnect, if the original SSID password is changed back to the original password, ESP Device happily reconnects to the original SSID as if the password changed through /reconnect never happened.  
So my assumption as of now is that the call from Lambda to change the password through WIFI component is temporary as long as there is a new SSID with that password given during /reconnect. if the original SSID with original password (given in YAML) is existing, it will never change to new SSID and password.
I would look into the WIFI component source code to see whats happening.  

For now the objective is met that if the SSID and password is changed, I can reconnect the ESP device to new SSID / password without regenerating the code and flash the firmware from ESPHome

OttoWinter, 2019-07-21:
- Not an issue, this is a feature request
- Duplicate of https://github.com/esphome/feature-requests/issues/86, https://github.com/esphome/feature-requests/issues/286

closed 2019-07-21 by OttoWinter