esphome/issues #527: Repeat or Wait_time not working for remote_transmitter

chris-jennings, 2019-07-17:
Ubuntu virtual environment installed with pip2

ESP8266 (Wemos mini d1)

[Remote_transmitter](https://esphome.io/components/remote_transmitter.html)

I upgraded to the latest ESPhome version and had to change the switch components that use the modified remote_transmitter configuration with the templates. After I made the modifications the IR transmitter no longer was able to send valid codes the the Sony TV. I managed to get it to work by tapping on the switch (in Homeassistant) to send the code with the correct delay and repeats as per the requirements of a Sony device (3 repeats at 40ms interval).

I modified the repeat configuration so I could see the time between repeats by viewing the IR transmitter using a camera (4000ms repeat interval) and I don't see any codes being repeated. I can't work out if the repeats are not working or the wait_time is not being honoured and perhaps the code is being sent immediately.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: wemos2
  platform: ESP8266
  board: d1_mini_lite

wifi:
  ssid: --------------------
  password: --------------------------

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

switch:
  - platform: template
    name: Sony TV Power
    turn_on_action:
      remote_transmitter.transmit_sony:
        data: 0xA90
        nbits: 12
        repeat:
          times: 3
          wait_time: 40ms
  - platform: template
    name: Sony TV Volume Up
    turn_on_action:
      remote_transmitter.transmit_sony:
        data: 0x00000490
        nbits: 12
        repeat:
          times: 3
          wait_time: 40ms
  - platform: template
    name: Sony TV Volume Down
    turn_on_action:
      remote_transmitter.transmit_sony:
        data: 0x00000C90
        nbits: 12
        repeat:
          times: 3
          wait_time: 40ms

remote_transmitter:
  pin: D5
  carrier_duty_percent: 50%

```

**Logs (if applicable):**

```
INFO Successfully connected to wemos2.local
[15:57:11][I][app:096]: esphome version 1.13.6 compiled on Jul 17 2019, 15:56:42
[15:57:11][C][wifi:372]: WiFi:
[15:57:11][C][wifi:254]:   SSID: 'Edgey'
[15:57:11][C][wifi:255]:   IP Address: 192.168.86.94
[15:57:11][C][wifi:257]:   BSSID: 70:3A:CB:91:A6:3F
[15:57:11][C][wifi:258]:   Hostname: 'wemos2'
[15:57:11][C][wifi:262]:   Signal strength: -34 dB ▂▄▆█
[15:57:11][C][wifi:263]:   Channel: 6
[15:57:11][C][wifi:264]:   Subnet: 255.255.255.0
[15:57:11][C][wifi:265]:   Gateway: 192.168.86.1
[15:57:11][C][wifi:266]:   DNS1: 192.168.86.1
[15:57:11][C][wifi:267]:   DNS2: 0.0.0.0
[15:57:11][C][template.switch:058]: Template Switch 'Sony TV Power'
[15:57:11][C][template.switch:059]:   Restore State: NO
[15:57:11][C][template.switch:060]:   Optimistic: NO
[15:57:11][C][template.switch:058]: Template Switch 'Sony TV Volume Up'
[15:57:11][C][template.switch:059]:   Restore State: NO
[15:57:11][C][template.switch:060]:   Optimistic: NO
[15:57:11][C][template.switch:058]: Template Switch 'Sony TV Volume Down'
[15:57:11][C][template.switch:059]:   Restore State: NO
[15:57:11][C][template.switch:060]:   Optimistic: NO
[15:57:11][C][template.switch:058]: Template Switch 'Sony TV Source'
[15:57:11][C][template.switch:059]:   Restore State: NO
[15:57:11][C][template.switch:060]:   Optimistic: NO
[15:57:11][C][template.switch:058]: Template Switch 'Sony TV Menu'
[15:57:11][C][template.switch:059]:   Restore State: NO
[15:57:11][C][template.switch:060]:   Optimistic: NO
[15:57:11][C][template.switch:058]: Template Switch 'Sony TV Menu Up'
[15:57:11][C][template.switch:059]:   Restore State: NO
[15:57:11][C][template.switch:060]:   Optimistic: NO
[15:57:11][C][template.switch:058]: Template Switch 'Sony TV Menu Down'
[15:57:11][C][template.switch:059]:   Restore State: NO
[15:57:11][C][template.switch:060]:   Optimistic: NO
[15:57:11][C][template.switch:058]: Template Switch 'Sony TV Menu Select'
[15:57:11][C][template.switch:059]:   Restore State: NO
[15:57:11][C][template.switch:060]:   Optimistic: NO
[15:57:11][C][logger:137]: Logger:
[15:57:11][C][logger:138]:   Level: DEBUG
[15:57:11][C][logger:139]:   Log Baud Rate: 115200
[15:57:11][C][logger:140]:   Hardware UART: UART0
[15:57:11][C][remote_transmitter:018]: Remote Transmitter...
[15:57:11][C][remote_transmitter:019]:   Carrier Duty: 50%
[15:57:11][C][remote_transmitter:020]:   Pin: GPIO14 (Mode: OUTPUT)
[15:57:11][C][ota:029]: Over-The-Air Updates:
[15:57:11][C][ota:030]:   Address: wemos2.local:8266
[15:57:11][C][api:103]: API Server:
[15:57:11][C][api:104]:   Address: wemos2.local:6053
[15:57:38][D][switch:021]: 'Sony TV Power' Turning ON.
[15:57:38][D][remote_transmitter:066]: Sending remote code...
[15:57:46][D][switch:021]: 'Sony TV Power' Turning ON.
[15:57:46][D][remote_transmitter:066]: Sending remote code...
[15:57:46][D][switch:021]: 'Sony TV Power' Turning ON.
[15:57:46][D][remote_transmitter:066]: Sending remote code...
[15:57:46][D][switch:021]: 'Sony TV Power' Turning ON.
[15:57:46][D][remote_transmitter:066]: Sending remote code...

```

I checked that the Sony code was valid but I haven't got around to viewing the code being sent with another ESP and an IR receiver. But the fact I can create the correctly formed code manually indicates that the Sony code is being formed correctly only that the repeat and/or wait_time is incorrect.

If my YAML file was incorrect I would have expected a parsing error instead of ignoring an incorrect entry.

The Platformio code seems to be correct as well:
` remote_base_sonyaction->set_send_times(3);
  remote_base_sonyaction_2->set_send_times(3);
  remote_base_sonyaction_3->set_send_times(3);
  remote_base_sonyaction_4->set_send_times(3);
  remote_base_sonyaction_5->set_send_times(3);
  remote_base_sonyaction_6->set_send_times(3);
  remote_base_sonyaction_7->set_send_times(3);
  remote_base_sonyaction_8->set_send_times(3);
  remote_base_sonyaction->set_send_wait(40);
  remote_base_sonyaction_2->set_send_wait(40);
  remote_base_sonyaction_3->set_send_wait(40);
  remote_base_sonyaction_4->set_send_wait(40);
  remote_base_sonyaction_5->set_send_wait(40);
  remote_base_sonyaction_6->set_send_wait(40);
  remote_base_sonyaction_7->set_send_wait(40);
  remote_base_sonyaction_8->set_send_wait(40);
`

chris-jennings, 2019-07-17:
Setting the repeats to 1000 and I certainly get 1000 repeats. So I assume the wait time must be wrong.

chris-jennings, 2019-07-17:
Sorry, it looks like this is related to [485](https://github.com/esphome/issues/issues/485) I changed the wait_time to 1000 times the value and now it works.

OttoWinter, 2019-07-21:
Closing as duplicate of https://github.com/esphome/issues/issues/485

closed 2019-07-21 by OttoWinter