esphome/issues #528: Multiple deep_sleep components

jeff-h, 2019-07-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 (nodemcu v2)

**Affected component:**

https://esphome.io/components/deep_sleep.html

**Description of problem:**
It's unclear whether multiple deep_sleep definitions are allowed. In the absence of anything stating you can't, my hope was that you can due to the precedent in other components. However, there's a validation error thrown when I have more than one.

**Problem-relevant YAML-configuration entries:**
```yaml
deep_sleep:
  # awake 40s in every 2 mins
  - id: sleep_a_little
    run_duration: 40s
    sleep_duration: 80s 

  # awake 40s in every 6 mins
  - id: sleep_a_lot 
    run_duration: 40s
    sleep_duration: 320s 
```

**Logs (if applicable):**

```
The live YAML validator reports "expected a dictionary".
```

**Additional information and things you've tried:**

I'd like my ESP8266 to sleep a little during the day, and a lot at night (it's running off a battery and solar panel, so there's reasonably ample power in the day). I was hoping to have a `time` component switch between the two `deep_sleep` definitions.

r-jordan, 2019-07-17:
Hi,

I have a similar device running off a battery and solar panel and I've coded the following to change the sleep duration depending on the battery voltage detected with a sensor in the analog pin:

<pre>
    on_value:
      then:
      - lambda: |-
          ESP_LOGW("main","battery = [%f]",x );
          if(x>3.69)
          {
            id(deep_sleep_1).set_sleep_duration(4*60000);
            id(deep_sleep_1).setup();
          }
          else if(x<3.50)
          {
            id(deep_sleep_1).set_sleep_duration(((3.50-x)*600+9)*60000);
            id(deep_sleep_1).setup();
          }
</pre>

The default sleep duration is defined in the deep_sleep definition:

<pre>
deep_sleep:
  run_duration: 50sec
  sleep_duration: 9min
  id: deep_sleep_1
</pre>

This way if a very low solar energy is received during a day the device won't rest running without sleeping because running out of battery. If that situation occurrs then the battery would not have a chance to recharge because the device is running for an undefined long time.

Hope that helps!

jeff-h, 2019-07-17:
@r-jordan Fantastic! Thanks for posting that — very clever indeed. I'll almost certainly steal a bunch of your code :) It constantly blows my mind how sophisticated you can get with ESPHome.

Although if ESPHome did allow multiple `deep_sleep` components you can see how simple that would make things in my particular use-case.

OttoWinter, 2019-07-21:
@r-jordan That's quite hacky and should definitely not be used that way (calling `setup` could do all sorts of stuff, you should not call it yourself - but in this case it probably will work fine).

- Also, this is a feature request, not an issue
- I think the premise behind this is wrong - the ESP has only one deep_sleep hw component, so esphome also just has one deep_sleep. What would be more of a solution would be to have these options be exposed through automation actions so that they can be changed at runtime.

closed 2019-07-21 by OttoWinter