esphome/issues #535: MCP23017 Switch config order

alexyao2015, 2019-07-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESP12-E
**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

MCP23017
https://esphome.io/components/mcp23017.html

**Description of problem:**
Output switch is required to be before the mcp23017 configuration entry. There is no indication of this in the docs that this was intentional. 

YAMLs were copied directly from the docs but the only difference is the order where switch is in the config.

Please fix it to allow it in any order or add this detail to the docs.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
WORKING:
mcp23017:
  - id: 'mcp23017_hub'
    address: 0x20

# Individual outputs
switch:
  - platform: gpio
    name: "MCP23017 Pin #0"
    pin:
      mcp23017: mcp23017_hub
      # Use pin number 0
      number: 0
      # One of INPUT, INPUT_PULLUP or OUTPUT
      mode: OUTPUT
      inverted: False

NOT WORKING:
# Individual outputs
switch:
  - platform: gpio
    name: "MCP23017 Pin #0"
    pin:
      mcp23017: mcp23017_hub
      # Use pin number 0
      number: 0
      # One of INPUT, INPUT_PULLUP or OUTPUT
      mode: OUTPUT
      inverted: False

mcp23017:
  - id: 'mcp23017_hub'
    address: 0x20
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

OttoWinter, 2019-07-21:
Could you explain what error you see when they are not in that order? It should not matter in what order they are initialized from a validation point of view. And the setup order should also not be a factor because mcp23017 has setup_priority before switch

alexyao2015, 2019-07-21:
There's no error or any indication of an error. Config validates fine and uploads fine. The switch even turns on fine but there is no output. Logs show no errors at all.

I spent hours messing with the config until I found out this was the issue.

closed 2019-10-17 by OttoWinter