esphome/issues #538: Infrared remote transmitter sending wrong signals

tiandeyu, 2019-07-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32
**Affected component:**

https://esphome.io/components/remote_transmitter.html

**Description of problem:**
The IR transmitter doesn't work. Get the wrong signal from IR receiver compare to my config.
Console logs attached below.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

remote_receiver:
  pin: GPIO23
  dump: all
  buffer_size: 20kb
  idle: 50ms
  
remote_transmitter:
  pin: GPIO22
  carrier_duty_percent: 50%

binary_sensor:
  - platform: remote_receiver
    name: "Remote Input"
    raw:
      code: [-3480, 1774, -414, 467, -415, 1339, -414, 467, -414, 467, -414, 468, -414, 468, -414, 467, -414, 468, -415, 467, -414, 466, -414, 468, -414, 467, -414, 467, -415, 1338, -415, 468, -413, 468, -414, 467, -414, 466, -415, 467, -413, 469, -414, 
1338, -415, 467, -414, 1339, -415, 1338, -415, 467, -414, 466, -414, 1340, -414, 468, -414, 467, -415, 1339, -415, 467, -414, 468, -415, 468, -414, 1338, -414, 467, -414, 467, -414, 1339, -415, 1339, -414, 468, -414, 1340, -415, 1339, -413, 468, -414, 
468, -413, 467, -414, 1340, -413, 469, -414, 467, -414, 447, -433, 1343, -411, 470, -414, 1340, -412, 470, -412, 470, -413, 1341, -410, 472, -411, 472, -412, 470, -409, 472, -410, 471, -410, 471, -413, 469, -411, 471, -411]
          

switch:
  - platform: template
    name: Raw Code Power Button
    turn_on_action:
      - remote_transmitter.transmit_raw:
          code: [-3480, 1774, -414, 467, -415, 1339, -414, 467, -414, 467, -414, 468, -414, 468, -414, 467, -414, 468, -415, 467, -414, 466, -414, 468, -414, 467, -414, 467, -415, 1338, -415, 468, -413, 468, -414, 467, -414, 466, -415, 467, -413, 469, -414, 
1338, -415, 467, -414, 1339, -415, 1338, -415, 467, -414, 466, -414, 1340, -414, 468, -414, 467, -415, 1339, -415, 467, -414, 468, -415, 468, -414, 1338, -414, 467, -414, 467, -414, 1339, -415, 1339, -414, 468, -414, 1340, -415, 1339, -413, 468, -414, 
468, -413, 467, -414, 1340, -413, 469, -414, 467, -414, 447, -433, 1343, -411, 470, -414, 1340, -412, 470, -412, 470, -413, 1341, -410, 472, -411, 472, -412, 470, -409, 472, -410, 471, -410, 471, -413, 469, -411, 471, -411]
          carrier_frequency: 38000Hz
          
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
[15:47:27][D][binary_sensor:033]: 'Remote Input': Sending state ON
[15:47:27][D][binary_sensor:033]: 'Remote Input': Sending state OFF
[15:47:32][D][switch:021]: 'Raw Code Power Button' Turning ON.
[15:47:32][D][remote.raw:028]: Received Raw: -1688, 521, -428, 442, -1263, 498, -428, 443, -394, 498, -428, 443, -393, 498, -429, 443, -393, 500, -427, 466, -369, 499, -428, 444, -392, 499, -428, 444, -1285, 476, -426, 467, -395, 475, -427, 467, -370, 499, -427, 467, -392, 477, -1295, 
[15:47:32][D][remote.raw:028]:   466, -425, 446, -1295, 466, -1296, 443, -393, 499, -427, 443, -1264, 498, -427, 466, -372, 498, -1295, 468, -427, 443, -425, 467, -428, 443, -1240, 520, -428, 443, -372, 520, -1297, 467, -1290, 447, -427, 467, -1296, 442, -1263, 498, -428, 443, -393, 
[15:47:32][D][remote.raw:041]:   499, -427, 443, -1287, 475, -426, 468, -394, 475, -376, 517, -1297, 442, -427, 466, -1296, 442, -373, 521, -428, 442, -1315, 445, -428, 468, -427, 443, -425, 467, -428, 442, -427, 466, -426, 444, -427, 466, -427
```

**Additional information and things you've tried:**

glmnet, 2019-07-23:
Try inverting the input of the receiver

tiandeyu, 2019-07-23:
> Try inverting the input of the receiver

Thx, will try

glmnet, 2019-07-23:
Sorry for the short answer.
For testing try using common IR Transmitters as many as you have, like TVs, etc. ESPHome should decode them without problems, you should see NEC or Panasonic, etc.
Also your raw there starts with a e.g. -1688, that is a negative number, it means do nothing at the start then transmit 521 microseconds then do nothing... that is a sign that the signal is inverted, and usually because of that ESPHome cannot decode the protocols and you end up with a lot of numbers.

tiandeyu, 2019-07-23:
Hi Ruffino,
Thanks very much for your reply. I tried to invert the receiver pin and got the right signals. And I'm using a Panasonic bathroom ventilator remote. Again appreciate your help.

tiandeyu, 2019-07-23:
The transmitter still not woking.

My config
```ymal
remote_receiver:
  pin: 
    number: GPIO23
    inverted: True
  dump: all
  buffer_size: 20kb
  idle: 50ms
  
remote_transmitter:
  pin: GPIO22
  carrier_duty_percent: 50%

binary_sensor:
  - platform: remote_receiver
    name: "Remote Input"
    panasonic:
      address: 0x4004
      command: 0x0B244D08
    
switch:
  - platform: template
    name: Raw Code Power Button
    turn_on_action:
      - remote_transmitter.transmit_panasonic:
          address: 0x4004
          command: 0x0B244D08
```
logs when toggle the switch.
[21:46:11][D][switch:021]: 'Raw Code Power Button' Turning ON.
[21:46:11][D][remote.raw:028]: Received Raw: 3434, -1824, 435, -474, 469, -1275, 467, -444, 460, -429, 467, -444, 468, -447, 383, -503, 467, -445, 429, -459, 467, -445, 465, -423, 466, -445, 471, -443, 385, -1357, 430, -459, 467, -444, 467, -421, 466, -446, 468, -446, 406, -481, 468, 
[21:46:11][D][remote.raw:028]:   -1275, 467, -444, 468, -1275, 470, -1274, 468, -444, 433, -455, 467, -1275, 440, -472, 469, -444, 408, -1335, 432, -455, 468, -443, 468, -423, 463, -1303, 406, -482, 467, -445, 434, -1308, 468, -1275, 468, -446, 406, -1336, 432, -456, 468, -443, 467, 
[21:46:11][D][remote.raw:041]:   -423, 463, -448, 469, -1275, 466, -445, 434, -454, 468, -444, 469

tiandeyu, 2019-07-23:
I'm using 5mm 940nm infrared transmitter. I hooked up the transmitter directly to GPIO and ground or using a 2N2222 transistor get the same result.

OttoWinter, 2019-10-13:
> Panasonic bathroom ventilator remote

That's the problem - that protocol hasn't been implemented yet. And it's likely not a simple protocol where you can just re-transmit a raw code (because it may contain other stuff like mode etc). This would rather be a feature request then, if you still need this, please open a feature request with the exact device you would like to integrate and the protocol it uses.

closed 2019-10-13 by OttoWinter