esphome/issues #539: Multiple I2C buses, one bus shadows another

tdamsma, 2019-07-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP-WROOM-32 

**Affected component:**

I2C / BME280

**Description of problem:**

I configured 3 i2c buses, and all seems to work. On closer inspection however the readings for K1 (on bus 1) are almost identical to the readings from K3. K1 and K2 are next to each other, and temp should be about 20 degrees. K3 is somewhere else, and it's temperature is much higher. So for some reason the values that are read for K1 are actually those of K3. 

I tried the following:

 - Only use K1+K3, or K1+K2 ==> all working as expected
 - K1 disconnected: no errors, giving readings from K3
 - Change pin number for i2c buses, same issues
 - Add a bus0 on (not connected) pins 19 and 21: 
    A device is found even though there is none
     ```
    [20:00:52][C][i2c:028]: I2C Bus:
    [20:00:52][C][i2c:029]:   SDA Pin: GPIO19
    [20:00:52][C][i2c:030]:   SCL Pin: GPIO21
    [20:00:52][C][i2c:031]:   Frequency: 25000 Hz
    [20:00:52][I][i2c:033]: Scanning i2c bus for active devices...
    [20:00:53][I][i2c:040]: Found i2c device at address 0x76
    ```
    Still K3 values on K1 bus

I checked the generated code, and I couldn't spot any obvious errors (but I am not familiar with C++). Of note, The same three buses worked fine with a micropython based experiment

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: "102"
  platform: ESP32
  board: esp32dev

wifi:
  ssid: ""
  password: ""

i2c:
  - sda: 17
    scl: 16
    id: bus1
    frequency: 25kHz
  - sda: 12
    scl: 14
    id: bus2
    frequency: 25kHz
  - sda: 26
    scl: 27
    id: bus3
    frequency: 25kHz

sensor:
  - platform: bme280
    i2c_id: bus1
    temperature:
      name: "K1"
    address: 0x76
    update_interval: 4s
  - platform: bme280
    i2c_id: bus2
    temperature:
      name: "K2"
    address: 0x76
    update_interval: 4s
  - platform: bme280
    i2c_id: bus3
    temperature:
      name: "K3"
    address: 0x76
    update_interval: 4s

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
```

**Logs (if applicable):**

```
[19:20:44][I][app:096]: esphome version 1.13.6 compiled on Jul 21 2019, 19:20:21
[19:20:44][C][i2c:028]: I2C Bus:
[19:20:44][C][i2c:029]:   SDA Pin: GPIO17
[19:20:44][C][i2c:030]:   SCL Pin: GPIO16
[19:20:44][C][i2c:031]:   Frequency: 25000 Hz
[19:20:44][I][i2c:033]: Scanning i2c bus for active devices...
[19:20:44][I][i2c:040]: Found i2c device at address 0x76
[19:20:44][D][bme280.sensor:197]: Got temperature=28.6°C pressure=1022.6hPa humidity=35.0%
[19:20:44][D][sensor:092]: 'K1': Sending state 28.64000 °C with 1 decimals of accuracy
[19:20:44][C][i2c:028]: I2C Bus:
[19:20:44][C][i2c:029]:   SDA Pin: GPIO12
[19:20:44][C][i2c:030]:   SCL Pin: GPIO14
[19:20:44][C][i2c:031]:   Frequency: 25000 Hz
[19:20:44][I][i2c:033]: Scanning i2c bus for active devices...
[19:20:45][I][i2c:040]: Found i2c device at address 0x76
[19:20:45][C][i2c:028]: I2C Bus:
[19:20:45][C][i2c:029]:   SDA Pin: GPIO26
[19:20:45][C][i2c:030]:   SCL Pin: GPIO27
[19:20:45][C][i2c:031]:   Frequency: 25000 Hz
[19:20:45][I][i2c:033]: Scanning i2c bus for active devices...
[19:20:45][I][i2c:040]: Found i2c device at address 0x76
[19:20:45][C][logger:137]: Logger:
[19:20:45][C][logger:138]:   Level: DEBUG
[19:20:45][C][logger:139]:   Log Baud Rate: 115200
[19:20:45][C][logger:140]:   Hardware UART: UART0
[19:20:45][C][bme280.sensor:142]: BME280:
[19:20:45][C][bme280.sensor:143]:   Address: 0x76
[19:20:45][C][bme280.sensor:155]:   IIR Filter: OFF
[19:20:45][C][bme280.sensor:156]:   Update Interval: 4.0s
[19:20:45][C][bme280.sensor:158]:   Temperature 'K1'
[19:20:45][C][bme280.sensor:158]:     Unit of Measurement: '°C'
[19:20:45][C][bme280.sensor:158]:     Accuracy Decimals: 1
[19:20:45][C][bme280.sensor:158]:     Icon: 'mdi:thermometer'
[19:20:45][C][bme280.sensor:159]:     Oversampling: 16x
[19:20:45][C][bme280.sensor:161]:     Oversampling: 16x
[19:20:45][C][bme280.sensor:163]:     Oversampling: 16x
[19:20:45][C][bme280.sensor:142]: BME280:
[19:20:45][C][bme280.sensor:143]:   Address: 0x76
[19:20:45][C][bme280.sensor:155]:   IIR Filter: OFF
[19:20:45][C][bme280.sensor:156]:   Update Interval: 4.0s
[19:20:45][C][bme280.sensor:158]:   Temperature 'K2'
[19:20:45][C][bme280.sensor:158]:     Unit of Measurement: '°C'
[19:20:45][C][bme280.sensor:158]:     Accuracy Decimals: 1
[19:20:45][C][bme280.sensor:158]:     Icon: 'mdi:thermometer'
[19:20:45][C][bme280.sensor:159]:     Oversampling: 16x
[19:20:45][C][bme280.sensor:161]:     Oversampling: 16x
[19:20:45][C][bme280.sensor:163]:     Oversampling: 16x
[19:20:45][C][bme280.sensor:142]: BME280:
[19:20:45][C][bme280.sensor:143]:   Address: 0x76
[19:20:45][C][bme280.sensor:155]:   IIR Filter: OFF
[19:20:45][C][bme280.sensor:156]:   Update Interval: 4.0s
[19:20:45][C][bme280.sensor:158]:   Temperature 'K3'
[19:20:45][C][bme280.sensor:158]:     Unit of Measurement: '°C'
[19:20:45][C][bme280.sensor:158]:     Accuracy Decimals: 1
[19:20:45][C][bme280.sensor:158]:     Icon: 'mdi:thermometer'
[19:20:45][C][bme280.sensor:159]:     Oversampling: 16x
[19:20:45][C][bme280.sensor:161]:     Oversampling: 16x
[19:20:45][C][bme280.sensor:163]:     Oversampling: 16x
[19:20:45][C][ota:029]: Over-The-Air Updates:
[19:20:45][C][ota:030]:   Address: 102.local:3232
[19:20:45][C][api:103]: API Server:
[19:20:45][C][api:104]:   Address: 102.local:6053
[19:20:47][D][bme280.sensor:197]: Got temperature=19.8°C pressure=1022.7hPa humidity=69.2%
[19:20:47][D][sensor:092]: 'K2': Sending state 19.77000 °C with 1 decimals of accuracy
[19:20:47][D][bme280.sensor:197]: Got temperature=28.7°C pressure=1022.6hPa humidity=34.8%
[19:20:47][D][sensor:092]: 'K3': Sending state 28.71000 °C with 1 decimals of accuracy
[19:20:48][D][bme280.sensor:197]: Got temperature=28.7°C pressure=1022.7hPa humidity=34.8%
[19:20:48][D][sensor:092]: 'K1': Sending state 28.73000 °C with 1 decimals of accuracy
[19:20:51][D][bme280.sensor:197]: Got temperature=19.8°C pressure=1022.6hPa humidity=69.2%
[19:20:51][D][sensor:092]: 'K2': Sending state 19.77000 °C with 1 decimals of accuracy
[19:20:51][D][bme280.sensor:197]: Got temperature=28.8°C pressure=1022.6hPa humidity=34.6%
[19:20:51][D][sensor:092]: 'K3': Sending state 28.79000 °C with 1 decimals of accuracy
[19:20:52][D][bme280.sensor:197]: Got temperature=28.8°C pressure=1022.6hPa humidity=34.6%
[19:20:52][D][sensor:092]: 'K1': Sending state 28.81000 °C with 1 decimals of accuracy
[19:20:55][D][bme280.sensor:197]: Got temperature=19.8°C pressure=1022.7hPa humidity=69.2%
[19:20:55][D][sensor:092]: 'K2': Sending state 19.76000 °C with 1 decimals of accuracy
[19:20:55][D][bme280.sensor:197]: Got temperature=28.9°C pressure=1022.7hPa humidity=34.4%
[19:20:55][D][sensor:092]: 'K3': Sending state 28.86000 °C with 1 decimals of accuracy
[19:20:56][D][bme280.sensor:197]: Got temperature=28.9°C pressure=1022.6hPa humidity=34.4%
[19:20:56][D][sensor:092]: 'K1': Sending state 28.88000 °C with 1 decimals of accuracy
[19:20:59][D][bme280.sensor:197]: Got temperature=19.8°C pressure=1022.7hPa humidity=69.2%
[19:20:59][D][sensor:092]: 'K2': Sending state 19.76000 °C with 1 decimals of accuracy
[19:20:59][D][bme280.sensor:197]: Got temperature=28.9°C pressure=1022.6hPa humidity=34.3%

```

**Additional information and things you've tried:**

tdamsma, 2019-08-21:
As the ESP32 supports 2 hardware i2c buses I think that is causing the issues. In principle you could do software i2c but that probably (apparently) is not supported.

closed 2019-08-21 by tdamsma

OttoWinter, 2019-08-27:
@tdamsma Correct, ESP32 only has two hw i2c buses, and I don't think porting some sw i2c solution to the ESP32 would be worth the effort, it might be interesting though to integrate i2c bus expanders (should be possible due to the modular design of I2CDevice).

OttoWinter, 2019-08-27:
@brandond Correct, problem is that's not really possible right now (`MULTI_CONF` makes it impossible to validate the full conf struct, including the length of the values in i2c block).