esphome/issues #540: Esphome dashboard: Invalid username and password

marcob1281, 2019-07-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Ubuntu 18.04, HAssio via Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**
I did follow the instructions provided with the addon, including doing the dashboard login using Hassio credential, but it doesn't work.
I get "Invalid Username or password"
I am sure the credential I used are fully working with Hassio.
Addon version 1.13.6, release or beta, no changes

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 00-banner.sh: executing... 

-----------------------------------------------------------
 Hass.io Add-on: ESPHome (beta)
 Beta version of ESPHome Hass.io add-on.
-----------------------------------------------------------
 Add-on version: 1.13.6
 You are running the latest version of this add-on.
 System: Ubuntu 18.04.2 LTS  (amd64 / qemux86-64)
 Home Assistant version: 0.96.3
 Supervisor version: 170
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
[cont-init.d] 00-banner.sh: exited 0.
[cont-init.d] 01-log-level.sh: executing... 
[cont-init.d] 01-log-level.sh: exited 0.
[cont-init.d] 10-requirements.sh: executing... 
[cont-init.d] 10-requirements.sh: exited 0.
[cont-init.d] 20-nginx.sh: executing... 
[cont-init.d] 20-nginx.sh: exited 0.
[cont-init.d] 30-esphome.sh: executing... 
[cont-init.d] 30-esphome.sh: exited 0.
[cont-init.d] 40-migrate.sh: executing... 
[cont-init.d] 40-migrate.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[00:28:42] INFO: Waiting for dashboard to come up...
[00:28:42] INFO: Starting ESPHome dashboard...
INFO Starting dashboard web server on unix socket /var/run/esphome.sock and configuration dir /config/esphome...
[00:28:43] INFO: Starting NGINX...
INFO 302 GET / (0.0.0.0) 1.24ms
INFO 304 GET /login (0.0.0.0) 13.94ms
WARNING 401 POST /login (0.0.0.0) 61935.08ms
INFO 302 GET / (0.0.0.0) 0.83ms
INFO 304 GET /login (0.0.0.0) 1.86ms

```

**Additional information and things you've tried:**

fah, 2019-07-22:
I'm no expert but maybe some stupid questions do help you:

Why are you using the beta version? I'd rather reinstall a stable version then search for the PW.

Did you enter the password into the username field to check that the keyboard-layout is the expected one?

What was the reason for this delay?:
```
WARNING 401 POST /login (0.0.0.0) 61935.08ms
```

marcob1281, 2019-07-23:
hi,
I tried either stable and beta version, no changes.
Keyboard layout is fine.
I don't know why there is that delay, all the communications are running on
the same device (Intel ATOM N3350 board, UP Squared from UP/AAEON)

Marco Barbato
http://flickr.com/photos/marcobarbato

«..Si quieres cambio verdadero …pues camina distinto..»
*Calle13*

marcob1281, 2019-07-24:
@fah  I owe you some info more.
I tried with Chrome and Firefox.
the address of hassio is http://localhost:8123
Looks esphome tries to connect to 0.0.0.0
shall I have to configure the Hassio address somewhere?

OttoWinter, 2019-10-13:
Nah, 0.0.0.0 here is the source address - and because you're using Hass.io ingress, HA handles the request and forwards it to esphome with the source address 0.0.0.0

> What was the reason for this delay?:

That looks like the culprit. When you try to log in, esphome contacts the Hass.io host system to validate the username+password. After 60 seconds of no reply from the host, it will bail out and raise an error. So it seems like your HA system is not accepting authentication requests.

It looks like you're using the docker image for Hass.io, but with a docker system. Please use the `esphome/esphome` image instead, it also has the dashboard. The Hass.io image is only meant for the Hass.io addon.

closed 2019-10-13 by OttoWinter