esphome/issues #541: Dual relay cover motor control

hugodelaroza, 2019-07-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HASSIO

**ESP (ESP32/ESP8266, Board/Sonoff):**

SONOFF DUAL R2
**Affected component:**

Dual relay cover motor control

**Description of problem:**
When the cover is going down and I press the sonoff button, the cover stops (as expected), but when I push the button again, it continues going down instead of going up, I have to wait untill it goes fully down to press the button again and it will go up 

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

I use this exactly configuration: https://www.esphome.io/cookbook/dual-r2-cover.html#see-also

glmnet, 2019-07-22:
Try this code
````yaml
      - lambda: |
          if (id(my_cover).current_operation == COVER_OPERATION_IDLE) {
            // Cover is idle, check current state and either open or close cover.
            if (id(cover_going_up) || id(my_cover).is_fully_closed()) {
              id(my_cover).open();
              id(cover_going_up) = false;
            } else {
              id(my_cover).close();
              id(cover_going_up) = true;
            }
          } else {
            // Cover is opening/closing. Stop it.
            id(my_cover).stop();
          }
````
You'll need to add a variable
````yaml
globals:
- id: cover_going_up
  type: bool
  initial_value: 'false'
````

> Note
Controlling the cover to quickly (sending new open/close commands within a minute of previous commands) might cause unexpected behaviour (eg: cover stopping halfway). This is because the delayed relay off feature is implemented using asynchronous automations. So every time a open/close command is sent a delayed relay off command is added and old ones are not removed.

@OttoWinter is this obsolete?

hugodelaroza, 2019-07-23:
Thanks a lot Guillermo!!. It now really works fine!!. I have to tell that my configuration has another difference from the original script, I added the option esp8266_restore_from_flash proposed by Otto Winter in another thread so the device remember the position after an AC failure.

My complete script now is the following:

esphome:
  name: persiana01
  platform: ESP8266
  board: esp01_1m
  esp8266_restore_from_flash: true

wifi:
  ssid: 'ssid'
  password: 'password'

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'password'

ota:
  password: 'password'
  
binary_sensor:
- platform: gpio
  pin:
    number: GPIO10
    inverted: true
  id: button
  on_press:
    then:
      # logic for cycling through movements: open->stop->close->stop->...
      - lambda: |
          if (id(my_cover).current_operation == COVER_OPERATION_IDLE) {
            // Cover is idle, check current state and either open or close cover.
            if (id(cover_going_up) || id(my_cover).is_fully_closed()) {
              id(my_cover).open();
              id(cover_going_up) = false;
            } else {
              id(my_cover).close();
              id(cover_going_up) = true;
            }
          } else {
            // Cover is opening/closing. Stop it.
            id(my_cover).stop();
          }

globals:
- id: cover_going_up
  type: bool
  initial_value: 'false'

switch:
- platform: gpio
  pin: GPIO12
  interlock: &interlock [open_cover, close_cover]
  id: open_cover
- platform: gpio
  pin: GPIO5
  interlock: *interlock
  id: close_cover

cover:
- platform: time_based
  name: "Cover"
  id: my_cover
  open_action:
    - switch.turn_on: open_cover
  open_duration: 20s
  close_action:
    - switch.turn_on: close_cover
  close_duration: 20s
  stop_action:
    - switch.turn_off: open_cover
    - switch.turn_off: close_cover

I tink that the cookboob should be updated with something like this

Thanks a lot again!!

closed 2019-07-23 by hugodelaroza

PierreScerri, 2019-08-20:
Hi I had the same issue and was also resolved.

How can I add support for an external rocker switch (Open, Off, Down). 
The switch is connected to the 4-pin header of a Sonoff Dual R2 (labelled Button 0 & Button 1 & Gnd). GPIO0=Open, GPIO9=Close.

I have tried using the following but it is erratic:

```
binary_sensor:
- platform: gpio
  pin:
    number: GPIO0
    inverted: true
  id: open_button
  on_press:
    then:
      - switch.turn_on: open_cover
  on_release:
    then:
      - switch.turn_off: open_cover
      
- platform: gpio
  pin:
    number: GPIO9
    inverted: true
  id: close_button
  on_press:
    then:
      - switch.turn_on: close_cover
  on_release:
    then:
      - switch.turn_off: close_cover

```

Or have I got it all wrong?