esphome/issues #542: ESPHome Dashboard add a auto generated version number

fah, 2019-07-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HA ESPHome editor

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32dev

**Affected component:**

ESPHome Dashboard

**Description of problem:**

If you edit multiple devices you get lost in the changes you have done and have been uploaded.

##### Use case 1:
ESPHome dashboard:
For example, I edit a file, start the upload, go away and the upload fails. I don't see this in all the browser tabs that are open. 

##### Use case 2:
I plan to change some boards / names / definitions all over the place and prepare the code but do not upload. (Scenario: Mass change that needs a longer preparation.)
Currently I don't see what devices are up to date.

##### Solution:
Add an icon to see that the current code has been uploaded (and is running).

##### Implementation:
In the dashboard overview you could calculate a hash value out of the source code.

The device needs a new API call that returns the hash value of its source code.
Upload the hash value within the new API call to the device. 
So the dashboard can compare and show the result.

In the end, a new icon that symbolize a "device source code == dashboard source code" or the opposite, would show the need to upload the code.

For a beginner it is very helpful that they see that "there is something wrong". This new icon would be another, very important, indicator.

Why so complicated?
Just to remember that the upload of the current code has **been done** might be misleading as the device might be a different one or other problems with the upload could have been occured.

**Problem-relevant YAML-configuration entries:**
none

**Logs (if applicable):**

none

**Additional information and things you've tried:**

OttoWinter, 2019-10-13:
Yeah it's been suggested before and it is a really useful feature. But there are some technical challenges that make it hard.

1. 

> In the dashboard overview you could calculate a hash value out of the source code.

The problem here is the code is only generated once you click the compile button - so the hash would need to be based off the YAML file, not the .cpp file.

But that has another problem: Lots of people have setup where YAML files are interdependent - so you have one base file `base.yaml` shared across all devices and a per-device YAML file. The hash would need to include all dependencies as well, but the problem is that the YAML is (again) only loaded when you click on one of the action buttons (compile, upload, logs, ...).

Constantly loading the files in background would of course be an option but: 1. might be bad performance-wise;  2. loading+validating can generate exceptions, how should those be handled?

2.

> The device needs a new API call that returns the hash value of its source code

Second problem here. There's no unified API for all devices. Some use MQTT, some use native API, some use the webserver (which of course gives the user a lot of freedom). We cannot create implementations for each of them. Possible solution would be to use mDNS with a TXT record.

OttoWinter, 2019-11-04:
Whoops, noticed this is in the wrong repo. This is a feature request, not a bug report. Please file it in the correct repo: esphome/feature-requests

closed 2019-11-04 by OttoWinter