esphome/issues #543: AsyncMqttClient compile failed

b2un0, 2019-07-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
docker, esphome/esphome:latest (1.13.6)

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

[mqtt](https://esphome.io/components/mqtt.html)

**Description of problem:**

compiler false for new ESP32 Boards (would add more to my setup after a few weeks)

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp32_05
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "ABC"
  password: "XYZ"
  
ota:

web_server:
  port: 80

mqtt:
  broker: 192.168.177.10
  discovery: false
  discovery_retain: false
  topic_prefix: home
  log_topic: home/esp32_05/log
  birth_message:
    topic: home/esp32_05/status
    payload: online
  will_message:
    topic: home/esp32_05/status
    payload: offline
    
logger:
```

**Logs (if applicable):**

```
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esp32_05
Processing esp32_05 (platform: espressif32@1.6.0; framework: arduino; board: nodemcu-32s)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 > NodeMCU-32S
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <AsyncMqttClient> 0.8.2
|   |-- <AsyncTCP> 1.1.0
|-- <ESP Async WebServer> 1.1.1
|   |-- <AsyncTCP> 1.1.0
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Preferences> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling .pioenvs/esp32_05/libaae/AsyncTCP_ID1826/AsyncTCP.cpp.o
Compiling .pioenvs/esp32_05/libb1e/AsyncMqttClient_ID346/AsyncMqttClient/Packets/PubAckPacket.cpp.o
Compiling .pioenvs/esp32_05/libb1e/AsyncMqttClient_ID346/AsyncMqttClient/Packets/PubCompPacket.cpp.o
Compiling .pioenvs/esp32_05/libb1e/AsyncMqttClient_ID346/AsyncMqttClient/Packets/PubRecPacket.cpp.o
.piolibdeps/AsyncTCP_ID1826/src/AsyncTCP.cpp: In function 'bool _start_async_task()':
.piolibdeps/AsyncTCP_ID1826/src/AsyncTCP.cpp:205:141: error: 'xTaskCreateUniversal' was not declared in this scope
xTaskCreateUniversal(_async_service_task, "async_tcp", 8192 * 2, NULL, 3, &_async_service_task_handle, CONFIG_ASYNC_TCP_RUNNING_CORE);
^
Compiling .pioenvs/esp32_05/libb1e/AsyncMqttClient_ID346/AsyncMqttClient/Packets/PubRelPacket.cpp.o
Compiling .pioenvs/esp32_05/libb1e/AsyncMqttClient_ID346/AsyncMqttClient/Packets/PublishPacket.cpp.o
Compiling .pioenvs/esp32_05/libb1e/AsyncMqttClient_ID346/AsyncMqttClient/Packets/SubAckPacket.cpp.o
*** [.pioenvs/esp32_05/libaae/AsyncTCP_ID1826/AsyncTCP.cpp.o] Error 1
========================== [ERROR] Took 5.14 seconds ==========================
```

**Additional information and things you've tried:**

i copied the yaml from esp32_04 and deleted only the BLE stuff to bring up the new esphome device. its the same hardware device.

brumela, 2019-07-29:
I can confirm this error on esp32. 
esphome version is 1.13.6.

neographikal, 2019-07-29:
Confirmed on ESP32 nodemcu.

brandond, 2019-07-29:
Duplicate of esphome/esphome#653

This issue is fixed in dev.

closed 2019-07-29 by brandond