esphome/issues #545: ESP32 Ota Fail

leofalp, 2019-07-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant 0.96.3 
Based on x86 pc and Ubuntu server

**ESP (ESP32/ESP8266, Board/Sonoff):**

I'm using a esp32-cam module that works fine with Ai-Thinker configuration

**Affected component:**

OTA Update

**Description of problem:**
Ota update does not working:
```
INFO Successfully compiled program.
INFO Resolving IP address of espcamera1.local
INFO  -> 192.168.8.123
INFO Uploading /data/espcamera1/.pioenvs/espcamera1/firmware.bin (914384 bytes)
Uploading: [                                                            ] 0% 
ERROR Error sending data: [Errno 104] Connection reset by peer
```

**Problem-relevant YAML-configuration entries:**
```
esphome:
  name: espcamera1
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "***"
  password: "**"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "**"

ota:
  password: "**"

# Example configuration entry
esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32

  # Image settings
  name: "Telecamera 1"
  resolution: 640x480
  jpeg_quality: 10

```

**Logs (if applicable):**

```
[0;36m[D][ota:072]: Starting OTA Update from 192.168.8.5...[0m
[0;36m[D][ota:243]: OTA in progress: 0.1%[0m
abort() was called at PC 0x400ed1a1 on core 1

Backtrace: 0x4008f7f8:0x3ffb1970 0x4008fa29:0x3ffb1990 0x400ed1a1:0x3ffb19b0 0x40082b1b:0x3ffb19d0 0x400dcdc9:0x3ffb19f0 0x400dcaac:0x3ffb1a10 0x400dccd9:0x3ffb1a30 0x400d540a:0x3ffb1a50 0x400d5666:0x3ffb1f30 0x400d8f62:0x3ffb1f50 0x400d7c29:0x3ffb1f70 0x400d9912:0x3ffb1f90 0x400de02d:0x3ffb1fb0 0x40091bc5:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6400
entry 0x400806a8
[0;32m[I][logger:116]: Log initialized[0m
[0;35m[C][ota:364]: There have been 0 suspected unsuccessful boot attempts.[0m
[0;32m[I][app:028]: Running through setup()...[0m
[0;35m[C][wifi:029]: Setting up WiFi...[0m
[0;36m[D][wifi:272]: Starting scan...[0m
[0;36m[D][wifi:287]: Found networks:[0m
[0;32m[I][wifi:322]: - '***' [5m(E4:95:6E:43:23:37) [6m[0;33m▂▄[0;37m▆█[0m[0m
[0;36m[D][wifi:323]:     Channel: 2[0m
[0;36m[D][wifi:324]:     RSSI: -77 dB[0m
[0;36m[D][wifi:326]: - [5m'RLNet-681'[6m [5m(CC:2D:E0:DF:6C:A1) [6m[0;33m▂▄[0;37m▆█[0m[0m
[0;32m[I][wifi:164]: WiFi Connecting to '***'...[0m
[0;32m[I][wifi:380]: WiFi connected![0m
[0;35m[C][wifi:254]:   SSID: [5m'***'[6m[0m
[0;35m[C][wifi:255]:   IP Address: 192.168.8.123[0m
[0;35m[C][wifi:257]:   BSSID: [5mE4:95:6E:43:23:37[6m[0m
[0;35m[C][wifi:258]:   Hostname: 'espcamera1'[0m
[0;35m[C][wifi:262]:   Signal strength: -73 dB [0;33m▂▄[0;37m▆█[0m[0m
[0;35m[C][wifi:263]:   Channel: 2[0m
[0;35m[C][wifi:264]:   Subnet: 255.255.255.0[0m
[0;35m[C][wifi:265]:   Gateway: 192.168.8.1[0m
[0;35m[C][wifi:266]:   DNS1: 192.168.8.1[0m
[0;35m[C][wifi:267]:   DNS2: 0.0.0.0[0m
[0;35m[C][ota:029]: Over-The-Air Updates:[0m
[0;35m[C][ota:030]:   Address: espcamera1.local:3232[0m
[0;35m[C][ota:032]:   Using Password.[0m
[0;35m[C][api:030]: Setting up Home Assistant API server...[0m
[0;32m[I][app:060]: setup() finished successfully![0m
[0;32m[I][app:096]: esphome version 1.13.6 compiled on Jul 23 2019, 13:37:44[0m
[0;35m[C][wifi:372]: WiFi:[0m
[0;35m[C][wifi:254]:   SSID: [5m'***'[6m[0m
[0;35m[C][wifi:255]:   IP Address: 192.168.8.123[0m
[0;35m[C][wifi:257]:   BSSID: [5mE4:95:6E:43:23:37[6m[0m
[0;35m[C][wifi:258]:   Hostname: 'espcamera1'[0m
[0;35m[C][wifi:262]:   Signal strength: -73 dB [0;33m▂▄[0;37m▆█[0m[0m
[0;35m[C][wifi:263]:   Channel: 2[0m
[0;35m[C][wifi:264]:   Subnet: 255.255.255.0[0m
[0;35m[C][wifi:265]:   Gateway: 192.168.8.1[0m
[0;35m[C][wifi:266]:   DNS1: 192.168.8.1[0m
[0;35m[C][wifi:267]:   DNS2: 0.0.0.0[0m
[0;35m[C][logger:137]: Logger:[0m
[0;35m[C][logger:138]:   Level: DEBUG[0m
[0;35m[C][logger:139]:   Log Baud Rate: 115200[0m
[0;35m[C][logger:140]:   Hardware UART: UART0[0m
[0;35m[C][esp32_camera:043]: ESP32 Camera:[0m
[0;35m[C][esp32_camera:044]:   Name: camera1[0m
[0;35m[C][esp32_camera:045]:   Board Has PSRAM: YES[0m
[0;35m[C][esp32_camera:047]:   Data Pins: D0:5 D1:18 D2:19 D3:21 D4:36 D5:39 D6:34 D7:35[0m
[0;35m[C][esp32_camera:048]:   VSYNC Pin: 25[0m
[0;35m[C][esp32_camera:049]:   HREF Pin: 23[0m
[0;35m[C][esp32_camera:050]:   Pixel Clock Pin: 22[0m
[0;35m[C][esp32_camera:051]:   External Clock: Pin:0 Frequency:20000000[0m
[0;35m[C][esp32_camera:052]:   I2C Pins: SDA:26 SCL:27[0m
[0;35m[C][esp32_camera:053]:   Reset Pin: -1[0m
[0;35m[C][esp32_camera:074]:   Resolution: 640x480 (VGA)[0m
[0;35m[C][esp32_camera:097]:   JPEG Quality: 10[0m
[0;35m[C][esp32_camera:099]:   Contrast: 0[0m
[0;35m[C][esp32_camera:100]:   Brightness: 0[0m
[0;35m[C][esp32_camera:101]:   Saturation: 0[0m
[0;35m[C][esp32_camera:102]:   Vertical Flip: ON[0m
[0;35m[C][esp32_camera:103]:   Horizontal Mirror: ON[0m
[0;35m[C][esp32_camera:120]:   Test Pattern: NO[0m
[0;35m[C][ota:029]: Over-The-Air Updates:[0m
[0;35m[C][ota:030]:   Address: espcamera1.local:3232[0m
[0;35m[C][ota:032]:   Using Password.[0m
[0;35m[C][api:103]: API Server:[0m
[0;35m[C][api:104]:   Address: espcamera1.local:6053[0m
[0;36m[D][esp32_camera:157]: Got Image: len=26738[0m
[0;36m[D][api:573]: Client 'Home Assistant 0.96.3 (192.168.8.5)' connected successfully![0m

```

**Additional information and things you've tried:**

lethalwallabies, 2019-08-22:
Was there any solution to this?

OttoWinter, 2019-08-27:
If you haven't solved the issue yet, please reproduce the issue while viewing the logs with the esphome logs command. That will auto-decode the stacktrace.

ArielBaravalle, 2019-08-31:
Hello
I have the same problem
 any solution to this?
I can flsha by FTDI  usb ,  and the device are online , but  I can't  with OTA

ArielBaravalle, 2019-08-31:
[image]

OttoWinter, 2019-08-31:
This appears to be linked to `esp32_camera` - Also try disabling all camera output while uploading (by stopping HA, or removing the device from HA temporarily)

ArielBaravalle, 2019-08-31:
> uploading

Hello Otto , thanks for your help.   Now try to remove from HA  , and have the same problem.
I see  the parameter reboot_timeout in my actual  configuration ,  can be this option  the cause of problem? 

`api:
  password: "1234"
  reboot_timeout: 0s`

ArielBaravalle, 2019-08-31:
this is a serial port log from the ESP  in the moment of the OTA progress:
```

[1B][0;36m[D][esp32_camera:157]: Got Image: len=36818[1B][0m
[1B][0;36m[D][esp32_camera:157]: Got Image: len=36340[1B][0m
[1B][0;36m[D][esp32_camera:157]: Got Image: len=36291[1B][0m
[1B][0;36m[D][ota:072]: Starting OTA Update from 192.168.1.47...[1B][0m
[1B][0;36m[D][ota:243]: OTA in progress: 0.1%[1B][0m
abort() was called at PC 0x400f0149 on core 1

Backtrace: 0x4008f848:0x3ffb1970 0x4008fa79:0x3ffb1990 0x400f0149:0x3ffb19b0 0x40082b2b:0x3ffb19d0 0x400dfd71:0x3ffb19f0 0x400dfa54:0x3ffb1a10 0x400dfc81:0x3ffb1a30 0x400d7d36:0x3ffb1a50 0x400d7f96:0x3ffb1f30 0x400db8fa:0x3ffb1f50 0x400da589:0x3ffb1f70 0x400dc47e:0x3ffb1f90 0x400e0fd5:0x3ffb1fb0 0x40091c15:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6400
entry 0x400806a8
[1B][0;32m[I][logger:116]: Log initialized[1B][0m
[1B][0;35m[C][ota:364]: There have been 0 suspected unsuccessful boot attempts.[1B][0m
[1B][0;32m[I][app:028]: Running through setup()...[1B][0m
[1B][0;36m[D][binary_sensor:033]: 'esp32cam2 Motion': Sending state ON[1B][0m
[1B][0;35m[C][light:094]: Setting up light 'esp32cam_2 light'...[1B][0m
[1B][0;36m[D][light:261]: 'esp32cam_2 light' Setting:[1B][0m
[1B][0;35m[C][wifi:029]: Setting up WiFi...[1B][0m
[1B][0;36m[D][wifi:272]: Starting scan...[1B][0m
[1B][0;36m[D][binary_sensor:033]: 'esp32cam2 timpre': Sending state OFF[1B][0m
[1B][0;36m[D][wifi:287]: Found networks:[1B][0m
[
```

OttoWinter, 2019-08-31:
@ArielBaravalle Looks like an issue with the ESP32 Camera driver library interacting weirdly with the ESP32 Core. Not much I can do here, you can try flashing with some newer arduino_version to the chip manually via USB and seeing if subsequent OTAs work better.

OttoWinter, 2019-10-22:
@john-arvid Please try with esphome-flasher

Skaronator, 2020-01-13:
@OttoWinter Just run in the same issue with my very first ESP32. USB flashing with esphome-flasher works flawless but OTA doesn't work at all.

I did a lot of try and error but I didn't had any success. I got the logs directly from the device while doing a OTA. Looks like the watchdogs kills the ESP due to a IDLE cpu core?

```
[19:18:22][D][debug:023]: ESPHome version 1.14.3
[19:18:22][D][debug:025]: Free Heap Size: 235780 bytes
[19:18:22][D][debug:053]: Flash Chip: Size=4096kB Speed=40MHz Mode=DOUT
[19:18:22][D][debug:086]: Chip: Model=ESP32, Features=WIFI_BGN,BLE,BT, Cores=2, Revision=1
[19:18:22][D][debug:088]: ESP-IDF Version: v3.2.3-14-gd3e562907
[19:18:22][D][debug:091]: EFuse MAC: 0000745A0FC4F5FC
[19:18:22][D][debug:143]: Reset Reason: Power On Reset
[19:18:22][D][debug:186]: Wakeup Reason: Unknown
[19:19:03][D][ota:072]: Starting OTA Update from 192.168.178.5...
[19:19:03][V][ota:097]: OTA features is 0x00
[19:19:03][V][ota:110]: Auth: Nonce is 3775663560c2d5a3ba47f13199347af1
[19:19:03][V][ota:130]: Auth: CNonce is aa73ab393d24078bb4efd9f9ae2b6be3
[19:19:03][V][ota:137]: Auth: Result is ced70704b272dfd3718c158b4172da2d
[19:19:03][V][ota:145]: Auth: Response is ced70704b272dfd3718c158b4172da2d
[19:19:03][V][ota:171]: OTA size is 814384 bytes
[19:19:03][D][Updater.cpp:136] begin(): OTA Partition: app1
[19:19:03][V][ota:219]: Update: Binary MD5 is f2774aed9ef41aabdab22856696de3bd
[19:19:03][D][ota:243]: OTA in progress: 0.1%
[19:19:04][D][ota:243]: OTA in progress: 9.6%
[19:19:05][D][ota:243]: OTA in progress: 19.2%
[19:19:06][D][ota:243]: OTA in progress: 29.2%
[19:19:07][D][ota:243]: OTA in progress: 38.8%
[19:19:08][D][ota:243]: OTA in progress: 48.8%
[19:19:09]E (50475) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[19:19:09]E (50484) task_wdt:  - IDLE0 (CPU 0)
[19:19:09]E (50484) task_wdt: Tasks currently running:
[19:19:09]E (50484) task_wdt: CPU 0: ipc0
[19:19:09]E (50484) task_wdt: CPU 1: IDLE1
[19:19:09]E (50484) task_wdt: Aborting.
[19:19:09]abort() was called at PC 0x400e260b on core 0
[19:19:09]
[19:19:09]Backtrace: 0x4008c5d0:0x3ffbe170 0x4008c801:0x3ffbe190 0x400e260b:0x3ffbe1b0 0x4008490d:0x3ffbe1d0 0x40082675:0x3ffb9d60 0x40086bde:0x3ffb9d80 0x40082d27:0x3ffb9da0 0x40088d19:0x3ffb9dc0
[33mWARNING Found stack trace! Trying to decode it[0m
[32mINFO Need to fetch platformio IDE-data, please stand by[0m
[32mINFO Running:  platformio run -d esp32_1 -t idedata[0m
[19:19:10]
[19:19:10]Rebooting...
[19:19:10]ets Jun  8 2016 00:22:57
[19:19:10]
[19:19:10]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[19:19:10]configsip: 0, SPIWP:0xee
[19:19:10]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[19:19:10]mode:DOUT, clock div:2
[19:19:10]load:0x3fff0018,len:4
[19:19:10]load:0x3fff001c,len:1044
[19:19:10]load:0x40078000,len:8896
[19:19:10]load:0x40080400,len:5828
[19:19:10]entry 0x400806ac
[19:19:10][I][logger:156]: Log initialized
[19:19:10][VV][preferences:038]: LOAD 0: valid=YES, 0=0x00000003 1=0x60BCD276 (Type=233825507, CRC=0x60BCD276)
[19:19:10][C][ota:364]: There have been 3 suspected unsuccessful boot attempts.
[19:19:10][VV][preferences:051]: SAVE 0: 0=0x00000004 1=0x31811791 (Type=233825507, CRC=0x31811791)
[19:19:10][I][app:028]: Running through setup()...
[19:19:10][V][app:029]: Sorting components by setup priority...
[19:19:10][C][wifi:033]: Setting up WiFi...
[19:19:10][V][wifi_esp32:033]: Enabling STA.
```

My yaml looks like this, basically bare metal:

```
esphome:
  name: esp32_1
  platform: ESP32
  board: esp32doit-devkit-v1

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pw

api:
  password: !secret api_pw

ota:
  password: !secret ota_pw
```

GuillaumeCale, 2020-01-24:
Same problem here, randomly while uploading in progress 20~85% ...

Skaronator, 2020-01-24:
@GuillaumeCale I made a PR to (presumably) fix this issue. Would be cool if you can try it out https://github.com/esphome/esphome/pull/959

GuillaumeCale, 2020-01-25:
Thank you for your response. I tried many things : more wifi power, more supply power... finally it worked when I restart the ESPhome docker + close all tabs ... maybe luck ?

Maco65, 2020-04-21:
I have the same issue. Randomly stopping OTA with progress from 0% to 92%.
Though that it was connected to WiFi signal bit then even when ESP32-cam is in the same room with router the issue exists.

Skaronator, 2020-04-21:
@Maco65 if you're using docker you could try my docker images which contains [a possible fix](https://github.com/esphome/esphome/pull/959) for this issue.

```
docker pull skaronator/esphome:latest
```

Only caveat is that you need to flash this on your device first. After that the OTA should work correctly.

Maco65, 2020-04-22:
@Skaronator - thanks. I am not sure if I am using docker. :( I use Home Assistant HassOS 3.13 and ESPHome Add-on. Can I use your fix then ?

glmnet, 2020-04-22:
The fix is also on dev already, so you can use the official dev version of ESPHome https://esphome.io/guides/faq.html#how-do-i-use-the-latest-bleeding-edge-version

Maco65, 2020-04-22:
@glmnet Geat !
I am so happy for esphome add-on and the was is so easy to program esp boards ! Great Job !
And I easily did dev add-on and now loaded new code to my esp32-cam.
We will see how it behaves now. :)

pjrt, 2020-06-15:
Hey. So I updated to the dev channel, but am still getting that error:

```
Uploading: [=                                                           ] 2% 
ERROR Error sending data: [Errno 104] Connection reset by peer
```
Odd that it goes up to 2%. 
Flashing the binary over USB works fine. 

However that raises a question: how does OTA update work? When flashing over USB I have to set the chip to flash mode. For OTA, I can't do that obviously.

closed 2020-12-05 by stale[bot]