esphome/issues #548: Template Switch Trigger Not Updating State in Home Assistant

cryptelli, 2019-07-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Python VENV

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**Affected component:**

https://esphome.io/components/switch/template.html

**Description of problem:**
After turning the switch off via Home Assistant the ESPHome configuration via the `on_turn_off` trigger should turn the switch back on if `switch1` isn't off. 

Through checking the logs of the flashed ESP I can see that the following is printed to the console, but the state of the switch inside Home Assistant remains off, any attempt to toggle the switch just causes it to toggle on then off straight away.

**Problem-relevant YAML-configuration entries:**
```yaml
- platform: template
  name: "Template Switch"
  id: temp_switch
  optimistic: true
  on_turn_off:
    if:
      condition:
        switch.is_off: switch1
      then:
        switch.turn_off: switch2
      else:
        - switch.turn_on: temp_switch

```

**Logs (if applicable):**

```
[23:01:02][D][switch:025]: 'Template Switch' Turning OFF.
[23:01:02][D][switch:045]: 'Template Switch': Sending state OFF
[23:01:02][D][switch:021]: 'Template Switch' Turning ON.
[23:01:02][D][switch:045]: 'Template Switch': Sending state ON
```

**Additional information and things you've tried:**
I discovered if Home Assistant is restarted the correct state of the switch is represented. In the above case it shows the switch is `on`.

glmnet, 2019-07-24:
I guess that if you try to control the state of the switch you should not use `optimistic`

cryptelli, 2019-07-24:
Thanks for your comment. I realise that, the issue is the switch which I'm trying to control the state of is virtual. Any suggestions on what I can do in this instance?

glmnet, 2019-07-24:
Start by removing `optimistic: `
then you have two options:
a) implement `lambda: `
b) use `switch.template.publish:`

I guess this should work:
````yaml
- platform: template
  name: "Template Switch"
  id: temp_switch
  on_turn_on:
    switch.template.publish:
      id: temp_switch
      state: ON
  on_turn_off:
    if:
      condition:
        switch.is_off: switch1
      then:
        - switch.turn_off: switch2
        - switch.template.publish:
            id: temp_switch
            state: OFF
      else:
        - switch.template.publish:
            id: temp_switch
            state: ON
````

cryptelli, 2019-07-24:
I'd experimented with `switch.template.publish`. I've tried your suggested code but it only seems to send `'temp_switch' Turning ON`.

The last `else` is designed so that if the `then` statement fails it resets the state of the switch from `off` to `on` accurately displaying the current status of the switch that it was trying to turn off.

glmnet, 2019-07-24:
If you remove the `optimitstic` the switch state should not change by Home assistant request, hence removing the else effectively keeps temp_switch on when switch1 is off

brandond, 2019-07-24:
What are you actually trying to do with this template switch? Perhaps if you posted the full YAML along with an explanation of what you're trying to do we could suggest a better way to accomplish it.

cryptelli, 2019-07-25:
I posted the YAML which I felt was relevant to the question (apart from the switch definitions). 

After a little bit of experimentation I was able to solve my problem. The key was using `turn_on_action` and `turn_on_action` actions.

```yaml
switch:
  - platform: gpio
    pin: GPIO4
    name: "Switch 1"
    id: switch1
    inverted: true
    on_turn_on:
      switch.template.publish:
        id: temp_switch
        state: ON

  - platform: gpio
    name: "Switch 2"
    pin: GPIO5
    id: switch2
    inverted: true

  - platform: template
    name: "Template Switch"
    id: temp_switch
    turn_on_action:
      - switch.template.publish:
          id: temp_switch
          state: ON
      - switch.turn_on: fan_relay
    turn_off_action:
      if:
        condition:
          switch.is_off: switch1
        then:
          - switch.turn_off: switch2
          - switch.template.publish:
              id: temp_switch
              state: OFF
        else:
          - switch.template.publish:
              id: temp_switch
              state: ON
```

closed 2019-07-25 by cryptelli

glmnet, 2019-07-25:
My bad...
I thought we were using the `turn_on_action` istead of `on_turn_on`.
Those are a bit confusing, I already asked @OttoWinter  to show a warning if the user was not implementing those and also `optimistic` was not set to true.
I cannot think of a case that would make sense. @brandond what do you think?
- Using `optimistic` true is neat, you can quickly have a switch hosted in ESPHome which you can control from HA, then you can use this switch `switch.is_on` condition on another automation.
- If you don't set `optimistic` then you want to know as soon as the switch is toggled from HA, so you need to implement `turn_on_action` otherwise the switch state is not updated and you are not doing anything with it either.

You can anyway publish the state however in that case a `binary_sensor` should make more sense.
So I vote for a **WARN** message here

cryptelli, 2019-07-25:
All good. I have to agree with your comments @glmnet, it's somewhat confusing until I looked the function of `turn_on_action` and `on_turn_on` then realised I needed to change tack. Working exactly as intended now.

At least this issue has sparked a conversation, and hopefully a **WARN** could be introduced to prevent others from stumbling down the same path.