esphome/issues #550: status_led does not light up on a failing i2c sensor

valordk, 2019-07-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** HA installed in vmware

**ESP (ESP32/ESP8266, Board/Sonoff):** ESP32 WROOM 32D

The ESP32 is mounted on a custom breakout board with the required pullup resistors, capacitors and 3.3V LDO (NCP114)

**Affected component:** Status LED

https://esphome.io/components/status_led.htm

**Description of problem:**
The **sensor_led** component which is supposed to blink on warning or error events risen from other components does not do so when an i2c sensor is diconnected.

The DUT is an ESP32D with a red LED wired to IO12 through a 10K resistor.

During the bootup, or if the wifi gets diconnected, the LED blinks as expected until the connection is reestablished., so the wiring and the basic functionality of the Status LED appears to be as expected.

But if the i2c sensors are disconnected from the board, the Status LED does not begin to blink, eventhough the i2c component starts throwing following WARNING events:
`[16:42:03][W][i2c:070]: Received NACK on transmit of address 0x4B`

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: sensornode
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "ssid"
  password: "passwd"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "passwd"

ota:
  password: "passwd"
i2c:
  sda: 21
  scl: 22
  scan: True
  frequency: 50kHz  
sensor:
  - platform: wifi_signal
    name: "sensornode wifi signal sensor"
    update_interval: 60s
  - platform: uptime
    name: "sensornode uptime sensor"
    update_interval: 60s
  - platform: sht3xd
    temperature:
      name: "sensornode SHT30 0x44 Temperature"
      accuracy_decimals: 2    
    address: 0x44
    update_interval: 10s
  - platform: sht3xd
    temperature:
      name: "sensornode SHT30 0x45 Temperature"
      accuracy_decimals: 2    
    address: 0x45
    update_interval: 10s 
status_led:
  pin:
    number: 12

```

**Logs (if applicable):**

```
[16:59:09][D][api.connection:575]: Client 'Home Assistant 0.95.4 (x.x.x.x)' connected successfully!
[16:59:13][D][sensor:092]: 'SHT30 0x44 Temperature': Sending state 27.32853 °C with 2 decimals of accuracy
[16:59:16][D][sensor:092]: 'SHT30 0x45 Temperature': Sending state 27.15762 °C with 2 decimals of accuracy
[16:59:23][E][i2c:070]: Received NACK on transmit of address 0x44
[16:59:25][E][i2c:070]: Received NACK on transmit of address 0x45

```

**Additional information and things you've tried:**
As you can see from the above log, i've even forced the i2c component to log these events on an ERROR level instead of WARNING, but it did not help.

For information, the sensors begin to work fine after hot-plugging the connector back into the board.

glmnet, 2019-07-25:
Ahh yes...
Seems to be exiting [here](https://github.com/esphome/esphome/blob/a783637a7a4ccf354021df5577d3093aac02a128/esphome/components/sht3xd/sht3xd.cpp#L47) and not calling `this->status_set_warning()` before

valordk, 2019-07-25:
Good catch. I can try fixing it on the existing SHT3xd and the STS3x which I have pull-request for.

closed 2019-10-13 by OttoWinter