esphome/issues #551: esp32-cam stops working

Iv711, 2019-07-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

esp32-cam (https://ru.aliexpress.com/item/32990986108.html?spm=a2g0s.9042311.0.0.418f33edynENUb)
**Affected component:**

https://esphome.io/components/esp32_camera.html
https://esphome.io/cookbook/pir.html

**Description of problem:**
Immediately I apologize for the English language, I use Google.
"ESP32" freezes after 10-20 hours of operation. The camera ceases to give the image, the sensors hang. But restart.switch works programmatically. After the reboot, the board starts working normally, but again for several hours.
I'll try. All sensors and cameras are hanging, including uptime. except the reset button.

**Problem-relevant YAML-configuration entries:**
```yaml
esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  power_down_pin: GPIO32
  pixel_clock_pin: GPIO22
  resolution: 640x480
  jpeg_quality: 10
  name: $hostname cam
  
binary_sensor:
  - platform: gpio
    pin: GPIO2
    name: $hostname pir Sensor
    device_class: motion
  - platform: status
    name: $hostname status    
  - platform: gpio
    pin: GPIO16
    name: $hostname button

sensor:
  - platform: wifi_signal
    name: $hostname WiFi Signal
    update_interval: 60s
  - platform: uptime
    name: $hostname Uptime

text_sensor:
  - platform: version
    name: $hostname ESPHome Version
  - platform: wifi_info
    ssid:
      name: $hostname WiFi

switch:      
  - platform: restart
    name: $hostname restart

```

**Logs (if applicable):**

**Additional information and things you've tried:**

saikek, 2019-07-26:
I have simmilar issues - I'm using:
* ESP32camera + AiThinker module 
* 3dprinted case 
* with decent radiator attached. (which is getting hot).

I wish that there would be a turn on/off toggle so i can use camera only when motion is detected for example. (when i check logs i see that it's sending frames all the time to HA, even if no client is watching)

OttoWinter, 2019-07-28:
Probably related to https://github.com/esphome/issues/issues/552

> I wish that there would be a turn on/off toggle so i can use camera only when motion is detected for example.

The camera does not know that - the ESP only receives the *already encoded* JPEG frame, and cannot process it in any way (for example checking for motion).

> when i check logs i see that it's sending frames all the time to HA, even if no client is watching

Check the idle framerate, otherwise you probably have a client connected somewhere that is requesting frames.

AnthonyKNorman, 2019-09-06:
I am also having this issue. I have attached USB Serial to one of my cams and am recording the log. Will post here if I can see any messages on failure. Should I set debug to 'verbose'?.

mpettitt, 2019-09-06:
Looks like when Home Assistant opens the camera view, it switches to showing the max_framerate setting, which is correct, but when the camera view is closed, no signal is sent to switch the camera back to "idle" mode. That makes sense - for any other component in Home Assistant, closing the UI window means "I'm done, everything I've changed is what I wanted" - but for the camera, it seems to need an additional "OK, switch back to low frequency mode" call.

abmantis, 2019-09-06:
> Looks like when Home Assistant opens the camera view, it switches to showing the max_framerate setting, which is correct, but when the camera view is closed, no signal is sent to switch the camera [...]

I also see this behaviour.

wrobi, 2019-09-28:
Earlier I thinked the freezing period depends on power supply. I've been tested for few weeks. On first week only checked via browser. Stopped every day. After changed power supply it works 3-4 days. Today tested with motion detection and stoped after 3 hours. Looks like doesn't like continous operation. Maybe overheating can be the problem. Should be measured.

cvdabbeele, 2019-10-13:
Hi,
I am having the same issue: after a few hours of operation (anything between, say 4 and 24), I lose the camera feed in Home Assistant.  However, from HomeAssistant, **I can still control the LED on the ESP32-CAM board**.  My ESP32 chip does not get hot.  I also tried with different power supplies.

I also found a reference to `"reboot_timeout"`, mentioning that the ESP32 currently has issues with the IP stack and that a full reboot is required to get the interface back working.  ( https://esphome.io/components/api.html?highlight=reboot_timeout#native-api-component )
and this: https://esphome.io/guides/automations.html?highlight=reboot_timeout  
the last one is about MQTT, but I am using the native API
Anyhow, I'm giving it a try with reboot_timeout=1
@AnthonyKNorman  

> I have attached USB Serial to one of my cams and am recording the log

Did you find anything useful in your USB logging?
@mpettitt Your finding on 

> "Home Assistant, closing the UI window" 

may be related to the error I'm seeing:
2019-10-13 16:51:23 WARNING (MainThread) [homeassistant.core] Unable to remove unknown listener <function EventBus.async_listen_once.<locals>.onetime_listener at 0x72f43468

Thank you all for sharing any new findings or theories to test to fix this
chris

cvdabbeele, 2019-10-14:
ok, that didn't help either.
The http feed still stops working after a few hours.  The board remains responsive to GPIO commands

saikek, 2019-10-15:
Is there a chance that anyone have JTAG programmer that can be connected to ESP32 ? 
It would be possible to collect log and trace Deadlocks / memory leaks.

cvdabbeele, 2019-10-15:
Hi,
Here is some more information  (probably provided by others as well)
Is there a way to get debug logging via the esphome-flasher?

**Symptom**
ESP32 and cam are working fine.  After a few hours the cam stops sending images to Home Assistant.  The GPIOs still react to commands from Home Assistant.  My board was flashed with `esphome version 1.14.0-dev compiled on Oct 13 2019, 18:05:58`

**From the communication of the ESPhome-flasher on the USB port**
```
[[18:12:27][D][esp32_camera:157]: Got Image: len=31301
[18:12:27][D][esp32_camera:157]: Got Image: len=31376
[18:12:27][D][esp32_camera:157]: Got Image: len=31393
[18:12:28][D][esp32_camera:157]: Got Image: len=31351
[18:12:28][D][esp32_camera:157]: Got Image: len=31384
[18:12:28][D][esp32_camera:157]: Got Image: len=31198
[18:12:29][D][esp32_camera:157]: Got Image: len=31338
[18:12:29][D][esp32_camera:157]: Got Image: len=32000
[18:12:29][D][esp32_camera:157]: Got Image: len=31315
[18:12:30][D][esp32_camera:157]: Got Image: len=31293
[18:12:30][D][esp32_camera:157]: Got Image: len=31346
[18:12:30][D][esp32_camera:157]: Got Image: len=31286
[18:12:31][D][esp32_camera:157]: Got Image: len=31302
[18:12:31][D][esp32_camera:157]: Got Image: len=31319
[18:12:31][D][esp32_camera:157]: Got Image: len=31325
[18:12:31][D][esp32_camera:157]: Got Image: len=31307
[18:12:32][D][esp32_camera:157]: Got Image: len=31360
[18:12:32][D][esp32_camera:157]: Got Image: len=31308
[18:12:32][D][esp32_camera:157]: Got Image: len=31254
[18:35:13][D][light:261]: 'camfrontgarden flash' Setting:
[18:35:13][D][light:270]:   State: ON
[18:35:15][D][light:261]: 'camfrontgarden flash' Setting:
[18:35:15][D][light:270]:   State: OFF

```

At 18:12:32, all of a sudden, it stops sending images
At 18:35:13, I had noticed it and I turned on the LED on the ESP32-cam board, from Home Assistant. The board reacted as expected and I then turned the LED (the flash) back off.  This functionality seems to still work

**From the LogBook**
Note that there is no entry for that time (6:12PM)

```6:40 PM  FrontGarden changed to unavailable
6:40 PM  camfrontgarden flash changed to unavailable
6:35 PM  camfrontgarden flash turned off
6:35 PM  camfrontgarden flash turned on
4:46 PM  Home123 changed to cloudy
2:38 PM  PIR Sensor on ESP32cam detected motion
2:38 PM  FrontGarden changed to idle
2:38 PM  camfrontgarden flash turned off
2:37 PM  PIR Sensor on ESP32cam changed to unavailable
2:37 PM  FrontGarden changed to unavailable
2:37 PM  camfrontgarden flash changed to unavailable
1:39 PM  Home123 changed to rainy
12:58 PM  PIR Sensor on ESP32cam detected motion
12:58 PM  FrontGarden changed to idle
12:58 PM  camfrontgarden flash turned off
12:58 PM  PIR Sensor on ESP32cam changed to unavailable
12:58 PM  FrontGarden changed to unavailable
12:58 PM  camfrontgarden flash changed to unavailable
12:48 PM  camfrontgarden flash turned off
12:48 PM  camfrontgarden flash turned on
12:48 PM  PIR Sensor on ESP32cam detected motion
12:48 PM  FrontGarden changed to idle
12:48 PM  camfrontgarden flash turned off
12:38 PM  PIR Sensor on ESP32cam changed to unavailable
```
**States**
[image]
Trying to manually change the state to different states ("on", "unavailable", "idle", back to "on") doesn't seem to have any effect.  The state changes in the "state" view and under "states" I do see the  `state_changed` events, but the camera does not re-engage

**/config/home-assistant.log**
no entries for the last hours previous to the "event"

**Power Supply**
I have seen this issue with different prower supplies.  Including a 5V / 2.5A psu and the USB port from my monitor

**Chip Temperature**
During operation, the chip gets warm, but not hot

**Next steps**
I do not have a JTAG programmer, but **_I'd be happy to insert additional text/commands in the YAML file and re-flash the board so that maybe we can get a clearer view on the situation.  Unfortunately I have no idea what to put in there.  I am open to suggestions._**  My YAML is pretty similar to the one that  @Iv711  posted earlier

OttoWinter, 2019-10-15:
@cvbabbeele That sounds a lot like the behavior seen when the camera overheats (or least that was the case when I developed the integration with cheap camera modules like the M5Cam) - the camera would no longer receive any images and if given enough time to cool down with power down it works again. Please try using the test pattern option, if that works then it's overheating.

KiLLeRRaT, 2019-10-15:
I use my at the front door purely after a PIR motion trigger event.  I'm going to create a gpio switch that turns off the `power_down_pin: GPIO32`.  If I get motion on the PIR, it turns on the pin, and takes my snapshots, and turn off the gpio again.  

If I want to view the stream in HA, I will have to turn on the GPIO pin, watch the stream, and then turn off the GPIO pin again. Not ideal but if this is more reliable, I can live with it.

Let's see if that works eh...

cvdabbeele, 2019-10-15:
@OttoWinter thank you for the tips Otto.  You seem to know what you are talking about.  The last micro controller I worked with was a Z80 (yes, the 1980's) I'm getting my second wind now with ESPhome and Hassio.  
Anyway, the Test pattern is on.  I'll let it braise for a night or two and will report back.

**My nickel just dropped.**  
When you say that the camera overheats, you actually mean the camera, the _**camera sensor**_ and not the ESP32 chip/board.  Duh,... how could I have missed that.  And yes. It was **hot** !  On top of that, it was almost pressed upon the warm ESP board.  Recipe for disaster.
Having high hopes  
ChrisV

cvdabbeele, 2019-10-15:
@OttoWinter I'm not sure that I understand the purpose of testing with the Test Pattern.  I noticed that with the test pattern, the sensor is still getting very hot.  Could it be that the test pattern is generated by the sensor and not by the ESP32 chip?   
So, unfortunately, after a while, the same issue returned:  the camera stopped sending images
I will do some more testing with cooling etc.. during the weekend, but I am afraid those cheap boards are not made to run 24/7 and we may have to look into a solution like the one that @killerrat suggested, where we turn on the camera from a PIR sensor.  
But that's a discussion for another forum
chrisV

OttoWinter, 2019-10-17:
>  I noticed that with the test pattern, the sensor is still getting very hot

Ok for me that was different. With the test pattern the camera didn't get that hot.

> Could it be that the test pattern is generated by the sensor and not by the ESP32 chip?

Yes, it is.

saikek, 2019-10-17:
I've pointed my camera to the celling and am capturing logs for 24H+ now.
Since not much is going on the frame it seems to be ok. Are there any controls that can decrease camera / ESP32 calculations - exposure, frames count, frame quality, night vision, etc... ?

cvdabbeele, 2019-10-17:
ok, the heat issue is probably more related to the frame rate than to the content of the image (test pattern, static/ceiling)  I noticed that the frame rate changes whether home assistant has to serve the cam view to a client or not.  This seams to match with the docs: [https://esphome.io/components/esp32_camera.html](url)   

> max_framerate (Optional, float): The maximum framerate the camera will generate images at. Up to 60Hz is possible (with reduced frame sizes), but beware of overheating. Defaults to 10 fps.

> idle_framerate (Optional, float): The framerate to capture images at when no client is requesting a full stream. Defaults to 0.1 fps.

I'll experiment a bit by capping the max_framerate

diogopms, 2019-10-18:
Hey, thanks for creating this issue.... 🙏 👍 
Do you know a good esp32 cam hardware model, without overheating problem?

cvdabbeele, 2019-10-20:
@diogopms So far, it is my understanding that the problem lies with the camera sensor of the ESP32-cam.  This sensor is the OV2640.  Please verify if it is your Cam sensor (lens) which is getting hot, or if it is your ESP32 chip/board.  AFAIK, all of those ESP boards use the OV2640.  I am not familiar with any other cam sensors

OttoWinter, 2019-10-20:
OV3660 is theoretically also supported

bastshoes, 2019-10-27:
Changing board to esp32dev brought stability to module.

Iv711, 2019-10-29:
> Changing board to esp32dev brought stability to module.

I also started working in this configuration